RIOT-OS/RIOT #9938: Executing commands like udp from main.c without user input

npcode15, 2018-09-14:
Hello everyone, I would like to execute functions like udp from main.c i.e. instead of executing udp command over serial command line, I would like to execute that as a part of main.c file (right after compilation). I looked online but I couldn’t find anything relevant. Please help, thanks.

Also, if possible, just for information, I would like to know why was this post rejected by user mailing list. Thanks.

OlegHahm, 2018-09-14:
I guess you're referring to the shell command `udp` from the `gnrc_networking` example application. The implementation of the function that is called by the shell command can be found [here](https://github.com/RIOT-OS/RIOT/blob/334f828599d7cc25e06a8ceaf6e88914fc4160a7/examples/gnrc_networking/udp.c).

> Also, if possible, just for information, I would like to know why was this post rejected by user mailing list.

I don't think that happened on purpose. Could you forward the notification you get to me, please? oleg@riot-os.org Thanks!

npcode15, 2018-09-14:
Sure, I will forward the email.

Regarding my question, I will have a look at the implementation and get back, if I have questions.

Thank you.

npcode15, 2018-09-15:
I also have a follow-up question. I don't know, if I should post this as another question. I was wondering if its possible to differentiate between the virtual tap interfaces from main.c file. I basically want to send some information from sender (tap0) to receiver (tap1), but I would like to vary the code slightly based on sender/receiver. Thanks.

OlegHahm, 2018-09-19:
Please take a look at https://github.com/RIOT-OS/Tutorials/tree/master/task-06

npcode15, 2018-09-19:
@olegart, I already looked at this example.

Basically, I want to be able to repeat the ipc_pingpong example (_https://github.com/RIOT-OS/RIOT/tree/master/examples/ipc_pingpong_) but for 2 nodes and not for 2 processes.
In order to reply, I would like to know (within c file) that I received a message. I did find something, please have a look at line 163 in file gntc_pktdump.c 
_https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktdump/gnrc_pktdump.c_

Screenshot of the same.
[image]

Could you please explain, why is there a need to check that the type of '**msg**' could be '**GNRC_NETAPI_MSG_TYPE_SND**'. I think it could be possible to use/modify this code to achieve what I want.

Please let me know if the question was unclear. Thanks.

closed 2018-09-21 by npcode15