RIOT-OS/RIOT #9943: Device not Detected by RIOT

npcode15, 2018-09-15:
Hello, I am trying to use nrf5280dk with RIOT OS. I am following these steps to enable access to the device from os.
https://devzone.nordicsemi.com/b/blog/posts/setting-up-ipv6-over-ble-using-nrf52-series-and-ri

**Setup**: Running RIOT OS with vagrant (Virtual Box) in Windows 10 Host machine.

I installed the required tools. But, I am getting the following error when I try to connect with the device (nrF52dk)
[image]

Windows successfully detects the device, I am able to see the JLink folder. I enabled the device in VirtualBox settings. So, as soon as I run the RIOT OS (vagrant up + vagrant ssh), the device gets detached from Host Machine. But, I have no idea if the virtual machine is detecting the device. I could not find a way to check it. I could not find a 'mnt' folder to check if the device got mounted. There no UI. 
I tried checking RIOT/dev. But, it did not help either.

Also, I tried using commands like dmesg. 
[image]

Please help. Thank you.

aabadie, 2018-09-16:
Normally the system in the VM is Linux so if the board is correctly detected in the VM, you should have a new serial port in `/dev/ttyACM0`.
Another way to check if the board is detected inside the VM is to look at Linux kernel messages with `dmesg` just after you plugged the board:
```
dmesg -T
```

kYc0o, 2018-09-17:
AFAIK JLink is not part of the automatically attached USB devices. You might need just to attach it through the GUI of VirtualBox.

npcode15, 2018-09-19:
I could not attach the device with RIOT with vagrant. So, I switched to RIOT within Ubuntu (using VMWare).

Thanks, @kYc0o @aabadie. I am gonna close this issue.

closed 2018-09-19 by npcode15