RIOT-OS/RIOT #9950: Cannot Flash nrf52dk.

npcode15, 2018-09-18:
There seems to a conflict between toolchain and RIOT OS (hard compilation vs soft compilation). Does anything come to your mind? I am using GCC ARM toolchain.

Refer to this image for exact error.
[image]

Is there a particular toolchain I should be installing.

**Setup**: RIOT inside Virtual Ubuntu (Windows Host)
**Device**: nrf52840dk

**NOTE: Although, Hello World example gets flashed, but anything other than that doesn't.**

Thanks.

aabadie, 2018-09-18:
You need arm-none-eabi-gcc 7.2 at least and you are using 6.3.1

npcode15, 2018-09-18:
Hello @aabadie, I tried updating. Ubuntu says it is already updated.
[image]

Is this a correct link to update toolchain?
https://gnu-mcu-eclipse.github.io/toolchain/arm/install/

or launchpad one?

Thanks.

aabadie, 2018-09-18:
> or launchpad one?

Yes prefer this one

npcode15, 2018-09-18:
@aabadie 
There is no installer available for version 7.
RIOT is not finding the newly added version. (Added to path)

Is there a way to configure RIOT to check the new directory for arm-toolchain?

aabadie, 2018-09-18:
> Successfully compiled. Thanks.

Great! So let's close this one then.

closed 2018-09-18 by aabadie