RIOT-OS/RIOT #9957: Ping no NETDEV_IEEE802154_ACK_REQ enabled

Josar, 2018-09-18:
While Tetsing for https://github.com/RIOT-OS/RIOT/pull/9509 it seemed that the `NETDEV_IEEE802154_ACK_REQ` bit was removed from the ping packets.

in Previous releases i could see ACK packages in Wireshark now they are not send anymore.
Also the ACK request bit is not set in the packet.
And with packets which needs fragmentation a lot of packets are lost. 
Any idea why?

gebart, 2018-09-19:
Do you have the ACK_REQ option set on the interface?
If not, that should probaly be set by gnrc during init as that would likely be what most users expect..
```
ifconfig
Iface  7  HWaddr: 6d:4e  Channel: 26  Page: 0  NID: 0x23
          Long HWaddr: 23:35:63:2c:63:01:6d:4e 
           TX-Power: 0dBm  State: SLEEP  max. Retrans.: 0 
          ACK_REQ  PRELOAD  MTU:1280  HL:64  RTR           <<<<<<<======= Here
          IPHC  
          Source address length: 8
          Link type: wireless
          inet6 addr: fe80::2135:632c:6301:6d4e  scope: local  VAL
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff01:6d4e
          inet6 group: ff02::1a
          
          Statistics for Layer 2
            RX packets 2656  bytes 204499
            TX packets 0 (Multicast: 0)  bytes 208775
            TX succeeded 53710 errors 11606
          Statistics for IPv6
            RX packets 1857  bytes 217658
            TX packets 1952 (Multicast: 1063)  bytes 223632
            TX succeeded 1952 errors 0
```

miri64, 2018-09-19:
For the record, I assigned @bergzand because I remember there were some PRs of his that changed the place *where* the flags are set, but IIRC not how they were set. Maybe he can shed some light on this.

bergzand, 2018-09-29:
Okay, If I read the code correctly (it is saturday afternoon), the ACK_REQ flag was never configured in the first place. And because I fixed a "bug" in #9581, it is now no longer enabled by default.

PeterKietzmann, 2018-10-04:
> it is now no longer enabled by default.

Again here: is it intended to disable by default? Personally I would say no!

bergzand, 2018-10-04:
> Again here: is it intended to disable by default? Personally I would say no!

Let me get a few things straight before this escalates. I agree that ACK requests should be enabled by default.

What I suspect happened is that the ack req was always enabled due to a small bug in the flag handling and that this bug became a feature at some point.

@PeterKietzmann I plan on finalizing #9592 one of these days and revert to the old behaviour (having ack requests enabled by default)

PeterKietzmann, 2018-10-04:
Super, thanks!

bergzand, 2018-10-17:
@PeterKietzmann Since I didn't find any time to finalize #9592 before the high impact merge window (and I consider that one to be high impact), how about I add the ACK_REQ flag to the radio initialization for now to have an easy fix for restoring the old behaviour?

PeterKietzmann, 2018-10-19:
Sounds reasonable in general. @jia200x opinions?

jia200x, 2018-10-25:
yes, let's do it. Otherwise the release tests might give unexpected results :/

jia200x, 2018-10-25:
ping @bergzand

bergzand, 2018-10-25:
@jia200x @PeterKietzmann fixed in #10250

miri64, 2018-10-25:
#10250 was merged. Can we close this?

jia200x, 2018-10-25:
before closing it, @bergzand is the ACK_REQ flag enabled by default in the other radios?

bergzand, 2018-10-25:
@jia200x AFAIK I didn't change the behaviour of the other radios

jia200x, 2018-10-25:
ok, thanks @bergzand for the fix! Then, let's close this issue ;).

closed 2018-10-25 by jia200x