RIOT-OS/RIOT #9964: Initialization sequence is not in a common format

danpetry, 2018-09-19:
#### Description

Currently there is a lot of variation between the way the startup sequence is implemented between boards. More specifically, this relates to the cpu- and board-specific code, i.e. the code which is run between the reset vector and `kernel_init()`. This leads to the following problems:

1. Unit tests for sections of the startup sequence that are common across all boards cannot be implemented (see https://github.com/RIOT-OS/RIOT/issues/9342#issuecomment-422816773)
2. The startup sequence not being well defined could lead to issues with maintainability and potential bugs when porting new boards, which could have been avoided.

gebart, 2018-09-19:
Good arguments @danpetry. I agree that a common behavior across all of the boards would be useful.

kaspar030, 2018-09-19:
I suggest waiting for cross file arrays before tackling this. They might provide a nice way of ordering & specifying code to run at boot time.

closed 2019-09-10 by stale[bot]