RIOT-OS/RIOT #9965: BOARD_INSUFFICIENT_MEMORY alignment

MrKevinWeiss, 2018-09-19:
It is a hassle to add a board (especially that start with early characters) and keep alignment.  If we add a abc_board we should push every other board down.

```
BOARD_INSUFFICIENT_MEMORY := airfy-beacon arduino-duemilanove arduino-mega2560 \
                             arduino-uno b-l072z-lrwan1 blackpill bluepill calliope-mini \
                             cc2650-launchpad cc2650stk hifive1 maple-mini \
                             mega-xplained microbit msb-430 msb-430h \
                             nrf51dongle nrf6310 nucleo-f031k6 nucleo-f042k6 \
                             nucleo-f303k8 nucleo-l031k6 nucleo-f030r8 \
                             nucleo-f070rb nucleo-f072rb nucleo-f103rb \
                             nucleo-f302r8 nucleo-f334r8 nucleo-l053r8 \
                             nucleo-l073rz opencm904 spark-core \
                             stm32f0discovery telosb waspmote-pro \
                             weio wsn430-v1_3b wsn430-v1_4 yunjia-nrf51822 z1
```

It would be nice to have a script to do the following:
```
BOARD_INSUFFICIENT_MEMORY := airfy-beacon \
                             arduino-duemilanove \
                             arduino-mega2560 \
                             arduino-uno \
                             b-l072z-lrwan1 \
                             blackpill bluepill \
                             calliope-mini \
                             cc2650-launchpad \
                             cc2650stk \
                             hifive1 \
                             maple-mini \
                             mega-xplained \
                             microbit \
                             msb-430 \
                             msb-430h \
                             nrf51dongle \
                             nrf6310 \
                             nucleo-f031k6 \
                             nucleo-f042k6 \
                             nucleo-f303k8 \
                             nucleo-l031k6 \
                             nucleo-f030r8 \
                             nucleo-f070rb \
                             nucleo-f072rb \
                             nucleo-f103rb \
                             nucleo-f302r8 \
                             nucleo-f334r8 \
                             nucleo-l053r8 \
                             nucleo-l073rz \
                             opencm904 \
                             spark-core \
                             stm32f0discovery \
                             telosb \
                             waspmote-pro \
                             weio \
                             wsn430-v1_3b \
                             wsn430-v1_4 \
                             yunjia-nrf51822 \
                             z1
``` 
and maybe even add it to murdock with alphabetizing.  Or we can think about adding the "standard"/murdock build sizes (since this is just to make murdock happy) that way it can just be a comparison instead of every single board below 256kB being manually entered.

kaspar030, 2018-09-19:
#9081 would vastly mitigate this problem by being able to create groups based on memory size.

vincent-d, 2018-09-19:
I did this at some point, but it's far for perfect https://github.com/RIOT-OS/RIOT/pull/6818

I have no time to improve it though, so feel free to take it over or replace it ;)

miri64, 2019-09-10:
@MrKevinWeiss any chance you can look into this? Also maybe related https://github.com/RIOT-OS/RIOT/issues/11128.

MrKevinWeiss, 2019-09-10:
Yup.  I proposed a solution which involved a hacky script doing 90% of the work then manually taking care of the rest.  It was then implemented in a better way and fed into the CI with @x3ro script but never got finished.  I haven't noticed it in a while so I let it stale.

closed 2020-04-13 by stale[bot]