RIOT-OS/RIOT #9966: [RFC] pkg: PKG_VERSION with tags or with commit hash

bergzand, 2018-09-19:
in #9947 was some discussion whether we want to use a commit hash or also allow git tags for the `PKG_VERSION` variable.

I'd like to use this issue to start a discussion on which option we prefer here.

Arguments I've seen so far are:
 - Tags are more human readable
 - Tags can (but should not) be changed to a different commit.

After the discussion here dies down I'd like to call a vote.

kaspar030, 2018-09-19:
I'm all for commit hashes + tag as comment. It is desirable that a certain commit hash *always* results in the same code being checked out. That includes code that's pulled as dependency.

It is considered bad practice to change the commit a git tag points to, but imagine our faces if we find out... Note that we point to repositories which are out of our control. Just using hashes makes sure there won't be any surprises. Adding the intended tag as comment should serve as "readability bonus".

miri64, 2018-09-19:
I favor the `tag` as `PKG_VERSION` solution. If the external project force pushes their tags (or force pushes to public/stable branches in general) they aren't a serious project and shouldn't be supported by RIOT anyway. Yes, we could add the tag name as comment, but it wouldn't be the first time that we forget to update a comment after the change.

As a compromise or third option I would suggest using the tag's object hash (which is based on the content of the *tag*, i.e. includes the commit hash, but also the message, author, signiture and more) for to verify the version.

Same goes for HTTP-based packages (or if we ever support different transports for packages). Who guarantees us that the file(s) we download weren't changed on the project's server? The usual approach would be verify with a hash provided by the project, which most projects that want to be taken serious, do.

kaspar030, 2018-09-19:
> If the external project force pushes their tags (or force pushes to public/stable branches in general) they aren't a serious project and shouldn't be supported by RIOT anyway.

It doesn't matter if the external project is never going to do that *on purpose*. 

github account credentials get lost all the time. Remember last time you allowed some github application to access *all private and public repositories*? Well, personally I'm sensitive in that regard, but can we say that much about *everyone that has access to any repository we link by tag*?

Using commit hashes removes an entire class of security issues.

> As a compromise or third option I would suggest using the tag's object hash

While that might seem as if it is the best of both worlds, I've never seen a tag object hash and don't know how to retrieve it. Surely that's a hole in my git-foo that can be fixed, but commit hashes are *everywhere* and IMO, much more accessible.

gebart, 2018-09-20:
Definitely use hashes (commit or otherwise), I agree with the arguments of @kaspar030 above.
Also, see this thread for an example of a RIOT supported pkg which (used to) update its tags: https://github.com/ARMmbed/littlefs/issues/79

dylad, 2018-09-20:
I agree that tags are more readable but some external projects don't use tags (e.g EDBG). Since every project use hashes I think its a more reasonable choice.

OlegHahm, 2018-09-24:
As @miri64 wrote: for HTTP packages we would use a hash to verify that the upstream version hasn't changed. Git basically provides this hash as the commit hash automatically. Hence, I changed my mind and would follow @kaspar030's proposal and use the git commit hash. This also allows us to pin a particular package to a certain commit if the project doesn't provide tags (e.g., rolling releases) or introduces a critical fix early on in a consistent manner.

miri64, 2018-09-27:
While I start to agree with the hash argument side of the discussion, I'd like to point out that this still means a user needs to trust *us* that the hash actually points to the release noted in the comment. Just be cause the hash is pointing to lwIP v2.1 it doesn't necessarily have to. Sure, the user can check but this is

a) extra work and
b) still not guaranteeing to be the "right" commit because as you pointed out, the tag might have been tempered with in the package's repository.

miri64, 2018-09-29:
What I wanted to say with that is: An extra, distinct verification process within the pkg build pipeline apart from the "just trust the self-signing nature of git objects" method might be a more trust-worthy approach.

miri64, 2018-09-29:
(I am aware that this would put a strain on build-times, but there are solutions in the works I think (#10038?) to mitigate this.

closed 2019-09-10 by stale[bot]