RIOT-OS/RIOT #9971: No output in PyTerm after flash

npcode15, 2018-09-20:
Hello, I am trying to flash hello-world example. Flash is successful but the pyterm doesn't output anything. 

Please refer to the following image for reference.
[image]

I am not if this is a  RIOT issue.

**Setup**: Ubuntu 10 within a Windows 10 Host (using VMWare).
**Device**: nrf52dk
**arm-toolchain**: v7.3

Thanks.

dylad, 2018-09-20:
Did you tried to reset the board after flashing ?
Usually I had to reset this board to see some outputs in the beginning.
Another solution is to open an instance of pyterm in a second terminal which is always connected, then use your first terminal to flash, compile and do all the stuff you want.

dylad, 2018-09-20:
Are you using nrf52dk or nrf52840dk ? 
In fact are you using nrf52832 or nrf52840 chip ?
For nrf52832 you need to setup some register because the reset pin is a mere GPIO.

dylad, 2018-09-20:
Looks like you need to setup the pin to the reset function to enable it. Otherwise the pin acts as a GPIO

npcode15, 2018-09-20:
@dylad I will set it. Thanks.

npcode15, 2018-09-26:
@dylad I have a follow-up question.

Is it possible for me to execute RIOT shell commands in Pyterm.

If yes, how? I am not able to do so.

Thank you.

dylad, 2018-09-26:
Pyterm is intend to execute RIOT shell commands. So yes you can.
Typing "help" plus enter key should give the list of available commands.
Not all RIOT tests applications use the shell, some of them only print information without any interaction. So please check that your application is calling the shell init.
If this still don't work, please check your serial connection and your serial driver.

npcode15, 2018-09-26:
when I enter commands like ps, ifconfig. Pyterm does absolutely nothing. Please look at the following image for reference.
[image]

npcode15, 2018-09-26:
I was able to make it work after re-installing openocd (I don't know if it has any relevance or not).

So, it works but is a bit sketchy. I do not want to confuse anyone. Apparently, even after specifying PORT=/dev/ttyACM0 or PORT=/dev/ttyACM1, RIOT was flashing the same board. I was able to make it work by connecting second one only after the first flash suceeded. 

I just wanted to inform. I will close this one but I might have a question regarding this in the future. So, it possible that I might re-open this again. 

Thanks.

closed 2018-09-26 by npcode15

dylad, 2018-09-27:
Remember that openOCD don't use serial port to flash, so telling RIOT build system which port to use will not help openOCD. In this case you need to add `DEBUG_ADAPTER_ID=xxxxxxx` to tell which debugger you want to use, see `makefiles/tools/openocd-adapters` for details.

dylad, 2018-09-28:
see #10072 for resetting your board.