RIOT-OS/RIOT #9975: RFC: Introduce "Discussion: " label category

miri64, 2018-09-20:
During the maintainer assembly at the RIOT summit last week, we discussed the introduction of a `NEEDS CONSENSUS` label, to mark PRs and issues where the discussion is stalled or so stuck in opposing arguments to involve a neutral arbiter to resolve it.

I propose that we introduce a new category of labels all-together (which IMHO also "Requests for Comments" belongs to) that also includes `NEEDS CONSENSUS`, that more fine-grainly would help to steer discussions before it even goes into the `NEEDS CONSESUS` category. Some other example that go into the direction that @danpetry proposed during that discussion would be a `Discussion: Final Comments` (to give people the time to give their final arguments) or `Discussion: Finished` (to mark that the decision making process was finished and now only implementation details are commented on).

miri64, 2018-09-25:
A recording of the discussion can be found on [YouTube](https://www.youtube.com/watch?v=IBPHqAwLdI4&t=8949s) btw.

jia200x, 2018-09-26:
+1 for trying it :)

miri64, 2018-10-02:
No objections, only likes, in the last 2 weeks, so I guess I just implement it ;-).

miri64, 2018-10-02:
Done

closed 2018-10-02 by miri64