RIOT-OS/RIOT #9977: feature `periph_gpio_irq` not fully implemented/made use of

haukepetersen, 2018-09-20:
#### Description
#9845 made it possible to build GPIO drivers without support for external interrupts, very useful indeed. But sadly, except for the `cpu/msp430fxyz`, this feature is not implemented (e.g. missing ifdefs) for any other existing GPIO driver. So in other words, using the `PERIPH_GPIO_IRQ` module (or not) has no effect on most boards... 

IMHO when touching a core interface like the `periph/gpio` driver, it should be made sure that the changes are actually implemented for existing implementations (as in this case). Or at least there should be some tracking on the conversion status or similar.

Please simply close this issue if I overlooked any ongoing PR or already open issue on this topic...

gebart, 2018-09-20:
You are right in that this is only implemented for the msp430. When creating that change I only wanted to get the GPIO driver size down in order to make #8711 mergable.
Let's make this issue into the tracker for implementing this feature check on other platforms.

gebart, 2018-09-20:
Updated the top post

cladmi, 2018-09-20:
Some are already in https://github.com/RIOT-OS/RIOT/pull/8711 PR I will start splitting them and do the other ones.

cladmi, 2018-09-20:
I think this belongs here https://github.com/RIOT-OS/RIOT/pull/9978

haukepetersen, 2018-09-20:
I went through the drivers and added the fixes: #9981

kaspar030, 2018-09-20:
> So in other words, using the `PERIPH_GPIO_IRQ` module (or not) has no effect on most boards...

Actually, using the module has the effect of also using periph_gpio.

> But sadly, except for the cpu/msp430fxyz, this feature is not implemented (e.g. missing ifdefs) for any other existing GPIO driver. 

As written [here](https://github.com/RIOT-OS/RIOT/pull/9978#issuecomment-423245763), where's the problem with that? Implementations that are "updated" only conditionally compile the irq code. The others work as before. We can now incrementally implement this optimization.

haukepetersen, 2018-09-26:
> We can now incrementally implement this optimization.

done :-)

haukepetersen, 2018-10-16:
everything done -> issue solved.

cladmi, 2018-10-17:
This one has not been done yet I think ? maybe I am missing something

* [ ] Adapt gpio_init_int name as suggested by #9845 (comment) with an alias for the old name

While reviewing https://github.com/RIOT-OS/RIOT/pull/10135 I noticed that two boards `samr30-xpro` @aabadie and `lobaro-lorabox` @leandrolanzieri  do not set `periph_gpio_irq`. But they actually compile when adding `FEATURES_PROVIDED += periph_gpio_irq`. I did not run the `tests/periph_gpio` as I do not have the boards though.
As they are new boards they may have missed the update.
If that is the case, they should be updated to provide the new feature.

leandrolanzieri, 2018-11-22:
> This one has not been done yet I think ? maybe I am missing something
>
> * [ ]  Adapt gpio_init_int name as suggested by #9845 (comment) with an alias for the old name
>
> While reviewing #10135 I [...]

This has been addressed for the `lobaro-lorabox` now in #10449.

cladmi, 2018-11-22:
@leandrolanzieri Cool
@aabadie can you check if it should also be done for `samr30-xpro` ?

aabadie, 2018-12-05:
> can you check if it should also be done for samr30-xpro ?

see #10434

PeterKietzmann, 2018-12-18:
#9845, #10434 and #10553 were all merged so I'm closing this one again

closed 2018-12-18 by PeterKietzmann