RIOT-OS/RIOT #9982: Nanocoap Example |  nrf52dk device IP

npcode15, 2018-09-20:
Hello, is it possible for me to connect the nrf52dk board with the internet?

I have 2 of them and I want to make them talk with each other using UDP for which I would need their IP addresses. Before I can find get the IP,  I would need to connect them to the network.

I am following this tutorial from nordic, right? 
https://devzone.nordicsemi.com/b/blog/posts/setting-up-ipv6-over-ble-using-nrf52-series-and-ri

The example mentioned in the tutorial was microcoap-example but I think RIOT replaced it with nanocoap-example. Anyway, this produces a Link-Local IPv6 (please refer to the screenshot below).
[image]

I am trying to ping the IP and the output says invalid argument. I would like to know what exactly is?

Note: I was not able to complete the tutorial. I got stuck at step, 'sudo hcitool lescan'. So, I wanted to know what is the significance of the IP displayed as output.

aabadie, 2018-09-21:
If I understand correctly, your are facing issues with the Nordic tutorial. Note that Nordic tutorial is not using RIOT but their own software libraries.

I suggest that you ask them directly.

aabadie, 2018-09-21:
Sorry, I replied a bit fast. It is actually about RIOT...

Maybe @dylad could answer that question. I remember he already played a bit with IPv6 and nrf52850dk.

npcode15, 2018-09-21:
Okay, I did post a question on Nordic's forum as well.

dylad, 2018-09-21:
@npcode15 In the current state, I think GNRC stack will not work with nrf52840dk board.
You will need #9473 to get merged before.
To explain things more clearly : 
The GNRC network stack of RIOT depends on the Nordic Softdevice blob for nRF devices. Currently RIOT supports Softdevice from Nordic SDK 14.2 (I think) but support for nRF52840 only start at Nordic SDK 15.
@miri64 Am I all wrong here ?

You can try to grab #9473 and redo your tests from there.

miri64, 2018-09-28:
> @miri64 Am I all wrong here ?

I never used the soft device with GNRC myself, but when porting `gnrc_netif` I saw that it is supposed to support 6Lo. So I would guess it *should* work. Maybe @kaspar030 can provide a better answer (sorry for hot-potatoing this question, but I really don't know the answer)?

npcode15, 2018-10-05:
@dylad @miri64 

"@npcode15 In the current state, I think GNRC stack will not work with nrf52840dk board." - 
would changing the networking stack to OpenThread make any difference?

PeterKietzmann, 2019-02-21:
> would changing the networking stack to OpenThread make any difference?

AFAIK OpenThread is currently limited to 802.15.4 radios by atmel. However, we now have some other BLE features, among others, we provide BLE support by the nimBLE stack as external software package. Please have a look at these applications under examples/[nimle_gatt](https://github.com/RIOT-OS/RIOT/tree/master/examples/nimble_gatt), [sklad_eddystone](https://github.com/RIOT-OS/RIOT/tree/master/examples/skald_eddystone), [sklad_ibeacon](https://github.com/RIOT-OS/RIOT/tree/master/examples/skald_ibeacon)

closed 2019-09-25 by stale[bot]