RIOT-OS/RIOT #9987: Ping-Pong between 2 nodes using UDP

npcode15, 2018-09-21:
Hello, I would like to repeat the ipc_pingpong example (_https://github.com/RIOT-OS/RIOT/tree/master/examples/ipc_pingpong_) but for 2 nodes and not for 2 processes.
Basically, communicate between 2 nodes via UDP. 

I was able to communicate one way but in order to reply, I would like to know that I received a message. 
I would like write some code in main.c, something like

if packet_received
{
 reply back
}

I did find something, please have a look at line 163 in file gntc_pktdump.c 
_https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktdump/gnrc_pktdump.c_
is possible to copy functions such as gnrc_udp_init and _eventloop and reuse them for my purpose?
I am not sure as this file was included as a header file.

Screenshot of the referred code.
[image]
Also, could you please explain, why is there a need to check that the **type** of '**msg**' could be '**GNRC_NETAPI_MSG_TYPE_SND**'. I mean, I am assuming this class is all about dumping the packets received to display the output.

Please let me know if the question was unclear. Thanks.

aabadie, 2018-09-21:
> Basically, communicate between 2 nodes via UDP.

I would suggest to start a UDP listener on each nodes and when a packet is received it sends back to the other an UDP message.

Use the sock API for this (interestingly there's no example for this), see the [online documentation of the API](http://doc.riot-os.org/group__net__sock.html). There's a [test application](https://github.com/RIOT-OS/RIOT/blob/master/tests/gnrc_sock_udp/main.c) which shows some code.

You can play with `sock_udp_recv` and `sock_udp_send` in a while loop in main.

npcode15, 2018-09-21:
Listeners are fine, but I was basically looking for a method which would inform me as soon as the msg was received.

I had an idea, I am still working on it. Apart from sock_udp methods, do you think _eventloop(msg) would work?

I would create a thread. Event loop will register to udp and the port. I will initiate a thread with eventloop. Let me know your views thanks.

PeterKietzmann, 2018-09-25:
@npcode15 I'm not sure I understand your question correctly. Which layer/component/thread should be informed about which message? Do you mean when sending the IPC message `GNRC_NETAPI_MSG_TYPE_SND` from your application to the UDP thread for example, that the UDP thread acknowledges IPC message arrival?

npcode15, 2018-09-25:
@PeterKietzmann Issue #9990 answered this question.

Basically, I referring to all the hex data and message dump.

Thank you.

closed 2018-09-25 by npcode15