RIOT-OS/RIOT #9990: Communication without utilizing the GNRC stack

npcode15, 2018-09-21:
Hello, is it possible to communicate between two nodes without actually utilizing the network stack?

**Example**: I would like to send raw data (just the payload no header i.e. no packets)? Maybe using a MAC layer.

Thanks

aabadie, 2018-09-21:
I assume you want to send data using 802.15.4 radio.

Yes it is possible, see the shell command `txtsnd` in `examples/default`. The packets are sent at MAC Layer.
The code is the underlying functions is [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/shell/commands/sc_gnrc_netif.c#L1194).
You also need to add at least this line `USEMODULE += gnrc_pktdump` to your application Makefile to dump received packets to stdout. See `examples/default` [Makefile](https://github.com/RIOT-OS/RIOT/blob/master/examples/default/Makefile).
Unfortunately, RIOT is missing straightforward examples for this kind of things, so you have to dig a little bit in the code.

npcode15, 2018-09-21:
@aabadie thanks for letting me know that. I didn't that txtsnd was sent at MAC Layer. I will look into the code. 

I got another question, how do I know that a node has received a message so that it can either reply or acknowledge??

This is question kinda duplication of #9987

aabadie, 2018-09-21:
> how do I know that a node has received a message so that it can either reply or acknowledge?

I think you can have a look at [this task](https://github.com/RIOT-OS/Tutorials/tree/master/task-07) in [the official RIOT tutorials](https://github.com/RIOT-OS/Tutorials).
If you didn't already, I recommend that you play them at least once.

npcode15, 2018-09-21:
I did have a look but I didn't complete the Tutorial. Thanks a lot. I will close this one.

npcode15, 2018-09-25:
Need to reopen this one. I have a follow-up question.

@aabadie, according to the tutorial, I need to subscribe to a particular type of packet in order to receive the packets. So, I am wondering what should be the type for 'txtsnd'. The method also requires a port number, assuming I can use NULL, I still a protocol type for 'txtsnd'. Thanks.

aabadie, 2018-09-25:
I'm not an expert of the Tutorials but maybe @miri64 or @cgundogan could give you more accurate answers

npcode15, 2018-09-25:
@aabadie sure. 

I would like to add another thing. I am trying to copy functions like **__dump_**, **__dump_snip_** to enable complete logging. My Aim is to get the message sent with the packet. I could not find any other way but to copy those functions to my main.c file. This way I will have the message in Hex form.

**The problem:** 
[image]

 Thanks.

PeterKietzmann, 2018-09-25:
@npcode15 be aware that the referenced task incorporates the GNRC UDP/IPv6 stack. You may want to have a look at [task-05](https://github.com/RIOT-OS/Tutorials/tree/master/task-05) as well.

For the hex dump, did you include `USEMODULE += od` in you r applications Makefile?

PeterKietzmann, 2018-09-25:
Does this answer your question? Can we close the issue again?

npcode15, 2018-09-25:
Yes, it worked.

please close after another question. Do not wait for my response.

I would like to know if there is a better way to get the message sent along with the packet. Currently, I would convert from hex to ASCII to retrieve the message.

Thanks a lot for help.

PeterKietzmann, 2018-09-25:
@npcode15 `txtsnd` is simply a command that executes a shell handler. Behind the scene this handler  builds a raw packet (of type `GNRC_NETTYPE_UNDEF`) and sends it over the network interface. See the actual implementation: [_gnrc_netif_send](https://github.com/RIOT-OS/RIOT/blob/master/sys/shell/commands/sc_gnrc_netif.c#L1194).
Now on the receiver side, just for demonstration, we have the `pktdump` module that you already referenced. This module starts a thread that subscribes to IPC messages which relate to all compiled GNRC protocol implementations. Each received `snip` calls a `print` function to simply dump the relevant information behind it (see [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktdump/gnrc_pktdump.c#L52)). I think this would be your method of choice to "subscribe" to a certain message type.

`txtsnd` does not require a port number. Usage is as follows:
```
usage: txtsnd <if> [<L2 addr>|bcast] <data>
```
Which port would that be anyway? There is no further protocols such as UDP on top.

npcode15, 2018-09-25:
No, I am talking about the usage of txtsnd. I am talking from the perspective of receiving messages sent using txtsnd.

If am sending a message using UDP, I would subscribe for UDP and port 8888. I will receive them.
But, to recieve txtsnd, I would susbcribe to which protocol. Please look at the following snippet.
[image]

But, I do appreciate your explanation of how txtsnd works under the hood.

PeterKietzmann, 2018-09-25:
But `txtsnd` does **not** send UDP packets.

npcode15, 2018-09-25:
I know.

I just I want to send messages using txtsnd and capture them using the customised code I wrote (tutorial task 7.2).

To capture the packets, I need to tell the code to register for a protocol. For example, if I want UDP, I can register for UDP. If I want TCP, I can register for TCP. Similarly, if want to register for txtsnd packets how do I register?

I gave reference to this code in the image in the last comment.

PeterKietzmann, 2018-09-25:
`GNRC_NETTYPE_UNDEF`

PeterKietzmann, 2018-09-25:
If this does not answer your question I suggest you share the complete code. Please point to a branch then instead of screenshots

npcode15, 2018-09-25:
I tried '_GNRC_NETTYPE_UNDEF_' already. It results in a compilation error.

PeterKietzmann, 2018-09-25:
1. In future please provide code in a repository or paste it here (formatted properly)
2. The error says you make integer from pointer. I guess this is due to the `NULL` in `GNRC_NETREG_ENTRY_INIT_PID(NULL, sched_active_pid);`. Try to use [`GNRC_NETREG_DEMUX_CTX_ALL`](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/gnrc/netreg.h#L80) instead.
3. I am still confused what you are trying to do. The issue says you want plain packets and this is why we started talking about `txtsnd`. On the other hand your application looks as if you're targeting UDP.

npcode15, 2018-09-26:
@PeterKietzmann 

1. Sure, I will keep that in mind. I did not have the code in the repository that is why I could not provide repository reference.

3. I am not targetting UDP packets. I want to be able to ping-pong messages between 2 nodes using txtsnd. I know that possible with txtsnd using shell. But I also want control over the sending and receiving of packets rather than packets getting dumped by a system. I need this to customize the packet dump to include a timestamp, not provided by the regular packet dump. Hence, I need this kind of code.

PeterKietzmann, 2018-09-26:
But did you manage to get this running now? I'd like to close this issue then and continue with the other issues subsequently...

npcode15, 2018-09-26:
No, not yet, but I can close this one. I got the answer.

I am having error, udp.h: no such file or directory.

PeterKietzmann, 2018-09-26:
The error you're referring to, does it relate to #10040? Otherwise you can always open an other issue which describes the actual bug in the title and issue description.

npcode15, 2018-09-26:
No. But, yes, I will create another issue if I am not able to resolve it.

npcode15, 2018-09-30:
> 1. In future please provide code in a repository or paste it here (formatted properly)
> 2. The error says you make integer from pointer. I guess this is due to the `NULL` in [...]

@PeterKietzmann I am still not able to receive packets sent via _'txtsnd'_. I used the following code to receive the packet sent via txtsnd. But, my code becomes irrelevant as the program use gnrc_pktdump.c to receive the packets instead of using my version of the same code.

```c
void *rcv_pkt(void *arg) {
 (void) arg;

 int num = 4;
 msg_t msg_queue[num];
 msg_init_queue(msg_queue, num);

 /* initialize and register pktdump to print received packets */
 gnrc_netreg_entry_t server = GNRC_NETREG_ENTRY_INIT_PID(GNRC_NETREG_DEMUX_CTX_ALL,
                                gnrc_pktdump_pid);
 gnrc_netreg_register(GNRC_NETTYPE_UNDEF, &server);

 msg_t msg;
 while (1) {
    msg_receive(&msg);
    gnrc_pktsnip_t *pkt;

    pkt = (gnrc_pktsnip_t *)msg.content.ptr;
    // od_hex_dump(pkt->data, pkt->size, OD_WIDTH_DEFAULT);

    printf("Sending Reply\n");
    send(mac_address, "Hello Node M3-100");

    gnrc_pktbuf_release(pkt);
 }

 return NULL;
}
```
Basically, when I use, GNRC_NETTYPE_UNDEF a new thread is created and _eventloop thread ()is called and packets are dumped (https://github.com/RIOT-OS/RIOT/blob/b78f78d3a172957f7f57233fd1b984733e8c39a0/sys/net/gnrc/pktdump/gnrc_pktdump.c#L189). 

This was not the case with UDP, in UDP I was providing a pid and that is why new thread was not started by pktdump and the code I wrote was receiving the packet. 

I tried providing a pid (the pid number created when initialized a thread for rcv_pkt()), and using that instead of using GNRC_NETTYPE_UNDEF but in that case I receive no packets at all.

Basically, this is the primary issue but if full code is needed I will upload the complete code to the repository. Hope I was clear with the issue.