RIOT-OS/RIOT #10015: Strategic Mediatek MTK6261D support?

Citrullin, 2018-09-22:
Hey guys,

I remember that we talked about commercial usage of RIOT at the summit. So, I somehow got into the topic of smartwatches. I found the X9 plus with a NRF52832 and bought two. It shouldn't be the problem to get RIOT OS up and running on this device. This watch costs around 20 USD. But I found another MCU which is used in even cheaper watches: the Mediatek MTK6261D. There are a lot of watches on Aliexpress/eBay/Amazon for around 5 - 15 USD. There is also a little hack-community out there. I bought a bunch of them. So, how about getting RIOT working on these watches? They have GSM/GPRS, Bluetooth 3.0. The CPU has around 260 Mhz. These watches run Nucleus OS. So I think an open source microkernel alternative could be interesting for the Chinese companies behind these watches and for RIOT a really good strategy to grow the market share. These are just my thoughts. Since I am not really a CPU expert I probably need some help to get RIOT running on this CPU. 

Some interesting links around this topic:
- https://www.cnx-software.com/2015/11/24/no-1-d3-smartwatch-review/
- https://www.dr-lex.be/hardware/china_phone_flashing.html
- https://forum.xda-developers.com/smartwatch/other-smartwatches/readback-extractor-mtk6260-firmware-t3289272

dylad, 2018-09-23:
This could be interesting, looks like this an ARM926EJ-S processor. Problem is (with Mediatek) the lack of good documentation about low level peripheral for implement drivers like UART or SPI.
Do you know if there is a way to physically access the JTAG pins ? Using weird/voodoo russian/chinese software is ... tricky.

Citrullin, 2018-09-23:
@dylad CNX-software mentiones this reverse engineering tool: https://github.com/xobs/fernly Maybe that can be helpful.
Yes, that's one reason why Mediatek is soo cheap. They just take the ARM reference designs and create a MCU out of it, without any optimization and they documentation is usally shit.
I've looked in some videos where they use these weird vodoo chinese tools. The configuration there says MT6261D instead of MTK6261D. So seems like MTK is somehow the light version of MTK. So maybe this datasheet helps then: https://datasheet4u.com/datasheet-pdf/MEDIATEK/MT6261D/pdf.php?id=1266527 This project around this MCU is really interesting and might be helpful: https://www.kosagi.com/w/index.php?title=Fernvale_Main_Page

I watched some teardown videos. Looks like they don't expose the JTAG via smd pins. But this micro usb port exposes some kind of COM port. Looks like the Fernvale people wrote already a open source flash tool. 

So, if there are people interested in it and located in Berlin, I am happy to sponsor a bunch of these U8 smart watches.

closed 2019-03-04 by Citrullin