RIOT-OS/RIOT #10016: Tutorial task-07 Error | ASSERTION FAILED

npcode15, 2018-09-22:
Hello, I am trying to solve the task-07 (7.2 to be precise) from the tutorials.

I wrote a new function: rcv_pkt() and called it at the end of the main function.
Please refer to the following screenshot (of function code) for a better understanding.
[image]

**Error:** Compilation succeeds with 'make all' but when I use the command 'make all term' I get the following error.
[image]

I tried using make clean, touch to solve the probelm of skewed clock but didn't succeed so far.

I appreciate any help and thanks in advance.

aabadie, 2018-09-25:
@npcode15, your question seems related to the Tutorial, maybe you should post [there](https://github.com/RIOT-OS/Tutorials/issues/new) ?

To give you an advice on your problem, you get a `FAILED ASSERTION` indeed, but if you look just above, the file and line where this happened is also given: `sys/net/gnrc/netreg/gnrc_netreg.c:49`. Could it be that you didn't initialize a message queue ?

npcode15, 2018-09-25:
@aabadie, Sorry I should have posted there. But, I will open an issue and close it referencing this one. So, that the issue will searchable.

Regarding the solution, yes it worked. I missed the initialization somehow. But, I am getting another error (please look at the following image).
[image]

I tried to initiate the UDP server from the main.c file. This was fine before. I have no idea what broke.
I even cloned a fresh copy but didn't help. Does anything spring to your mind?

**Current gnrc_minimal_code**
[image]

npcode15, 2018-09-25:
Nope, that was not the issue but I am getting a segmentation dump when I commented out line 126.
But, it turns out I do not need udp.c file because I subscribe to UDP packets. So, I was able to fix the issue for my purpose. But, it still fails.

[image]
**One possibility:** Because I am already subscribed to receive UDP and I am again trying to do so by starting UDP server. There is a conflict, hence the error. Just a guess.

Also, Should I close the issue?

closed 2018-09-25 by npcode15