RIOT-OS/RIOT #10018: nRF51822: Cannot erase protected sector

Citrullin, 2018-09-23:
Hi,
some time ago I bought a smartwatch. So I just look it up: Luckily it has a nRF51822 MCU. So theoretical I can run RIOT on it. So I soldered the exposed SWDIO and SWCLK pins. So far, so good. Looks fine. Can reset the device and it restarts. When I try to flash the device, I get the following error:
Has anyone an idea how erase it to override the software with RIOT?

```
info : Listening on port 44749 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* nrf51.cpu          hla_target little nrf51.cpu          halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
auto erase enabled
Info : nRF51822-QFAC(build code: A1) 256kB Flash
Error: Cannot erase protected sector at 0x0
Error: failed erasing sectors 0 to 18

Warn : Flash driver of nrf51.flash does not support free_driver_priv()
Warn : Flash driver of nrf51.uicr does not support free_driver_priv()
```

dylad, 2018-09-24:
Did you tried to erase the flash with nrfjprog software from Nordic ? It's based on Jlink software.
You can then erase your flash with `nrfjprog -f nrf51 --eraseall` or `nrfjprog -f nrf51 --recover`

Citrullin, 2018-09-24:
@dylad Thanks. I will try that. :)

Citrullin, 2018-09-25:
@dylad So, tried it and needed to resolder the wires. While I did that, I somehow broke the watch. So I close this for now :D

closed 2018-09-25 by Citrullin