RIOT-OS/RIOT #10030: RFC: Categorize labels

miri64, 2018-09-25:
We talked off-line a lot about categorizing labels in the past, so I finally went through all of the existing labels (or proposed ones) and tried to find categories for all of them. Here are the ones I came up with (including a proposed color scheme for the labels).
The goal is to make the labeling of issues and PRs more clearer and thus make it easier to find a certain issue of interest, either to review it, contribute to it, or discard it faster.

- [image] **Architecture (Arch):** Labels PRs and issues related to a certain (set of) architecture(s)
- [image] **Area:** Marks an area of expertise within the code base that an issue or PR relates to
- [image] **Category:** Category of an issue or PR
- [image] **CI:** Labels to control or give hints to the CI system
- [image] **Community:** Labels PRs or issues to solicit community involvement
- [image] **Discussion:** Marks status of an ongoing discussion within an issue or PR
- [image] **Host OS (OS):** Marks issues or PRs related to a specific (non-Linux) host OS
- [image] **Integration Process (Process):** Marks a state within the integration Process of a PR
- [image] **Reason:** Marks a reason to close a PR or issue
- [image] **Reviewed:** Marks the current state of the review of a PR
- [image] **PR status (Status):** Overall status of a PR
- [image] **Task force (TF):** Marks all issues and PRs related to the work of an active task force

I also propose to remove some labels, since they are either unclear, outdated or pretty similar to a different label.

Below you can find a table how I would categorize the existing labels. In some cases the renamed label has a quite different name, which I reasoned with in the footnotes below. Mind you, none of this is set in stone and all of it (the categories themselves, the colors, the ordering, etc) is up for discussion. But please don't add new labels that aren't on the table yet. These should go into separate discussions.

<table>
  <thead>
    <tr><th>Old Label</th><th>Renamed Label / Reasoning</th></tr>
  </thead>
  <tbody>
    
    <tr><th colspan="2">Architecture (Arch)</th></tr>
    <tr><td>arm</td><td>Arch: ARM</td></tr>
    <tr><td>avr</td><td>Arch: AVR</td></tr>
    <tr><td>esp</td><td>Arch: ESP</td></tr>
    <tr><td>mips</td><td>Arch: MIPS</td></tr>
    <tr><td>msp</td><td>Arch: MSP</td></tr>
    <tr><td>native</td><td>Arch: native</td></tr>
    <tr><td>risc-v</td><td>Arch: RISC-V</td></tr>
    
    <tr><th colspan="2">Area</th></tr>
    <tr><td>arduino</td><td>Area: arduino API (1)</td></tr>
    <tr><td>board</td><td>Area: boards</td></tr>
    <tr><td>BLE</td><td>Area: BLE</td></tr>
    <tr><td>C++</td><td>Area: C++</td></tr>
    <tr><td>CI</td><td>Area: CI</td></tr>
    <tr><td>CoAP</td><td>Area: CoAP</td></tr>
    <tr><td>core</td><td>Area: core</td></tr>
    <tr><td>cpu</td><td>Area: cpu</td></tr>
    <tr><td>crypto</td><td>Area: crypto</td></tr>
    <tr><td>documentation</td><td>Area: doc</td></tr>
    <tr><td>driver</td><td>Area: drivers</td></tr>
    <tr><td>fs</td><td>Area: fs</td></tr>
    <tr><td>LoRa</td><td>Area: LoRA</td></tr>
    <tr><td>make</td><td>Area: build system (2)</td></tr>
    <tr><td>network</td><td>Area: network</td></tr>
    <tr><td>ota</td><td>Area: OTA</td></tr>
    <tr><td>pkg</td><td>Area: pkg</td></tr>
    <tr><td>pm</td><td>Area: pm</td></tr>
    <tr><td>posix</td><td>Area: POSIX</td></tr>
    <tr><td>saul</td><td>Area: SAUL</td></tr>
    <tr><td>security</td><td>Area: security</td></tr>
    <tr><td>tests</td><td>Area: tests</td></tr>
    <tr><td>timer</td><td>Area: timers</td></tr>
    <tr><td>toolchain</td><td>Area: toolchains</td></tr>
    <tr><td>tools</td><td>Area: tools</td></tr>
    
    <tr><th colspan="2">Category</th></tr>
    <tr><td>bug</td><td>Category: bug</td></tr>
    <tr><td>cleanup</td><td>Category: cleanup</td></tr>
    <tr><td>enhancement</td><td>Category: enhancement</td></tr>
    <tr><td>Feature Request</td><td>Category: feature request</td></tr>
    <tr><td>new feature</td><td>Category: new feature</td></tr>
    <tr><td>question</td><td>Category: question</td></tr>
    
    <tr><th colspan="2">CI</th></tr>
    <tr><td>CI: ready for build</td><td>CI: ready for build (3)</td></tr>
    <tr><td>CI: run tests</td><td>CI: run tests (3)</td></tr>
    <tr><td>NEEDS SQUASHING</td><td>CI: NEEDS SQUASHING</td></tr>
    
    <tr><th colspan="2">Community</th></tr>
    <tr><td>good first issue</td><td>Community: good first issue</td></tr>
    <tr><td>Hack'n'ACK candidate</td><td>Community: Hack'n'ACK</td></tr>
    <tr><td>help wanted</td><td>Community: help wanted</td></tr>
    
    <tr><th colspan="2">Discussion</th></tr>
    <tr><td>contested</td><td>Discussion: contested</td></tr>
    <tr><td>Request for Comments</td><td>Discussion: RFC</td></tr>
    <tr><td>-</td><td>Discussion: final comments (4) (see #9975)</td></tr>
    <tr><td>-</td><td>Discussion: finished (4) (see #9975)</td></tr>
    <tr><td>-</td><td>Discussion: NEEDS CONSENSUS (4) (see #9975)</td></tr>
    
    <tr><th colspan="2">Host OS (OS)</th></tr>
    <tr><td>osx</td><td>OS: Mac OS X</td></tr>
    <tr><td>windows</td><td>OS: Windows</td></tr>
    
    <tr><th colspan="2">Integration Process (Process)</th></tr>
    <tr><td>API CHANGE</td><td>Process: API change</td></tr>
    <tr><td>backport</td><td>Process: release backport</td></tr>
    <tr><td>backporting-needed</td><td>Process: NEEDS BACKPORT</td></tr>
    <tr><td>major</td><td>Process: NEEDS >1 ACK (1)</td></tr>
    
    <tr><th colspan="2">Reason</th></tr>
    <tr><td>duplicate</td><td>Reason: duplicate</td></tr>
    <tr><td>invalid</td><td>Reason: invalid</td></tr>
    <tr><td>memo</td><td>Reason: archived (1)</td></tr>
    <tr><td>quality defect</td><td>Reason: quality defect</td></tr>
    <tr><td>wontfix</td><td>Reason: won't fix</td></tr>
    
    <tr><th colspan="2">Reviewed</th></tr>
    <tr><td>-</td><td>Reviewed: 1-fundamentals (4) (see #9973)</td></tr>
    <tr><td>-</td><td>Reviewed: 2-code-design (4) (see #9973)</td></tr>
    <tr><td>-</td><td>Reviewed: 3-testing (4) (see #9973)</td></tr>
    <tr><td>-</td><td>Reviewed: 4-coding-conventions (4) (see #9973)</td></tr>
    <tr><td>-</td><td>Reviewed: 5-documentation (4) (see #9973)</td></tr>
    
    <tr><th colspan="2">PR status (Status)</th></tr>
    <tr><td>Waiting For Other PR</td><td>Status: waiting for other PR</td></tr>
    <tr><td>WIP</td><td>Status: WIP</td></tr>
    
    <tr><th colspan="2">Task force (TF)</th></tr>
    <tr><td>I2C REWORK</td><td>TF: I2C</td></tr>
    <tr><td>Task Force: Configuration</td><td>TF: Config</td></tr>
    
    <tr><th colspan="2"><del>Remove</del></th></tr>
    <tr><td>minor</td><td>meaning unclear</td></tr>
    <tr><td>GNRC</td><td>GNRC TF not a thing anymore</td></tr>
    <tr><td>PR-award-nominee</td><td>PR awards aren't a thing anymore</td></tr>
    <tr><td>x86</td><td>We currently don't have any x86 support</td></tr>
    <tr><td>Newbie-Task-Candidate</td><td>Duplicate of "good first issue" which seems to be <a href=https://help.github.com/articles/helping-new-contributors-find-your-project-with-labels/>preferred</a></td></tr>
  </tbody>
</table>

(1) rename for clarification
(2) rename for generalization
(3) no rename required
(4) new label

x3ro, 2018-09-25:
Yay 🎉 I also think an automation solution like [git-labelmaker](https://github.com/himynameisdave/git-labelmaker) would still make sense, to make this a pull-request based process. Compare #6267

miri64, 2018-09-25:
Oh sorry @x3ro, I totally forgot about #6267, otherwise I would have mentioned it in OP.

miri64, 2018-09-25:
An automated solution would help a lot, but I think in the long run, we would also need some kind of triageing session for labeling new (or re-labeling old) PRs and issues (a bot can't decide e.g. if something is a "good first issue"

x3ro, 2018-09-25:
@miri64 Oh I didn't mean for labelling itself, but for managing the labels. That is, make sure that a new label can only be created via PR :)

miri64, 2018-09-25:
Ah. Actually, I wouldn't restrict the labeling system that much. But this is a separate discussion IMHO. Would you mind open an issue about that, if you think this is the right direction?

jia200x, 2018-09-26:
> PR awards aren't a thing anymore

:(

Thanks so much for this!

I only have one additional comment on the Category entry.
Looking at the [Angular Commit Message Format](https://gist.github.com/stephenparish/9941e89d80e2bc58a153#allowed-type) types, I think it could also make sense to add a "Category: tests" and move the "Area: docs" to "Category: docs".

 We could extract useful information for the CI (show warning if there's a new feature with no test, not test everything if there's only a change in documentation, etc).

As a further step, we could enforce the usage of labels. See e.g [Danger CI](https://github.com/danger/danger)

miri64, 2018-09-26:
@jia200x Please note my remark in OP:

> […] But please don't add new labels that aren't on the table yet. These should go into separate discussions.

This issue is about categorizing existing labels. How we extend/manage/introduce new categories in the future goes beyond that discussion IMHO. I feel like most of your comments go into this direction (and the CI suggestions go even beyond the new / renamed labels discussion).

> I only have one additional comment on the Category entry.
Looking at the Angular Commit Message Format types, I think it could also make sense to add a "Category: tests" and move the "Area: docs" to "Category: docs".

Mhh... we also could move "Area: tests" to "Category: tests" instead of adding a new label. But it somehow did not fit to what I imagine to be a category of issues or PR (describing what the issue/PR *is*). A PR can fix a "bug" or be a "new feature" and an issue can describe a "bug" or request a new feature. But neither issues nor PRs can *be* a test or a documentation. They can only provide or improve one (Category: enhancement), make them more readable (Category: cleanup) or fix them if they are broken (Category: bug).

Come to think of it: maybe "Category: new feature" and "Category: feature request" are redundant as one is for issues and one is for PRs... We don't have e.g. a "Category: bug" for issues and "Category: bugfix" as well.

> Also, there are several issues tracking on-going efforts. How do we tag them?

I think a label: `Category: tracking` (or under our current "scheme": `tracking`) would make sense, but again: introducing new labels is not part of this discussion (either introduce it now or later).

miri64, 2018-09-26:
I made the title more specific, to hopefully steer the direction of this discussion ;-).

waehlisch, 2018-09-26:
Instead of ```category``` I would say ```type```. This should also include ```type:documentation```.

Should we also add```priority:{high|low|medium}```?

```Reason``` and ```Status``` are very similar. I would merge ```Reason``` into ```status```.

```Area``` is very generic and has overlap with ```Architecture```. Maybe someone finds a better name for this label.

miri64, 2018-09-26:
> Instead of ```category``` I would say ```type```. This should also include ```type:documentation```.

:+1:

> Should we also add ```priority:{high|low|medium}```?

I think it's a good idea (the rust label system which inspired this change definitely has it) and would have been one of my first steps after we done the change proposed here. But this would be an addition (and a whole new label category for that matter).

> ```Reason``` and ```Status``` are very similar. I would merge ```Reason``` into ```status```.

I don't see the similarity. `Reason` is to state the reason for closing a PR, while `Status` reflects the current working state of a PR. Maybe the doc on the two categories needs some clarification.

> ```Area``` is very generic and has overlap with ```Architecture```. Maybe someone finds a better name for this label.

Mh... Rust uses also [Area](https://github.com/rust-lang/rust/labels?utf8=%E2%9C%93&q=area) and sure it is quite generic, but I guess "Point of interest within the codebase" would be quite overkill ;-). Regarding overlap to `Architecture`: I agree that it might could be merged with `Area: cpu` (remove `Area: cpu` instead and use `Area: <arch>`) but for me `Area` directly maps to a file / directory / group of files or directories within the codebase. `Architecture` does not necessarily (a bug in GNRC could only be [relevant for MSP430 e.g.](https://github.com/RIOT-OS/RIOT/pull/10043) so I would label it with `Arch: MSP`, `Area: network` but not with `Area: cpu`, while a bug within the code of the MSP430 port I *would* label with `Area: cpu`, `Arch: MSP`).

miri64, 2018-09-26:
> This should also include `type:documentation`.

Is this for issues that themselves are documenting something (e.g. progress of an effort, though there would be overlap with the tracking label proposed by @jia200x)? Otherwise, I still don't see how documentation can be a type/category for an issue or a PR.

kaspar030, 2018-09-26:
Feels like a huge improvement!

Can we keep minor and major as is, maybe as "Impact: minor" and "Impact: major"?

I like to tag trivial typo changes as "minor" to give reviewers a hint that they can just blindly merge :wink:.

Also I tend to use "major" as a tool to nudge reviewers to not merge too quickly. IMO the proposed "Process: NEEDS >1 ACK (1)" carries only part of the information.

miri64, 2018-09-26:
I've updated the proposed changes that made sense to me above.

miri64, 2018-09-26:
> […] `Status` reflects the current working state of a PR. Maybe the doc on the two categories needs some clarification.

Maybe `Info` or `Condition` is a better category name? Mostly, labels of that category inform maintainers in what working condition the PR is.

waehlisch, 2018-09-26:
> > Reason and Status are very similar. I would merge Reason into status.
>
> I don't see the similarity. Reason is to state the reason for closing a PR, while Status reflects the current > working state of a PR. Maybe the doc on the two categories needs some clarification.
>
The reason for closing a PR reflects the working state. You really don't need two types. When a PR is closed, either the state or the reason are alive.

kaspar030, 2018-09-27:
Do we need "Priority: medium"? It feels like that would have to be set for every not low or high priority PR/issue (e.g., most). Or is it "no priority < Priority: low < Priority: medium < Priority: high"?

Would using "Urgency:" instead of "Priority:" clarify this, possibly with "Urgency: low" renamed to "Urgency: little"?

waehlisch, 2018-09-27:
> Or is it "no priority < Priority: low < Priority: medium < Priority: high"

That's my understanding.

waehlisch, 2018-09-27:
Term "priority" is also borrowed from Rust.

kaspar030, 2018-09-27:
> That's my understanding.

ok!

miri64, 2018-09-27:
> Do we need "Priority: medium"? It feels like that would have to be set for every not low or high priority PR/issue (e.g., most). Or is it "no priority < Priority: low < Priority: medium < Priority: high"?

Grml... that's exactly the reason why I didn't want to include new labels and categories into the discussion. I remove Priority for now. Maybe we can discuss prioritization via label in a separate issue?

miri64, 2018-09-27:
> "quality defect" is a type, not a reason, right?

As far as I remember it was introduced with the intention to tell that a PR was broken by design, so a closing reason. I couldn't find the discussion that introduced it, but I remember that it was the result of a lengthy discussion. If it *is* used primarily as a type (which I saw for some issues it is), then I see some overlap to "Type: bug" and would just remove it.

miri64, 2018-09-27:
> The reason for closing a PR reflects the working state. You really don't need two types. When a PR is closed, either the state or the reason are alive.

Ok, that convinced me.

miri64, 2018-09-27:
Merged "Reason" and "PR Status" into "State"

miri64, 2018-09-27:
> As far as I remember it was introduced with the intention to tell that a PR was broken by design, so a closing reason. I couldn't find the discussion that introduced it, but I remember that it was [...]

Maybe put it in the Discussion category? But then again, a quality defect isn't something we need to discuss since it is most of the time if not always objectively defect.

danpetry, 2018-09-27:
@miri64 thanks very much for this. I think it will help us greatly to organize work between ourselves.

Here are my comments:
- I agree with limiting the scope of this discussion to renaming, rather than adding. Perhaps we should open another PR which captures follow-on work exposed during this discussion. If there are no objections, I'll open one.
- Regarding the capitalized PRs, I don't think there's currently a clear distinction between those and the non-capitalized ones. If capitalization is to indicate blocking, we could also have e.g. `State: NEEDS OTHER PR` or `Community: NEEDS HELP`. I think this would be a useful way to go.
- Re @kaspar030's [comment](https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424836384) about `major` and `minor`: I think if `minor` == light review and `major` == thorough review, we should be explicit, because people won't necessarily realize this intention.
- `State: quality defect`, to me, is unclear, maybe suggesting an alert about a defect in quality of our code. Could we clarify it as, perhaps, `State: PR quality defect`?
- I would remove `Discussion: contested`. I think the other ones cover all states of a discussion - this seems to overlap with `NEEDS CONSENSUS`.
- Just curious why the `Remove` category is struck through in the table?

miri64, 2018-09-27:
> - I agree with limiting the scope of this discussion to renaming, rather than adding. Perhaps we should open another PR which captures follow-on work exposed during this discussion. If there are no objections, I'll open one.

Rather open an issue ;-). PRs against non-code would be difficult.

> - Regarding the capitalized ~~PRs~~ [labels?], I don't think there's currently a clear distinction between those and the non-capitalized ones. If capitalization is to indicate blocking, we could also have e.g. `State: NEEDS OTHER PR` or `Community: NEEDS HELP`. I think this would be a useful way to go.

I have to admit I didn't really follow any consistent pattern other than "it especially important so it should be capitalized". But yes why not capitalize "Waiting for other PR" then? Also: are abbreviations then also "especially important". I'm leaning towards removing this specialized capitalization. What do you think?

> - Re @kaspar030's [comment](https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424836384) about `major` and `minor`: I think if `minor` == light review and `major` == thorough review, we should be explicit, because people won't necessarily realize this intention.

:+1: maybe "Impact: light" / "Impact: significant"? Or your proposal, but something more review-related as a category name (e.g. "**Review requirement (Req)**")?

> - `State: quality defect`, to me, is unclear, maybe suggesting an alert about a defect in quality of our code. Could we clarify it as, perhaps, `State: PR quality defect`?

As explained before: If I remember correctly "quality defect" was introduced to mark PRs that are themselves of poor quality (not a quality defect in our codebase, which in my opinion is just a bug). In terms of our maintainer guidelines (which were introduced later) this would match a complete fail in the points of sections 1 and 2.

> - I would remove `Discussion: contested`. I think the other ones cover all states of a discussion - this seems to overlap with `NEEDS CONSENSUS`.

IMHO 'contested' is distinct from 'needs consensus'. 'contested' marks either a discussion where the proposed discussion item has already a consensus (the proposal is contested and we don't adopt it) or where enough voices offered clearly formulated doubts about the proposal that it would receive that label during the discussion face (which would lead it to reach the first alternative at some point). 'needs consensus' on the other hand would mark a discussion that is stuck, either by enough contending arguments on both sides or by being forgotten about (see #9975).

> - Just curious why the `Remove` category is struck through in the table?

Artefact from before I opened the issue. Fixed it.

danpetry, 2018-09-27:
> Rather open an issue ;-).

I've opened the issue: https://github.com/RIOT-OS/RIOT/issues/10057

>I'm leaning towards removing this specialized capitalization.

Personally I like the idea of indicating whether something is blocked. Perhaps we could decapitalise all labels (apart from abbreviations) and think about adding `Status: blocked`? (I put this in the issue above.)

> +1 maybe "Impact: light" / "Impact: significant"? Or your proposal, but something more review-related as a category name (e.g. "Review requirement (Req)")?

I think the second one would make sense.

> As explained before: If I remember correctly "quality defect" was introduced to mark PRs that are themselves of poor quality (not a quality defect in our codebase, which in my opinion is just a [...]

Sure - I think the labels should be self-explanatory as far as possible, though. The proposed change was in order to address this.

> 'contested' marks either a discussion where the proposed discussion item has already a consensus (the proposal is contested and we don't adopt it) or where enough voices offered clearly formulated [...]

To me, "contested" means that the proposal is being challenged, not that consensus has been achieved against the proposal.

miri64, 2018-09-27:
> To me, "contested" means that the proposal is being challenged, not that consensus has been achieved against the proposal.

That was what I was trying to say in the second half of the sentence ^^". However, only because a proposal is challenged that doesn't mean that a consensus now needs to immediately be called (which "needs consensus" is for).

miri64, 2018-09-27:
> Sure - I think the labels should be self-explanatory as far as possible, though. The proposed change was in order to address this.

As discussed offline: as its meaning seems to be rather blurry even among maintainers and depending on interpretation (when used to mark a fix for a quality defect) has a lot of overlap with `bug` or `enhancement`, I lean towards removing the label altogether.

miri64, 2018-09-27:
(the interpretation of that label I followed could also be covered by `State: won't fix`

danpetry, 2018-09-28:
From https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-425086582
>>To me, "contested" means that the proposal is being challenged, not that consensus has been achieved against the proposal.

>That was what I was trying to say in the second half of the sentence ^^". However, only because a proposal is challenged this doesn't mean that a consensus now needs to immediately be called (which "needs consensus" is for).

Aren't most proposals challenged? So, does this label mean that the question of whether the PR should even be merged or not is being discussed?

>> Sure - I think the labels should be self-explanatory as far as possible, though. The proposed change was in order to address this.

> As discussed offline: as its meaning seems to be rather blurry even among maintainers and depending on interpretation (when used to mark a fix for a quality defect) has a lot of overlap with bug or enhancement, I lean towards removing the label altogether.

Sounds like a solution - I think `invalid` would also cover it. More detailed reason should be obvious from the comments - and this would also probably be kinder on a contributor than having their PR labelled as `quality defect`.

>> I think if minor == light review and major == thorough review, we should be explicit, because people won't necessarily realize this intention.

> Nonono, these should just describe the impact, e.g., a change to a driver has less impact than a change to core/msg.c. That reviews can be more lax or need to be more thorough is just one of the aspects.

Ok, but we should consider that whatever is left to the interpretation of other maintainers will need to be explained to people, and will be missed occasionally. Especially as the number of maintainers grows, and we eventually have such a large number of people helping out around the world that we can't possibly explain all these things to everyone in person :) I think they should ideally be as self-documenting as possible.
Maybe we could add another category rather than changing these ones?

miri64, 2018-09-28:
> Ok, but we should consider that whatever is left to the interpretation of other maintainers will need to be explained to people, and will be missed occasionally. Especially as the number of [...]
For this reason, I think they should all ideally be as self-documenting as possible, which is why I'm being quite nitpicky here.
Maybe we could add another category rather than changing these ones?

BTW, when renaming, I'm planning to add a description to every label, so at least there we can clarify stuff like this. I agree that label names should be self-documenting, but if @kaspar030 likes to keep the impact category as is (with his given his explanation), I don't mind either.

I think making the labeling system as fine-grained as possible might also not be very useful. Contrarily, having something like those `Impact:` labels that carry a number of clearly defined implications rather than a bunch of labels that would always be set together anyways helps with that.

miri64, 2018-09-28:
> Aren't most proposals challenged? So, does this label mean that the question of whether the PR should even be merged or not is being discussed? Could be `merging contested`?

Again, as far as I have seen this label used in the past, it marks a discussion where the discussion already leans towards a decline of the proposal. This doesn't necessarily mean the consensus is reached yet; sometimes it is even set by a maintainer who dislikes the idea so much they use it to mark that fact. In the latter case this might of course mean that some moderation is required (triggering `Discussion: needs consensus`). `merging contested` would restrict this label for usage with PRs only, however a proposed RFC in an issue might also be contested e.g., so I'd rather stick to `Discussion: contested`.

miri64, 2018-09-28:
Updated TOP post with regards to all-caps labels.

miri64, 2018-09-28:
Also moved "quality defect" to the `Removed` "category".

danpetry, 2018-09-28:
Ok, I'm happy my comments have been addressed. Let's go with `contested` then, for your reasons above, and `minor`/`major`. I've added `review requirement` and `status: blocked` labels to the follow-up issue for discussion.

miri64, 2018-09-28:
Ok, I have the feeling we already converging to a consensus here. I think open points (or in rather points where there was no follow up to the objections by the contributors who brought it up) are:

* Should `Arch:` be merged into `Area:` (see https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424740959)? See objection [here](https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424748525).
* Should there be a `Type: test` and `Type: doc` label (see https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424691995 and also https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424740959). See objection [here](https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-424755781).

Everything else can be followed up on in #10057.

miri64, 2018-09-29:
Note to self or others who want to help implement this feature: Renaming `NEEDS SQUASHING` and `Waiting For Other PR` need to be adapted here:

https://github.com/RIOT-OS/RIOT/blob/b78f78d3a172957f7f57233fd1b984733e8c39a0/dist/tools/pr_check/pr_check.sh#L42-L50

As a solution that not introduces cyclic dependency (a change to `pr_check` change needs to be merged before the rename, pr_check requires the labels to exist, so rename needs to happer before `pr_check` change is merged) I propose to just introduce `CI: needs squashing` and `State: waiting for other PR` as new labels (rather than renaming `NEEDS SQUASHING` and `Waiting For Other PR`) and remove the old ones as soon as the change to `pr_check` was merged.

miri64, 2018-09-30:
FYI, I took a few minutes of my Sunday and did the most boring (and least intrusive) part of this: Documenting the meaning of a label. So now every label also comes with a short blurb about it (according to what we discussed here).

danpetry, 2018-10-01:
> Should Arch: be merged into Area: (see #10030 (comment))? See objection here.

> Should there be a Type: test and Type: doc label (see #10030 (comment) and also #10030 (comment)). See objection here.

Personally, I would find it useful to have the labels currently falling under `Arch` and `Area` as separate. `Area` is a tough one to name accurately, though.
Perhaps a nomenclature that could offer a clearer distinction could be `Platform` instead of `Arch`, and `Feature` or `Function` instead of `Area`?
Following this, `test` and `doc` could be moved into `Type`, because they're not really a `Function`, they're a contribution `Type` like `bugfix` or `new feature` for example.

miri64, 2018-10-01:
> Perhaps a nomenclature that could offer a clearer distinction could be Platform instead of Arch, and Feature or Function instead of Area?
If this were done, then test and doc could be moved into Type, because they're not really a Function, they're a contribution Type like bugfix or new feature for example.

Neither `Feature` nor `Function` fits for me. As I said: I see it more as a description of the area within the code base. To set a label `Feature: toolchains` feels weird, because toolchains are IMHO not a feature but a given and moving it to `Type: toolchains` doesn't work for me either. `Function` doesn't fit at all for me, but maybe I'm thinking too much of the CS/mathmatics concept of functions here.

danpetry, 2018-10-01:
Btw, I'm happy for these last two points to be moved to https://github.com/RIOT-OS/RIOT/issues/10057 so that we can get this closed off.

miri64, 2018-10-02:
Now that we have the "final comments" label I set it. Please let your give your final comments especially with regards to https://github.com/RIOT-OS/RIOT/issues/10030#issuecomment-425438788. I'd like to have this discussion be finalized by the end of the week.

smlng, 2018-10-02:
I would also like to point out that GitHub offers other features than labelling to organise PRs and issues, i.e. Milestones and Projects. The number of labels is already quite high, so we might think about using these feature in favour of a label, e.g. Task Forces are meant to be non-permanent so it might be better to setup a Project and collect related PRs and Issues there instead of labelling these.

miri64, 2018-10-02:
> I would also like to point out that GitHub offers other features than labelling to organise PRs and issues, i.e. Milestones and Projects. The number of labels is already quite high, so we might [...]

While I agree that we should use also other GitHub features (more?), labels have proven as quite a stable mechanism to get a rough overview over the global state of issues and PRs (while projects and milestones e.g. are very target-specific). I agree that the TF workflow better fits with the project system instead of introducing labels that, once a TF is "done" the label is removed / forgotten / misused for other stuff. Our number of labels is IMHO still quite manageable; at the moment we are at 73 and that already includes the newly introduced labels from #9973 and #9975 (and some labels that will be removed if this discussion finds an agreement). Compare that to the [180 labels of the rust project](https://github.com/rust-lang/rust/labels) which was used as a template for this discussion (and which uses its labeling system quite successfully).

jia200x, 2018-10-05:
>  I agree that the TF workflow better fits with the project system instead of introducing labels that, once a TF is "done" the label is removed / forgotten / misused for other stuff.

I can create a Project for the Configurations PR and then we can remove the `TF: Configurations` label.

miri64, 2018-10-05:
Ok, but I guess we keep the I2C-label for now, since lots are done, lots are TODO, lots ar WIP.

miri64, 2018-10-05:
I guess we can conclude this discussion now. I'll go ahead and rename the labels as proposed. Apart from the deleted one we still can rename them differently if we don't like the current version ;-).

miri64, 2018-10-05:
And done.

closed 2018-10-05 by miri64