RIOT-OS/RIOT #10031: pkg/openthread: fails to build on macOS

smlng, 2018-09-25:
#### Description

Cannot build RIOT with OpenThread on macOS due to some 32/64 Bit issue.

#### Steps to reproduce the issue

Run `make -C tests/openthread` on macOS.

#### Expected results

Test application compiles.

#### Actual results

Build error right after downloading/cloning OpenThread:

```
 make -C tests/openthread/
Warning: no PORT set!
Compile OpenThread for FTD device
$OPENTHREAD_ARGS is [--enable-cli-app=ftd --enable-application-coap]
make[1]: Nothing to be done for `Makefile.include'.
Building application "openthread" for "samr21-xpro" with MCU "samd21".

Compile OpenThread for FTD device
$OPENTHREAD_ARGS is [--enable-cli-app=ftd --enable-application-coap]
rm -Rf /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread
mkdir -p /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread
/Volumes/devel/github/smlng/RIOT/dist/tools/git/git-cache clone "https://github.com/openthread/openthread.git" "fbfd76a990b81f007957e1bd774e51bce742e53e" "/Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread"
Cloning into '/Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread'...
remote: Enumerating objects: 33, done.
remote: Counting objects: 100% (33/33), done.
remote: Compressing objects: 100% (33/33), done.
remote: Total 37733 (delta 6), reused 4 (delta 0), pack-reused 37700
Receiving objects: 100% (37733/37733), 51.88 MiB | 7.69 MiB/s, done.
Resolving deltas: 100% (28098/28098), done.
Checking out files: 100% (3489/3489), done.
Checking out files: 100% (3421/3421), done.
HEAD is now at fbfd76a9 Apply `#include` code style. (#1723)
touch /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/.git-downloaded
"/Library/Developer/CommandLineTools/usr/bin/make" -C /Volumes/devel/github/smlng/RIOT/pkg/openthread
Compile OpenThread for FTD device
$OPENTHREAD_ARGS is [--enable-cli-app=ftd --enable-application-coap]
cd /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread && PREFIX="/" ./bootstrap
/Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/third_party/nlbuild-autotools/repo/scripts/bootstrap: line 180: /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/third_party/nlbuild-autotools/repo/tools/host/i386-apple-darwin/bin/libtoolize: No such file or directory
chmod: /tmp/tmp.bootstrapmu0tcg/libtoolize: No such file or directory
/Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/third_party/nlbuild-autotools/repo/scripts/bootstrap: line 239: /tmp/tmp.bootstrapmu0tcg/libtoolize: No such file or directory
cd /Volumes/devel/github/smlng/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread && CPP="arm-none-eabi-gcc -E" CC="arm-none-eabi-gcc" CXX="arm-none-eabi-g++"\
		OBJC="" OBJCXX="" AR="arm-none-eabi-ar" RANLIB="arm-none-eabi-ranlib" NM="arm-none-eabi-nm" \
		STRIP="" \
		CPPFLAGS="-fdata-sections -ffunction-sections -Os -Wno-implicit-fallthrough -mcpu=cortex-m0plus -mlittle-endian -mthumb -mfloat-abi=soft -march=armv6s-m " \
		CFLAGS="-fdata-sections -ffunction-sections -Os -Wno-implicit-fallthrough -mcpu=cortex-m0plus -mlittle-endian -mthumb -mfloat-abi=soft -march=armv6s-m " \
		CXXFLAGS="-fdata-sections -ffunction-sections -Os -Wno-implicit-fallthrough -mcpu=cortex-m0plus -mlittle-endian -mthumb -mfloat-abi=soft -march=armv6s-m -fno-exceptions -fno-rtti " \
		LDFLAGS="-fdata-sections -ffunction-sections -Os -Wno-implicit-fallthrough -mcpu=cortex-m0plus -mlittle-endian -mthumb -mfloat-abi=soft -march=armv6s-m -nostartfiles -specs=nano.specs \
		-specs=nosys.specs -Wl,--gc-sections -Wl,-Map=map.map " \
		./configure --disable-docs --host=arm-none-eabi --target=arm-none-eabi \
		--prefix=/ --enable-default-logging --enable-cli-app=ftd --enable-application-coap
/bin/sh: ./configure: No such file or directory
make[1]: *** [all] Error 127
make: *** [pkg-build-openthread] Error 2
```

#### Versions

output from `./dist/tools/ci/print_toolchain_versions.sh`:

```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.13.6
                 Kernel: Darwin 17.7.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Apple LLVM version 10.0.0 (clang-1000.10.44.2)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 8.1.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: Apple LLVM version 10.0.0 (clang-1000.10.44.2)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.1
               cppcheck: Cppcheck 1.84
                doxygen: 1.8.14
                 flake8: 3.5.0 (mccabe: 0.6.1, pycodestyle: 2.3.1, pyflakes: 1.6.0) CPython 2.7.15 on Darwin
                    git: git version 2.18.0
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15
                python2: Python 2.7.15
                python3: Python 3.7.0
             coccinelle: spatch version 1.0.6 compiled with OCaml version 4.05.0
```

Lotterleben, 2018-10-07:
I've prodded around in this issue a little and I've found that even if I manage to get rid of the error message @smeiling posted (for example by hardcoding `host=x86_64-apple-darwin` in `RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/third_party/nlbuild-autotools/repo/scripts/bootstrap`, line 135, which obviously isn't a satisfying solution but just to see where it takes us), I get a new error:

`cd /Users/lotte/codestuff/riot/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread && PREFIX="/" ./bootstrap
autom4te: need GNU m4 1.4 or later: /Users/lotte/codestuff/riot/RIOT/tests/openthread/bin/pkg/samr21-xpro/openthread/third_party/nlbuild-autotools/repo/tools/host/x86_64-apple-darwin/bin/m4`

This error is consistent with openthread, i.e. if I clone https://github.com/openthread/openthread/ and check out the openthread revision used in the pkg Makefile (`fbfd76a990b81f007957e1bd774e51bce742e53e`), I get the same result when running `./bootstrap`. m4 v1.4 is added in commit `9905c3fee138a7aa756ca544252186cad80decd3`.

Setting version to `9905c3fee138a7aa756ca544252186cad80decd3` in the Makefile resolves both @smeiling s error as well as the `autom4te: need GNU m4 1.4 or later: ` issue, but introduces new errors in `pkg/openthread/contrib/platform_functions_wrapper.c`, since the return values of `otThreadGetDeviceRole()` appear to have changed in https://github.com/openthread/openthread/commit/427ef53900f4e16798877dc377f007bfb2776d78 .

So my questions are:
- does it even make sense to resolve this particular issue if we're going to run into the m4 issue in the next step (or should the `bootstrap` script be patched to use a local installation of m4 instead?)
- is there a reason `fbfd76a990b81f007957e1bd774e51bce742e53e` is used (I'm guessing API changes like the ones I stumbled across)?

jia200x, 2018-10-08:
hi @Lotterleben 

thanks for the pointers!

> does it even make sense to resolve this particular issue if we're going to run into the m4 issue in the next step (or should the bootstrap script be patched to use a local installation of m4 instead?)

There's currently this PR (#9336) that updates OpenThread to the release version. It should be merged really soon. Could you check if this one solves the issue?

> is there a reason fbfd76a990b81f007957e1bd774e51bce742e53e is used (I'm guessing API changes like the ones I stumbled across)?

The reason is because there was no OpenThread release back then. So, we took the latest commit at that point.
Now with #9336 it should be pointing to a proper commit.

Lotterleben, 2018-10-08:
Hi @jia200x,

> hi @Lotterleben
>
> thanks for the pointers!
>
>
> There's currently this PR (#9336) that updates OpenThread to the release version. It should be merged really soon. Could you check if this one solves the [...]

Yes! `examples/openthread` builds nicely. So that's two flies with one PR :)

> 
> > is there a reason fbfd76a990b81f007957e1bd774e51bce742e53e is used (I'm guessing API changes like the ones I stumbled across)?
> 
> The reason is because there was no OpenThread release back then. So, we took the latest commit at that point.
> Now with #9336 it should be pointing to a proper commit.

Ahhhh, good to know. Thank you for the explanation!

jia200x, 2018-10-10:
@Lotterleben good to know! Thanks.

I will close this issue as soon as #9336 gets merged