RIOT-OS/RIOT #10040: Error | Segmentation Fault

npcode15, 2018-09-26:
Hello, I am trying to repeatedly send-receive packets using UDP.

I am getting the following error, as soon as the first iteration gets completed and I try to send the next node.
[image]

**NOTE**: send method belongs to udp.c which I did not provide. There is no issue with the code, it works perfectly with loop if I comment out the _rcv_pkt()_ method.

**NOTE**: there seems to be no issues with the other node. It works differently, it has a receive-send combo.

I hope I was able to make the problem clear.

**Code**
----------
#include <stdio.h>
#include "msg.h"
#include "net/ipv6/addr.h"
#include "net/gnrc.h"
#include "net/gnrc/netif.h"
#include "net/gnrc/netreg.h"
#include "net/gnrc/pktbuf.h"
#include "sched.h"
#include <time.h>
//#include "udp.h"
#include "od.h"
#include "udp.c"
#include <inttypes.h>
#include "thread.h"
#include "shell.h"

static void send(char *addr_str, char *port_str, char *data, unsigned int num, unsigned int delay);
void get_interface_and_address(void);
void rcv_pkt(void);

int main(void)
{
    int cnt = 0;

    while (cnt<2){
    printf("UDP -> Send %d\n", cnt);
    printf("Timestamp I: %" PRIu32 "\n", xtimer_usec_from_ticks(xtimer_now()));
    send("fe80::648c:2eff:fe47:f74e", "8888", "Hello Node M3-101", 1, 2000000); **//Program Fails Here**
    printf("Timestamp II: %" PRIu32 "\n", xtimer_usec_from_ticks(xtimer_now()));

    get_interface_and_address();
    rcv_pkt();
    cnt = cnt + 1;
   }
    printf("Exiting");
    return 0;
}

void get_interface_and_address(void){
   /* get interfaces and print their addresses */
    gnrc_netif_t *netif = NULL;
    while ((netif = gnrc_netif_iter(netif))) {
        ipv6_addr_t ipv6_addrs[GNRC_NETIF_IPV6_ADDRS_NUMOF];
        int res = gnrc_netapi_get(netif->pid, NETOPT_IPV6_ADDR, 0, ipv6_addrs,
                                  sizeof(ipv6_addrs));

        if (res < 0) {
            continue;
        }
        for (unsigned i = 0; i < (unsigned)(res / sizeof(ipv6_addr_t)); i++) {
            char ipv6_addr[IPV6_ADDR_MAX_STR_LEN];

            ipv6_addr_to_str(ipv6_addr, &ipv6_addrs[i], IPV6_ADDR_MAX_STR_LEN);
            printf("My address is %s\n", ipv6_addr);
        }
    }
}

void rcv_pkt(void) {
 unsigned int count = 0;

 int num = 4;
 msg_t msg_queue[num];
 msg_init_queue(msg_queue, num);

 gnrc_netreg_entry_t server = GNRC_NETREG_ENTRY_INIT_PID(8888, sched_active_pid);
 gnrc_netreg_register(GNRC_NETTYPE_UDP, &server);

 msg_t msg;
 while (1) {
    gnrc_pktsnip_t *pkt;
    msg_receive(&msg);

    pkt = (gnrc_pktsnip_t *)msg.content.ptr;
    time_t t;
    time(&t);
    printf("Received UDP Packet No. %u at %s \n", ++count, ctime(&t));
    printf("NETTYPE_UNDEF (%i)\n", pkt->type);
    //od_hex_dump(pkt->data, pkt->size, OD_WIDTH_DEFAULT);
    //udp_hdr_print(pkt->data);
    gnrc_pktbuf_release(pkt);
    break;
 }
}

Thank you.

PeterKietzmann, 2018-09-26:
**Please** push your code to a branch. In this way it is extra work in order to review or reproduce your issues.

PeterKietzmann, 2018-09-26:
Furthermore,  how does the seg fault message continue? Instead for screenshots you could also copy-paste the text.

npcode15, 2018-09-26:
**1. Git Repo -** https://github.com/npcode15/RIOTOS_PingPong_Exp.git
Please follow the Readme for replicating the issue.

**2.** That is all that happens. It exits the native terminal to Linux one. 
**Error:**\
home/vagrant/Tutorials/exp/snd_udp_pkt/bin/native/client_udp.elf tap0
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.01-devel-1173-gbec3f-vagrant-HEAD)
UDP -> Send 0
Timestamp I: 3153
Success: sent "Hello Node M3-101" (17 byte(s)) to [fe80::648c:2eff:fe47:f74e]:8888
 at Tue Sep 25 18:41:29 2018
 &  4084Timestamp II: 7134
My address is fe80::9464:71ff:fe69:f1ab
Received UDP Packet No. 1 at Tue Sep 25 18:41:34 2018

NETTYPE_UNDEF (0)
UDP -> Send 1
Timestamp I: 5011745
/home/vagrant/Tutorials/exp/snd_udp_pkt/../../RIOT/Makefile.include:400: recipe for target 'term' failed
make: *** [term] Segmentation fault (core dumped)
vagrant@vagrant:~/Tutorials/exp/snd_udp_pkt$

**3.** send is implemented in udp.c
From my opinion, I don't think there is a problem with send. send works fine in the loop if I comment _rcv_pkt()_ method. I think the problem is somehow related to _break statement_ in _rcv_pkt()_ method.
But, that is just a guess.

PeterKietzmann, 2018-09-27:
You are referring to a README, which one do you mean? I only found the copied ones from our tutorials. I'm gonna give some comments to your code in your repository. However, please have a look at the [UDP sock API](https://riot-os.org/api/group__net__sock__udp.html#ga3cb61a4ee66c9c235e4f22860658698c). I think it does exactly what you want and is a sober solution.
One other thing: Which RIOT version do you work on?

npcode15, 2018-09-27:
Actually, I created a Readme for the Repo and the folders had the copied one. I had to force push and I think that destroyed the Readme. Sorry about that. I have created another readme if it matters. 

RIOT Version: RIOT-2017.10

I have 2 installations of RIOTS.
1. RIOT via vagrant (for Tutorial). I use this for native development. (RIOT-2017.10)
2. RIOT in virtual Linux. I use this for developing/compiling BOARD specific code. (RIOT-2018.07)

The code in the git repo I shared used RIOT-2017.10 for testing/development.

**Note:** After I updated gcc-arm compiler toolchain to version 7 (in the 2nd variation), compilation fails for native development. That is why I use the first installation.

PeterKietzmann, 2018-09-27:
@npcode15 it is hard to follow your process and to figure your problems. I propose the following procedure:
- decide for one development environment
- use RIOTs latest 2018.07 release
- write two slim applications based on [UDP sock API](https://riot-os.org/api/group__net__sock__udp.html#ga3cb61a4ee66c9c235e4f22860658698c)
- explore your applications by sniffing on the tap bride, when your're on native
- then, report in detail about problems: Build errors, runtime errors, ...
(- provide your code again, if actually needed)

npcode15, 2018-09-27:
@PeterKietzmann I believe it is simple enough. I mean, exactly similar to task-06 from Tutorials. Maybe I was not clear but one simply needs to create a tap interface and compile two different codes by specifying PORTs.

1. How can I have one development environment? NATIVE doesn't compile with arm-gcc toolchain version 7.

2. I will look at _UDP sock API_. Primary motive behind writing my own code was to gain control over the kind of packets I am receiving (I did not want '_pktdump_' to do it for me) and the data I am displaying in the output (I want the output to have timestamps).

3. What do you mean by sniffing tapbridge? are you suggesting it as a replacement for something?

PeterKietzmann, 2018-09-28:
To 1.: Cool!
To 2.: native doesn't use the arm complier. It uses the standard gcc. Your error indicates that some dependencies are missing,  probably *multilib*. Just google "sys/cdefs.h: No such file or directory" and install the missing dependencies.
To 3.: How about something like 
`ipv6_addr_from_str((ipv6_addr_t *)&remote.addr.ipv6, "<addr string>")`
4. Sock is a user API for networking which is independent from the underlying implementation. In your case it utilizes GNRC anyway **but** does everything that you tried to implement manually in *udp.c* for you.
5. You may want to read [this](https://en.wikipedia.org/wiki/TUN/TAP). To summarize, a tapbridge is a channel over which two native instances communicate with each other. You can use tools such as Wireshark to visualize the traffic/packets that are sent over the virtual channel

npcode15, 2018-09-29:
@PeterKietzmann thanks a lot for the information, I will read the article (about TAP/TUN).

And I am yet to check if the IP address solution you suggested works (for UDP_sock_API). So, I am closing this one for now. I will either reopen or create another issue to resolve UDP_sock_api issues.

Thanks again.

closed 2018-09-29 by npcode15