RIOT-OS/RIOT #10045: Add other radios support for OpenThread pkg

jia200x, 2018-09-26:
The current OpenThread package only supports `at86rf2xx` radios, although the port is done on top of `netdev`.

This is why:
1. The radio setup is in the [openthread_bootstrap function](https://github.com/RIOT-OS/RIOT/blob/master/pkg/openthread/contrib/openthread.c#L72). 
2. This is the only radio that [implements the NETDEV_EVENT_TX_COMPLETE_DATA_PENDING](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_netdev.c#L664) for handling the frame pending bit in the IEEE802.15.4 ACK

I open this issue to keep track of the OT support for other radios. For adding a new radio to OpenThread, do 1. (we should probably move it to auto_init though!) and add the netdev event NETDEV_EVENT_TX_COMPLETE_DATA_PENDING. It should be simple since it should only involve reading a register :)

TODO:
- [x] at86rf2xx
- [ ] cc2538
- [ ] cc2420
- [ ] mrf24j40 (@bergzand) 
- [ ] kw2x

(Am I missing one?)

gebart, 2018-09-26:
(added kw41z to the top post)

jia200x, 2018-10-05:
> Sure, only a register read, how hard could this be on a bunch of different radios, the manufacturer probably thought about exposing this information in an easily accessible way :)

Oh ja :)

> @jia200x To ask the dumb question, do we also have to implement something to configure ACK transmission with frame pending bit?

oh, I totally forgot that. Good catch.
Yes, it has to be implemented (it won't have any effects now since pending bit handling is not supported in routers, but it will be added soon).

jia200x, 2020-09-09:
This issues still makes sense, but probably the way to do it is different now considering #14950 or #14371 is there. See e.g #14979

jia200x, 2020-10-16:
I propose to close this one because:
- Since #14979 is there, the goal moves to port missing radios to the Radio HAL instead of specifically adding OpenThread support to them. See [Radio HAL tracker](https://github.com/RIOT-OS/RIOT/issues/14792)
- When `netdev_ieee802154_submac` is deprecated, we will need to implement a new layer on top of the Radio HAL or SubMAC anyway.

closed 2020-10-16 by jia200x