RIOT-OS/RIOT #10074: CC2650-Launchpad flashing procedure

rienafairefr, 2018-09-28:
#### Context
Deployment of CC2650-Launchpad on [FIT IoT-LAB](www.iot-lab.info)

#### Description
CC2650-Launchpad `flash` target uses Uniflash, which is closed source and only for x86 (maybe even only x64) Linux, not usable on ARM

#### Steps to reproduce the issue
Trying to erase the flash of a CC2650-Launchpad board using `make flash` (using dslite.sh of Uniflash under the hood) on an ARM based Single Board Computer

#### Expected results
Being able to erase the board flash and flash using RIOT `make flash`, even on systems where Uniflash is not supported

I also posted a related issue on [Contiki-NG](https://github.com/contiki-ng/contiki-ng/issues/679), their version which apparently uses cc2538-bsl is not working for my use cases.

PeterKietzmann, 2019-07-23:
@aabadie, @cladmi ideas?

cladmi, 2019-07-23:
Solution, try to run qemu-x86 on IoT-LAB to execute the binary. If it is registered in the `/proc/sys/fs/binfmt_misc/` it could be transparent.

One thing I found when updating `uniflash`, is that TI have their own fork of openocd that supposedly can flash the board. I never tried it.

When using a `bsl` you may need a special bootloader, and that `reset` would not work when `term` is running.

cladmi, 2019-07-24:
I just noticed that there is an `openocd` file https://github.com/RIOT-OS/RIOT/blob/master/boards/cc2650-launchpad/dist/openocd.cfg

It was added on the board introduction PR https://github.com/RIOT-OS/RIOT/pull/7624 and was not in the original one https://github.com/RIOT-OS/RIOT/pull/5416

@smlng is it working with the normal `openocd` and if not, which version should be used?

closed 2020-02-25 by stale[bot]

jeandudey, 2020-03-02:
Solved for CC1312 in #13535 . Can be applied to this board as well.

rienafairefr, 2020-03-03:
Thanks @jeandudey , this should work, pinging @aabadie if he wants he can try to use this TI openocd to deploy the launchpad boards in FIT IoT-LAB :-)

aabadie, 2020-03-03:
I already tried to flash cc2650-launchpad using the TI version of openocd and can confirm that it works.
Deploying these boards in IoT-LAB is in my todo list, just lacking time.