RIOT-OS/RIOT #10079: Codacy: different problems (false positives)

gschorcht, 2018-09-29:
For several days I get a large number of false positives for one of my PRs with Codacy:

1. **Skipping configuration 'X' since the value of 'X' is unknown. Use -D if you want to check it. You can use -U to skip it explicitly.**
A lot (probably all) conditional code fragments that depend on macro definitions result in this error message. For example, the code
```
void board_init(void)
{
    #ifdef LED0_PIN
    gpio_init (LED0_PIN, GPIO_OUT);
    #endif
}
```
gives the error "Skipping configuration 'LED0_PIN' since the value of 'LED0_PIN' is unknown". The same error happens for:
```
#ifdef LED0_PIN
{
    .name = "LED red",
    .pin = LED0_PIN,
    .mode = GPIO_OUT,
    .flags = SAUL_GPIO_INIT_CLEAR
}
#endif
```
Using preprocessor definitions and doing something depending on their existence is a standard approach. These are definitely false positives.

2. **struct member 'structname::variable' is never used.**
Using structured data types to describe the structure of a register or a message is a standard approach. Mostly there are also unused elements. This should not lead to an error message. For example:
```
struct {
      uint8_t dlc :4;  /* frame payload length */
      uint8_t zero:2;  /* don't care, should be zero */
      uint8_t rtr :1;  /* remote transmission request frame */
      uint8_t eff :1;  /* extended ID frame format */
}
```
Member ```zero``` is just a place holder in the structure which has not to be used. Another example is:
```
struct hw_timer_regs_t {
    /* see Technical Reference, section 17.4 */
    struct {
        uint32_t unused      : 10;
        uint32_t ALARM_EN    : 1;  /* alarms are enabled */
        uint32_t LEVEL_INT_EN: 1;  /* alarms will generate level type interrupt */
```
I guess it has something to with ```cppcheck```. Option ```--suppress=unusedStructMember``` is probably required.

jia200x, 2018-10-04:
@gschorcht oh, I just checked these options are already disabled. Could you pint me to where you are getting these errors?

jia200x, 2018-10-05:
@gschorcht I triggered Codacy again and the problem is solved. It was an old run.

I will close this issue then. Please re-open if you detect another problem

closed 2018-10-05 by jia200x

gschorcht, 2018-10-08:
@jia200x Hm, still a problem, with last commit today, I had again the problem with PR #9426, see https://app.codacy.com/app/danpetry/RIOT/pullRequest?prid=2158012

jia200x, 2018-10-08:
@gschorcht we just transferred Codacy ownership to the RIOT CI. Maybe it's related to that. Let's wait a couple of hours and re-run ;)

gschorcht, 2018-10-11:
@jose BTW, I don't know why only for my PR #9426, but I ran into the undef problem today again, see https://app.codacy.com/app/danpetry/RIOT/pullRequest?prid=2158012

jia200x, 2018-10-11:
@gschorcht it's strange. I'm currently investigating what's going on.

I don't know why the link points to `danpetry` instead of the `RIOT-OS` organization. I will check.

jia200x, 2018-10-11:
@gschorcht in fact, that was the problem.

This is the actual run: https://app.codacy.com/app/RIOT-OS/RIOT/pullRequest?prid=2065996

It's passing. Might be a bug in the Github <-> Codacy integration. It won't be a problem to merge since Codacy is non blocking ;)

gschorcht, 2018-10-11:
@jia200x Thanks

gschorcht, 2018-12-18:
@jia200x Currently, I'm struggling with another strange problem. I tried everything to satisfy Codacy in PR #10033. 

[Codacy check](https://app.codacy.com/app/miri64/RIOT/pullRequest?prid=2520085) complains in the `README.md` files in commit https://github.com/RIOT-OS/RIOT/pull/10033/commits/f9528ba8933b6215169fc8de4fe437d9a92c24a2 that I have no spaces in front of the list items:
`[list-item-indent] Incorrect list item indent: add 2 spaces`

But, if I add these 2 spaces in front of the list items it complains :worried:
`[list-item-bullet-indent] Incorrect indentation before bullet: remove 2 spaces`

Furthermore, it insists on a an empty line in between the items.
`[list-item-spacing] Missing new line after list item`

Both indicated problems are false positives.