RIOT-OS/RIOT #10089: Generic CPU status data

olegart, 2018-10-01:
I'm thinking about some kind of global struct variable to hold various CPU-specific data:

* flash size (on Cortex-M can be calculated in runtime)
* RAM size (on Cortex-M can be calculated in runtime)
* EEPROM size (on Cortex-M can be calculated in runtime)
* CPU ID
* current CPU temperature
* current VDD voltage
* current VBAT voltage
* maybe something else?

As for now, all this data live in form fo various functions, defines or isn't available at all (e.g. you can't really get precise STM32 VDD voltage using high-level API only)

Should be nice to have it all in one place, and more or less independent of specific CPU.

What do you think?

olegart, 2018-10-01:
Let's improvise a little bit how API and usage might look.

struct {
    uint8_t id[16];
    uint8_t size;
} cpu_id_t;

(I believe there are no CPUs with ID over 128 bits? Anyway, ID size can be #defined)

struct cpu_status {
    uint32_t ram_size; /* bytes */
    uint32_t flash_size; /* bytes */
    uint32_t eeprom_size; /* bytes */
    int16_t temperature; /* 0.1 degrees Celsius */
    int16_t vdd; /* mV */
    int16_t vbat; /* mV */
    cpu_id_t cpuid;
} cpu_status_t;

Structure should not be global, I've changed my mind. Function _cpu_status_t get_cpu_status(void)_ should be.

Flash, RAM, EEPROM sizes are filled once and for ever at cpu_init() by defines or (preferably) actual calculation - later provides a way to build a single firmware for multiple CPUs from the same lineup, e.g. STM32L151CBU6A and STM32L151CCU6, every time using all available memory. CPU ID structure is filled at cpu_init() too. Voltages and temperatures are calculated each time get_cpu_status() is called.

Some custom fields may be added for specific CPUs only. 0 values should be reported as memory sizes if there's no such kind of memory, INT16_MIN (-32767) should be reported if temperature or voltage data is not available on this CPU.

gebart, 2018-10-01:
How about using SAUL (possibly using #9810) for the current VBAT, VDD, coretemp?
#9810 could be further extended in a follow up to support scaling according to a reference measurement, for example when paired with a 1.0 V reference signal on one ADC channel, to rescale from ADC counts to millivolts.

olegart, 2018-10-01:
@gebart I'm thinking of putting different CPU-related parameters together, you know, something like /proc/cpu on linux systems, so it will be the place to look if you need something CPU-related, so it can't be all ADC-related stuff.

Even temperature sensor on some CPUs is digital, not analog (e.g. nRF52).

BTW, thanks for pointing to saul_adc, I should check if it helps with #10090 now...

closed 2019-09-10 by stale[bot]