RIOT-OS/RIOT #10090: ADC data on STM32 are useless

olegart, 2018-10-01:
I'm not sure if it's bug or a feature, but...

1) ADC scale on STM32 is related to VDDA voltage, which usually means VDD voltage, which often means not-so-stable battery voltage

2) VDD can be calculated and ADC values properly scaled by reading internal ADC VREF channel

3) Here comes the problem: VREF calibration values are CPU-specific, sometimes there are none, sometimes you need to know an address. Also, calibration data itself are CPU-specific.

Accessing calibration data registers from high-level software doesn't seem to be the right why, because HAL. **Without calibration, ADC data are completely meaningless in most cases**.

Should low-level adc_sample function be changed to return actual millivolts instead of ADC counts?

P.S. In our firmwares, we changed adc_sample quite so years ago and quite happy with it.

gebart, 2018-10-01:
The periph/adc API is intended to be a low level API, therefore it makes the most sense to keep the value in ADC counts instead of having the driver try to be smart and rescale everything to voltage.
Additionally, some analog sensors will have output signals which are relative to the VDD, so if the sensor is supplied from the same low accuracy VDDA as the CPU, the errors will cancel out as long as the error does not contain a lot of high frequency noise.
Perhaps improving the SAUL analog driver could solve some of the issues here?

olegart, 2018-10-01:
@gebart, correct me if I wrong, but it seems to me that SAUL is pretty high-level thing itself, i.e. there's no part with direct access to CPU registers, etc, it only uses existing low-level API?

If so, it won't work this way.

P.S. Ratiometric sensors is not a problem if you know precise VDD value.

closed 2019-09-10 by stale[bot]