RIOT-OS/RIOT #10103: semtech-loramac not joining with DR above 2 after update

leandrolanzieri, 2018-10-02:
#### Description
While trying to run the lorawan example I found that any join attempt after setting the Data Rate to a number above 2 fails. This seems to be related to the recently merged PR that updates the semtech-loramac implementation #8864. I tested the lorawan example with the previous version (checked out a commit before the PR was merged) and it works.

#### Steps to reproduce the issue
1. Enable debug on ```semtech_loramac.c```.
2. Run the lorawan example:
```
BOARD=b-l072z-lrwan1 make clean flash term -C examples/lorawan/
```
You can also try to modify the Data Rate, and check that it works:
```C
    semtech_loramac_set_dr(&loramac, LORAMAC_DR_2);
```

#### Expected results
Get something like 
```
INFO # Starting join procedure
INFO # [semtech-loramac] Starting join procedure: 0
INFO # [semtech-loramac] loramac cmd msg
INFO # [semtech-loramac] starting OTAA join
INFO # [semtech-loramac] Transmission completed
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MLME confirm event
INFO # [semtech-loramac] join succeeded
```

#### Actual results
```
INFO # Starting join procedure
INFO # [semtech-loramac] Starting join procedure: 0
INFO # [semtech-loramac] loramac cmd msg
INFO # [semtech-loramac] starting OTAA join
INFO # [semtech-loramac] Transmission completed
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] RX timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] RX timer timeout
INFO # [semtech-loramac] MAC timer timeout
INFO # [semtech-loramac] MLME confirm event
INFO # [semtech-loramac] join not successful
INFO # [semtech-loramac] loramac join notification msg
INFO # Join procedure failed
```
----

#### Further tests
I also tried running the ```tests/pkg_semtech-loramac``` application but it does not join when the Data Rate is above 2:
```
> loramac set dr 5
INFO #  loramac set dr 5

> loramac get dr
INFO #  loramac get dr
INFO # DATARATE: 5

> loramac join otaa
INFO #  loramac join otaa
INFO # Join procedure failed!

> loramac set dr 0
INFO #  loramac set dr 0

> loramac join otaa
INFO #  loramac join otaa
INFO # Join procedure succeeded!
```

leandrolanzieri, 2018-10-02:
Maybe @jia200x or @aabadie have an idea?

aabadie, 2018-10-02:
Can you try to increase the value [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/contrib/semtech_loramac_timer.c#L67) from 22 to 50 (as it is before) ?

leandrolanzieri, 2018-10-04:
By changing that timeout to 50 now it receives the downlink and joins properly.