RIOT-OS/RIOT #10104: Error | using Ping6 from shell

npcode15, 2018-10-02:
Hello,

I am trying to use ping6 from the shell and I am getting the following error:
[image]

I am sure I have the necessary packages.
```
# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps
USEMODULE += xtimer
#To include the default network device(s) on your board
USEMODULE += gnrc_netdev_default
#To auto-initialize these network devices as GNRC network interfaces
USEMODULE += auto_init_gnrc_netif
#You may choose to build either as an IPv6 Node
USEMODULE += gnrc_ipv6_default
#For ICMPv6 echo request/reply (ping) functionality:
USEMODULE += gnrc_icmpv6_echo
#For UDP support include
USEMODULE += gnrc_udp
USEMODULE += od
```

Also, please point out the underlying code used by the command 'ping6'.

Thank you.

miri64, 2018-10-02:
Hi @npcode15! The assertion gives you a hint what's going wrong (with `DEVELHELP=1` set you would even get more): The thread you are calling `ping6` from (usually the `main` thread if you are calling `shell_init()` from `main()`) is not having a message queue. You can initialize one by calling [msg_init_queue()](http://api.riot-os.org/group__core__msg.html#ga480e6f32c8ab18579b62a890f3fda2cd) somewhere in the beginning of that thread (or the `main()` function if that applies).

closed 2018-10-02 by miri64

npcode15, 2018-10-05:
@miri64 thanks for the answer. But, it would be great if you could also point out the backend code for the ping. Thanks.

miri64, 2018-10-05:
```
$ git grep ping6
[…]
sys/shell/commands/shell_commands.c:    { "ping6", "Ping via ICMPv6", _icmpv6_ping },
[…]
```

;-)

You can find the `_icmpv6_ping` handler also [in that directory](https://github.com/RIOT-OS/RIOT/blob/0d5276790ea4000c8678d125aba43d997256965b/sys/shell/commands/sc_icmpv6_echo.c).

npcode15, 2018-10-07:
@miri64 Thanks.