RIOT-OS/RIOT #10121: RIOT cannot compile with the latest version of macOS (10.14) and Xcode 10

Zhiyi-Zhang, 2018-10-07:
#### Description
RIOT cannot compile with the latest version of macOS.

#### Steps to reproduce the issue
Compile the example/hello_world with default configuration (native).

#### Expected results
No errors.

#### Actual results
ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
ld: warning: ignoring file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/lib/libdl.tbd, missing required architecture i386 in file /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/lib/libdl.tbd
Undefined symbols for architecture i386:
  "_strndup", referenced from:
      _tty_uart_setup in periph.a(uart.o)
  "_sigaltstack$UNIX2003", referenced from:
      _native_interrupt_init in cpu.a(irq_cpu.o)
...
...
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)

#### Versions
Operating system: Mac OSX
Build environment: 
native gcc: Apple LLVM version 10.0.0 (clang-1000.11.45.2)
clang: Apple LLVM version 10.0.0 (clang-1000.11.45.2)

gebart, 2018-10-07:
It looks like you need an older xcode version.
https://forums.developer.apple.com/thread/108704

Future MacOS releases may drop 32 bit support for applications though, so RIOT would have to add support for building as a 64 bit binary to continue supporting MacOS.

Zhiyi-Zhang, 2018-10-07:
> It looks like you need an older xcode version.
> https://forums.developer.apple.com/thread/108704

A downgrade is always not the best choice :-(
Hope RIOT team can handle this issue as soon as possible since new macOS has been revealed for some time.

jia200x, 2018-10-26:
(I added triple ticks to the log so it formats better)

larseggert, 2019-09-09:
Before I even get the linker error above, I needed to add
```
CFLAGS += -isystem /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include
```
to my `Makefile`, so the headers are being found.

smlng, 2019-09-09:
MacOS is not supported as RIOT native does not run on 64Bit CPUs (yet)

smlng, 2019-09-09:
Hence, this can be closed

miri64, 2019-10-29:
Weellll, this _could_ [become a problem in Ubuntu as well](https://ubuntu.com/blog/statement-on-32-bit-i386-packages-for-ubuntu-19-10-and-20-04-lts)...

jenswet, 2020-10-20:
Are there any news about this?

miri64, 2020-10-20:
There is https://github.com/RIOT-OS/RIOT/pull/13009, but development on that seems to be stalled.

miri64, 2021-07-12:
Now that Apple made the switch to ARM, [there are more problems](https://forum.riot-os.org/t/compiling-riot-on-apple-m1/3293)

Teufelchen1, 2022-11-02:
Since PR #18735 landed, macOS native is officially no longer supported. The problems with macOS go far beyond the 64bit issue. I believe this issue can be closed.

miri64, 2022-11-02:
Yepp

closed 2022-11-02 by miri64