RIOT-OS/RIOT #10133: gnrc_ipv6_ext: needs heavy refactoring

miri64, 2018-10-08:
I was mentioning this already several times offline, but I think our current IPv6 extension handling is not very well designed to put it mildly. Currently, only RPL source routing headers are supported, but they are already quite a mess and make GNRC's reception handling within IPv6 already quite hard to follow (not to mention debug or optimize). I've already tried to refactor some stuff, but it's quite a big knot to untangle. So I think its best to discuss this first.

My main issue is the very weird call hierarchy that is followed for the handling of the RPL routing handler:

```
gnrc_ipv6_demux();
+> gnrc_ipv6_ext_demux()           # after some prior special handling for extension headers
                                   # also some nested recursion to gnrc_ipv6_demux() in here.
                                   # Not great with limited stack space ;-)
   +> _handle_rh()                 # within gnrc_ipv6_ext.c
      # some weird stuff write-protecting everything until then
      # because now we throw away every context information GNRC provides us with
      +> ipv6_ext_rh_process()     # leaving the GNRC "realm" for a bit
         +> gnrc_rpl_srh_process() # returning to a GNRC-specific implementation of RPL
```

I believe we can get rid of most of the weird stuff and additional parameter handling if we do two things:

1. Removing `ipv6_ext_rh_process()` within that hierarchy (or moving it into GNRC and provide 
the context carrying `gnrc_pktsnip_t`s instead of `ipv6_hdr_t` and `ipv6_ext_rh_t`) and providing `gnrc_rpl_srh_process()` with the GNRC-specific information it needs to handle and write-protect the provided headers itself (bonus: no need for abstraction breaking `gnrc_pktbuf_duplicate_upto()` anymore).
2. Moving all extension header and encapsulated IPv6 header handling out of `gnrc_ipv6_demux()` and putting the whole handling of extension headers into a separate function. By their very nature those headers are too different from normal payload headers. By doing that iteratively we can also get rid of above mentioned recursion. It also hopefully, will get rid of all the [`current` and `interested` parameters](https://github.com/RIOT-OS/RIOT/blob/521d7bdfc27b7c090d51244d48814d52484e6a1a/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L105-L108) in the current code whose meaning make the reception code so hard to decipher and API changes like the one in #7235 could become unnecessary (we might even be able to make `gnrc_ipv6_demux()` private to `gnrc_ipv6.c` again).

tcschmidt, 2018-10-17:
Some random remarks:

- Handling of extension headers is certainly not well-situated within RPL. However, information exchange with other modules (such as RPL, IPSec Modules, ...) should be well designed. Note that they are often typed, which can be used to address modules by messaging. I would suggest to look at other IP-stacks first.
- Additional headers (RPL routing header) would also be needed, I suppose.
- For modularisation: Not all extension headers need to be supported by all stack functions. Hop-by-Hop for instance only appears at routers, destination option at both.
- Processing flow is important: Base header is of course first, but extension headers may change the information extracted from the base header (-> mobility destination, routing).

miri64, 2018-10-23:
I've created [a new project](https://github.com/RIOT-OS/RIOT/projects/24) to track this work.

miri64, 2018-10-23:
> Handling of extension headers is certainly not well-situated within RPL. 

I never said that. The handling happens within the `gnrc_ipv6_ext` module where it belongs to (even in current master). Just the RPL-specific parts (e.g. RPL SRH) are handled within a submodule of RPL. There the *are* well situated. The problem is that due to some header-handling foobar, the GNRC context is left to a more (non-GNRC ;-)) *generic* header handling, which throws out all the info the RPL SRH handler actually needs which is situated within the packet buffer.

miri64, 2018-12-18:
Since #10238 is merged and the rest are just additional features and tests, I consider this issue solved.

closed 2018-12-18 by miri64