RIOT-OS/RIOT #10134: `struct netdev_radio_rx_info` RSSI value is "wasteful"

maribu, 2018-10-08:
# Background
`netcev_radio_rx_info_t` is defined as:

``` C
struct netdev_radio_rx_info {
    int16_t rssi;       /**< RSSI of a received packet in dBm */
    uint8_t lqi;        /**< LQI of a received packet */
}; 
```

According to [Wikipedia's article on dBm](https://en.wikipedia.org/wiki/DBm) a value of 60 dBm equals an microwave oven and -192.5 dBm is about the received signal strength of on 1Hz bandwidth in outer space. I expect that no IoT device will ever transmit up 80 dBm and that being able to detect signals at -192.5 dBm is out of reach for the near future for any low power receiver.

# The Problem
This would fit perfectly in one byte of data (using an offset). On the other side, many transceivers will be able to output an RSSI value with a precision of mare than 1 dBm. (E.g. the CC1101/CC1100 has a resolution of 0.5 dBm.)

# The Proposal
By defining `struct netdev_radio_rx_info::rssi` to not be in dBm, but in E-02 dBm a resolution of 0.01 dBm would be possible while still having a range of -327.68 dBm to 327.67 dBm. This should be future proof for quite some time. (The range should be even sufficient for forever.)

maribu, 2018-10-08:
@miri64 Do you wan't to have a look?

bergzand, 2018-10-08:
Something like this was discussed before in #6887. One of the lessons from there is that we should also take the range of LoRa radios into account. This is not an argument against or in favor of your idea, but more a precaution against repeating a previous discussion

maribu, 2018-10-08:
@bergzard: -192.5 dBm as thermal noise of outer space should be a really good lower bound - everything below would be indistinguishable from that background noise. And a microwave oven (60 dBm) a pretty good upper bound. But up to 327 dBm could be stored. I lt seems to require more than 1E29 Watt of electricity for 327dBm. This is way above any IoT scenario I can image for the next few millennia. I feal confident that the range is big enough :-)

olegart, 2018-10-09:
* LoRa fits here, on hardware level it uses 1 byte with an offset anyway
* I'm not sure saving a byte is enough to justify offset introduction

maribu, 2018-10-09:
@olegart: What do think about keeping it two bytes long, but having it in E-02 dBm instead of dBm?

olegart, 2018-10-10:
@maribu it's a gift from heavens if you can get ±1 dB accuracy from a real RF IC (and you have to make multiple measurements, then account for LNA settings, then account for the signal duration...). Anything better is just impossible on generic chips. I, personally, wouldn't vouch for anything better than ±3 dB with a single measurement.

E.g. there's TI [design note DN055](http://www.ti.com/lit/an/swra114d/swra114d.pdf) on RSSI measurements on CC1101 — RSSI readings are varying periodically with time, see fig. 2, 3 and 4. On LoRa, you'll get better RSSI readings on longer packets, and so on and so forth.

So there's no reason to improve RSSI resolution.

maribu, 2018-10-10:
@olegart: Getting ±1 dBm *relative* accuracy is not uncommon with current technology. Assume you want to use the RSSI info for *relative* distance estimation and have a transceiver with ±1 dBm (relative) accuracy and 0.5 dBm resolution. By reducing the resolution from 0.5 dBm to 1 dBm you already have a precision loss from ±1dBm to ±1.5 dBm. A 50% precision loss for the application of distance estimations is already big with today's technology, but it will become *huge* when technology becomes more precise.

Also: The rssi info is already stored in 2 bytes. So there is no overhead in memory to pay. This feature would basically come for free.

olegart, 2018-10-10:
> use the RSSI info for relative distance estimation

Doesn't work, believe me. Distance estimation based on RSSI is the worst idea ever, RSSI in real world fluctuates by ±10 dBm at least because of object position, angle, antenna position, obstacles, reflections and god knows what else.

0.5 dBm difference you'll get by tilting antenna by 10°. If its a wearable/portable device, you'll get ±5 dBm just by holding it a bit differently (and I'm not talking about iPhone 4 :).

All precision positioning systems are based on ToF or, better, TDoA calculations.

> This feature would basically come for free

The problem is its counter-intuitive and I see no real benefit. Right now it's just decibels, tomorrow you'll have to know its format to use it.

maribu, 2018-10-10:
> Right now it's just decibels, tomorrow you'll have to know its format to use it.
Let me do a reductio ad absurdo by applying your argument to a different measurement problem. Lets say we want to measure lengths instead and the precision would be 1nm now. Is there any reason to expect length to no longer be given in meters (with some metric prefix for convenience) just because measurement tools get more precise? Obviously not. So why would you expect the unit dBm to be outdated in the future just by more accurate measurements?

> Doesn't work, believe me. 
There is [plenty of research going on](https://scholar.google.de/scholar?hl=en&q=rssi+distance)  in that direction. RSSI based distance estimation may not be reliable right now. But this is not a reason to prevent application developers from accessing the full resolution of the underlying data. Maybe RSSI distance estimation will become reliable in the near future - and maybe not. But are we in any position to decide this on behalf of someone who might want to use RIOT to do research on this problem?

My point is: Throwing away information which some application developer could potentially use is undesirable. If there is a trade off and the cost of keeping the information is considered too high, this would totally make sense to me. But right now I don't see any cost in keeping the resolution. Thus, I consider throwing away some resolution wasteful.

olegart, 2018-10-10:
There's no information. There's only meaningless noise. In physics, actually, it is considered an error to provide more digits in a number than measurement accuracy allows.

No, RSSI estimation will never be reliable, no matter what research you just found (I found some on flat Earth, so what?). I explained it already — it just doesn't matter how accurate your measurements are. Replace your radio with $20,000 Keysight analyzer with 0.01 dBm accuracy, and your RSSI-based location measurements are still exactly the same crap. Put your hand around antenna, you'll get -10 dB attenuation — and how exactly 0.01 dBm accuracy gonna fix that?

> from accessing the full resolution of the underlying data

You see, actual resolution of any analog sensor is infinitesimal. Still, we are perfectly ok with digitizing it with 12-bit, 10-bit or even 8-bit precision.

I can't say I hold something against 0.01 dBi RSSI resolution. I just don't see any meaning in it.

maribu, 2018-10-11:
> In physics, actually, it is considered an error to provide more digits in a number than measurement accuracy allows. 

If a physicist publishes a result with more decimal places than the accuracy allows, this is indeed an error. However, no physicist would reduce the resolution of intermediate results, as this would reduce the accuracy of the result to publish. I personally see the data in this API as an intermediate result. So in my opinion the application developer is in charge to decide to which degree the data is accurate for his particular use case. E.g. without calibration the accuracy of one Atmel device was 5.5 dBm, but its relative accuracy was 1 dBm. So for some use cases the application should treat the accuracy as 5.5 dBm, but for others it could be justifiable to use 1 dBm as accuracy.

Also: Let's assume we treat the RSSI value as the "final result" that should not be given in higher resolution than the accuracy allow. In that case we would need a resolution that depends on the accuracy of a particular device - so we would end up with different resolutions per device. This would be impractical.

> You see, actual resolution of any analog sensor is infinitesimal. Still, we are perfectly ok with digitizing it with 12-bit, 10-bit or even 8-bit precision.

Again, 100% true what you are saying. But the trade-off is different here: Using a 12-bit ADC instead of an 8-bit ADC increase the hardware cost and has impact on the sampling time. For a particular use case an 8-bit sample is enough, for another 12 bit or even a 16 bit sample is required.

RIOT's [adc_sample](http://api.riot-os.org/group__drivers__periph__adc.html#ga14de5ca183ebf595a4bb1f0b63df76a3) API returns in any case an `int`, which is (as far as I know) on every supported platform at least 16 bit wide. Thus, the API does not limit the maximum possible resolution here.

When giving RSSI in 0.01 dBm resolution instead of 1 dBm the API would not limit the maximum resolution for quite some time (or at least I believe so). Of course this would not magically increase the accuracy at all. (This could than be clearly documented. E.g. via Doxygen something like that: 

``` C
/**
 * ...
 * @warning Please consult the datasheet of the used network device what
 *          the actual accuracy of the device and its native resolution are
 */
```

> Put your hand around antenna, you'll get -10 dB attenuation 

Sure. In the example of RSSI based distance estimation: Using RSSI alone would only work with cutting edge hardware in outer space, where effects like interference and multipath propagation can be ignored. It wouldn't be an interesting research opportunity if it would be easy and reliable :-)

maribu, 2018-10-11:
Maybe I can get a college of mine involved into this conversation. He is interested in research of RSSI based distance estimation. It could be good idea to involve people into this conversation who would end up actually using this API :-D

olegart, 2018-10-11:
> RIOT's adc_sample API returns in any case an int, which is (as far as I know) on every supported platform at least 16 bit wide. Thus, the API does not limit the maximum possible resolution here.

By the way, it does. There are 24-bit ADCs on 16-bit CPUs, so it's better be changed to int32_t (besides, using int is a MISRA-C:2004 rule 6.3 violation) ;)

> who would end up actually using this API :-D

He will end up rewriting half of the driver anyway, as single RSSI measurement performed by current drivers is not accurate irregardless of storage type.

maribu, 2018-10-11:
> He will end up rewriting half of the driver anyway, as single RSSI measurement performed by current drivers is not accurate irregardless of storage type.

I could implement a SAUL interface to the RSSI value for the CC110x transceivers. These are constantly updating the RSSI while in RX mode anyway. The `phydat_t` conveniently has the scale member, so the resolution can there be chosen to fit the actual resolution of the device, instead of using 0.01 dBm as "one size fits all". (Interestingly the CC110x RSSI has a different semantic than the LoRa once: While the LoRa RSSI value is measured during the reception of the frame, the CC110x transceiver returns the most recent RSSI value measured before the Sync word was completely received.)

> By the way, it does. There are 24-bit ADCs on 16-bit CPUs, so it's better be changed to `int32_t`

I agree. Do you want to create a PR, or should I?

olegart, 2018-10-11:
#10151

Yours is to change RSSI to 0.01 dBm resolution :)

closed 2019-09-10 by stale[bot]