RIOT-OS/RIOT #10161: Support for asian bases Microcontroller, RTCs & ADCs

Citrullin, 2018-10-14:
#### Description
It's just a more general topic. Asia is an interesting market, especially for real-time operation system. It might be a good idea to push more into this market, since it is well known for more open source behavior. 

### Useful links
https://www.youtube.com/watch?v=VYhAGnsnO7w
https://lcsc.com/

jcarrano, 2018-10-15:
> https://www.youtube.com/watch?v=VYhAGnsnO7w

WOW!

MrKevinWeiss, 2018-10-16:
Gotta love Dave.

miri64, 2018-10-17:
Asia is a big continent and also a big market. Can you specify what kind of microcontrollers you are talking about?

Citrullin, 2018-10-17:
> Asia is a big continent and also a big market. Can you specify what kind of microcontrollers you are talking about?

It's more a general topic. But of course, I could take a look into that and try some microcontroller, RTCs & ADCs in the future. Maybe I can also find out which are the common used ones.

PeterKietzmann, 2019-02-21:
@Citrullin any progress finding commonly used 'asian based Microcontrollers'? I find it cool to have kind of idea-sharing in the community but I don't see why we should keep this issue open. How do the others  see it?

Citrullin, 2019-02-21:
@PeterKietzmann Since this currently also a topic at my work, I will pick this up again at some point. But you are right. I just close it.

closed 2019-02-21 by Citrullin