RIOT-OS/RIOT #10171: Move RDMs to an external repository

aabadie, 2018-10-16:
I open this issue to propose that RDMs are moved to another repository than the main upstream RIOT repo. This is for many reasons:
- RIOT repository should focus on what it is: the source code of a real-time, multi-threaded operating system for microcontrollers. RDMs are different from documentation, which documents the RIOT API.
- RDMs are higher lever documents describing RIOT philosophy and developer organization, etc but not code.
- The last opened RDMs, #10162, is generating a lot of debate and traffic, but not related to code or improvements of RIOT. See [this discussion]() for example. I fear that RDMs end up in endless discussions like this most of the time.
- I see RDMs as the equivalent of Python PEPs but for RIOT. Maybe it would be good to get more inspirations from Python there ? [Python PEPs repository](https://github.com/python/peps) is separate from Python upstream repository.

Why not open the `rdms` project under the RIOT-OS organisation ? And put there the RDMs ?

emmanuelsearch, 2018-10-16:
@aabadie in general: if a proposed RDM is deemed not related to "code or improvements of RIOT" let's drop this RDM proposal.
If an RDM proposal is related (directly or indirectly) then having some discussions inside the RIOT repo is not necessarily a bad thing?
There are advantages either way -- keeping RDMs in the repo or moving them in separate repo.
Personally I'm not opposed to moving them out.
Let's just make sure that whatever we decide, it does not result in a bunch of "useless or ignored RDMs".

Concerning the discussion in #10162 in particular, the RDM proposal is 
- in a pre-alpha state, 
- drafted by a single person, 
- thrown out out-of-the-blue, 
- on a topic that is too wide and high level not to be slippery... 

Bottom line: it is not surprising that the discussion is wild. 
One can expect that other RDM proposals could be crisper when PRed, yielding different types of discussions, more focused.

So how about we wait until a few more RDMs are proposed and/or published, and *then* decide whether to move to another repo?

danpetry, 2018-10-16:
I agree that we should move them to a different repo, for the reasons given above. +1

> RDMs are different from documentation

A few people have expressed this sentiment, and I believe it's incorrect. Anything that isn't code and is written down is documentation, including project plans, Kanban tickets, issues etc. When people say "it's not documentation" I think they're really saying "It's not code-based documentation such as commenting/Doxygen".

Re the discussion in https://github.com/RIOT-OS/RIOT/pull/10162, the discussion is going as intended from my side at least. The subject matter is broad and has high impact and so it's productive and healthy that such vibrant discourse has been generated.

jia200x, 2018-10-16:
> RDMs are different from documentation, which documents the RIOT API.

I would specify RDM are different from (code) documentation, as @danpetry pointed out

Citrullin, 2018-10-16:
Do I get this correct: RDMs are used for architectural changes. More conceptional changes? And organization things?

emmanuelsearch, 2018-10-17:
@Citrullin see [RDM0](https://github.com/RIOT-OS/RIOT/blob/master/doc/memos/rdm0000.md) section 1.
RDMs are expected to document things like architectural design decisions, code structure, pitfalls and best practices... and possibly also RIOT community processes, position with respect to some related external technical debates etc.

danpetry, 2018-12-05:
Bump on this. (https://github.com/RIOT-OS/RIOT/pull/10162 hasn't died, it's just waiting on checks from people before I reboot the discussion on the mailing lists.)
How do we decide stuff via issues, like here for example where you can't really raise a PR?
We could say that three people have to ACK and no blocks? If so, ACK from me...

closed 2019-09-10 by stale[bot]