RIOT-OS/RIOT #10177: Warn when casts do no preserve qualifiers (-Wcast-qual)

jcarrano, 2018-10-17:
### Description

With the compiler options we are using now, explicit casts that discard a qualifier (like `volatile` or `const`) do not trigger a warning.

I propose to enable these warnings by adding the compiler flag `-Wcast-qual`.

I tried compiling a couple of simple examples to see how feasible it is and I found I found several violations. Some of them are in the kernel. **That does not mean the kernel is buggy, read below**.

### Kernel

Before somebody says "we could disable this warning for stuff in /core". Yes, but we could also fix it, and in any case, the core is the most critical part, so we better have lots of checks there.

The kernel defines some stuff, like `sched_active_thread` as volatile, because the active thread can change at any moment.

Of course, when the kernel is running the thread is not changing because it is the kernel who changes threads, so in that situation it is "ok" to discard the qualifier.

I'd argue, however, that it is not that "it's ok to discard the qualifier", but  rather that "in that situation the variable is not const". The hard part is how we explain that to the compiler.

#### `containerof`

When using GCC `containerof` is defined as:

```c
#   define container_of(PTR, TYPE, MEMBER) \
        (__extension__ ({ \
            __extension__ const __typeof__ (((TYPE *) 0)->MEMBER) *__m____ = (PTR); \
            ((TYPE *) ((char *) __m____ - offsetof(TYPE, MEMBER))); \
        }))
```

Why is that `const` there? Is it necessary?

### Other code

Do a `WERROR=0 CFLAGS=-Wcast-qual make` and you will see a number of violations, most of them discarding `const`.

kaspar030, 2018-10-17:
> With the compiler options we are using now, explicit casts that discard a qualifier (like `volatile` or `const`) do not trigger a warning.

Luckily. What are you trying to optimize?

jcarrano, 2018-10-17:
> What are you trying to optimize?

Nothing. I'm trying to get all the false positives out of the way so that the warning can be enabled. Then we don't merge code that has incorrect casts. The kernel is clearly a false positive.

My main point here is that there is no point in setting a rule only to break it later. Why make something const if later we ignore the const? (the case with volatile is different, because nothing can break by not making a variable const, but it can break by not making it volatile).

keestux, 2019-01-04:
> My main point here is that there is no point in setting a rule only to break it later. Why make something const if later we ignore the const?

I agree with @jcarrano 

>  (the case with volatile is different, because nothing can break by not making a variable const, but it can break by not making it volatile).

Here I disagree, I **do** think it is bad to ignore `volatile`. It is explicitly written as undefined behaviour [1].

[1] https://en.cppreference.com/w/c/language/volatile

closed 2019-09-10 by stale[bot]