RIOT-OS/RIOT #10192: tests/bench_runtime_coreapis times out

bergzand, 2018-10-18:
#### Description

Sometimes the `tests/benc_runtime_coreapis` causes a build failure on murdock. Tiggered with #10183 at the moment.

#### Steps to reproduce the issue

Wait for the CI to cause another unrelated build failure.

#### Expected results

Murdock doesn't complain.

#### Actual results
```
-- running on worker mobi3.inet.haw-hamburg.de thread 21, build number 12258.
make: Entering directory '/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/tests/bench_runtime_coreapis'
Building application "tests_bench_runtime_coreapis" for "native" with MCU "native".

   text	   data	    bss	    dec	    hex	filename
  30901	    384	  47764	  79049	  134c9	/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/build/tests_bench_runtime_coreapis.elf
make: Leaving directory '/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/tests/bench_runtime_coreapis'
make: Entering directory '/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/tests/bench_runtime_coreapis'
/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/build/tests_bench_runtime_coreapis.elf   
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: buildtest)
Runtime of Selected Core API functions

                 nop loop:      2236us  ---   0.002us per call  ---  447227191 calls per sec

             mutex_init():      5704us  ---   0.005us per call  ---  175315568 calls per sec
        mutex lock/unlock:   5667514us  ---   5.667us per call  ---     176444 calls per sec

       thread_flags_set():   2562719us  ---   2.562us per call  ---     390210 calls per sec
     thread_flags_clear():   2458604us  ---   2.458us per call  ---     406734 calls per sec
thread flags set/wait any:   7559558us  ---   7.559us per call  ---     132282 calls per sec
thread flags set/wait all:   7153391us  ---   7.153us per call  ---     139793 calls per sec
thread flags set/wait one:   7515864us  ---   7.515us per call  ---     133051 calls per sec

Timeout in expect script at "child.expect_exact('[SUCCESS]', timeout=TIMEOUT)" (tests/bench_runtime_coreapis/tests/01-run.py:18)

/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/tests/bench_runtime_coreapis/../../Makefile.include:549: recipe for target 'test' failed
make: *** [test] Error 1
make: Leaving directory '/tmp/dwq.0.8292186702610634/429526858e9c1664e23af3f174ff6e40/tests/bench_runtime_coreapis'
-- build directory size: 1.1M
```

#### Versions

Murdock versions

jia200x, 2018-10-22:
I confirm the same happens to me in some boards (`b-l072z-lrwan1`)

danpetry, 2018-10-23:
We added a custom timeout to this test as described [here](https://github.com/RIOT-OS/RIOT/pull/9348#discussion_r196032265). Maybe needs to be lengthened more?

aabadie, 2019-01-04:
The automatic test script was improved in #10588 to avoid these timeout problems on nrf51 based boards (but also on atmega2560). Now it also works well on b-l072z-lrwan1.
Maybe we can close ?

closed 2020-02-26 by stale[bot]