RIOT-OS/RIOT #10197: ISR Stackoverflow (Radio Mode)

npcode15, 2018-10-18:
Hello, 

I received the following error when I try to send a message via radio mode.

[image]

**Payload size** - less than 250 bytes.
This only happens when I try to modify the string using 'asprintf'.

And if I use 'strcat' to modify the string, the receiving node never receives the packet sent via radio mode. Everything works normally if I do not modify the string at runtime.

**CODE:**
```
            strcat(_msg, "#");
            char str[10];
            sprintf(str, "%d", cnt-1);
            asprintf(&_msg,"%s%s", msg1, "#");
            asprintf(&_msg,"%s%s", "", str);
            
            timeStampI = xtimer_now();
            printf("Sending packet");
            xtimer_usec_from_ticks(timeStampI));
            
            if (cnt < 1000)
                send(mac_address, msg1);
            else
            if (cnt >= 1000 && cnt <2000)
                send(mac_address, msg2);
            xtimer_sleep(1);
            *_msg = '\0';
```

**NOTE:** Similar code is used for UDP, it has no such issue.

Thank you.

kaspar030, 2018-10-18:
The "ISR stack overflow" message always pops up after the panic handler has used ```printf()```, as that uses too much stack. The actual hard fault probably happened way before.
Can you do "make debug" at that point and do what the hard fault error message suggests (print a back trace)?

Other than that we might need more context. Which board? What is the surrounding code? Where and how is ```_msg``` initialized? What does ```send()``` do?

npcode15, 2018-10-18:
@kaspar030 

**BOARD** - IoTLab-m3
**_msg init* - *char *_msg[250] = ""; (Initialized gloabally, before main)
**send()** - sends the radio message. I found this code from the RIOT itself. (I am sure send() is not a problem)

PeterKietzmann, 2019-02-21:
@npcode15 are you still around? If you want further help we would need you to make the whole *main.c * file available. It should be fine if you just point to your RIOT branch on GitHub.

closed 2019-09-25 by stale[bot]