RIOT-OS/RIOT #10206: Lua_basic example doesn't flash in b-l072z-lrwan1

jia200x, 2018-10-19:
#### Description
The `lua_basic` example fails to flash on `b-l072z-lrwan1` in current RIOT master. It seems there's a tiny overflow that's not being detected by the build system.

#### Steps to reproduce the issue
Flash `lua_basic` example in `b-l072z-lrwan1`.

#### Expected results

The application is flashed without errors
#### Actual results

```
make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C /home/jialamos/Development/RIOT/examples/lua_basic flash-only test
make[1]: Nothing to be done for 'Makefile.include'.
make[1]: Nothing to be done for 'Makefile.include'.
/home/jialamos/Development/RIOT/dist/tools/openocd/openocd.sh flash
### Flashing Target ###
Open On-Chip Debugger 0.10.0+dev-00226-g1c2e3d41d (2018-01-02-18:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2-1.cfg is deprecated, please switch to interface/stlink.cfg
hla_swd
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_assert_srst
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.252174
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32l0.cpu        hla_target little stm32l0.cpu        reset
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000414 msp: 0x20000200
auto erase enabled
Info : Device: STM32L0xx (Cat.5)
Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
target halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x2000000e msp: 0x20000200
[... 321 lines cut ...]
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 7.0.0 (tags/RELEASE_700/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.12.1
               cppcheck: Cppcheck 1.84
                doxygen: 1.8.14
                 flake8: missing
                    git: git version 2.18.0
                openocd: Open On-Chip Debugger 0.10.0+dev-00226-g1c2e3d41d (2018-01-02-18:28)
                 python: Python 3.7.0
                python2: Python 2.7.15
                python3: Python 3.7.0
             coccinelle: missing
```

jia200x, 2018-10-19:
@aabadie could you confirm with your `b-l072z-lrwan1` board if this happens?

jcarrano, 2018-10-19:
I generated a bin using `arm-none-eabi-objcopy -O binary bin/b-l072z-lrwan1/lua_basic.elf lua_basic.bin`.

- The size of that binary is 145556 bytes (143k)
- The b-l072z-lrwan1 has a STM32L072CZ chip ([source](https://www.st.com/en/evaluation-tools/b-l072z-lrwan1.html))
- The [STM32L072CZ](https://www.st.com/resource/en/datasheet/stm32l072v8.pdf) has 196k flash (see p. 12)
- I'm getting a different output (but still a failure):
  - Yours:
     ```
     Info : Device: STM32L0xx (Cat.5)
     Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
     ```
  - Mine:
    ```
    Info : Device: STM32L0xx (Cat.5)
    Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
    Warn : couldn't use loader, falling back to page memory writes
    ```

I think there's something funny going on with the OpenOCD config. It may make sense to have a minimal fail case (i.e. no modules, just define a huge array in ROM and try to reproduce the issue).

jia200x, 2018-10-30:
Not critical for the Release. I remove it from the Project.

jcarrano, 2019-03-06:
@aabadie Has #10213 fixed this?

closed 2019-03-20 by cladmi