RIOT-OS/RIOT #10211: How to do performance test?

kbumsik, 2018-10-19:
#### Description

Hi, I came from a PR for porting a WiFi module, #7433. As this PR is getting close to the end, I would like to more extensive tests e.g. UDP bandwidth and latency, or like [the page 48 on Martine Lenders' master thesis](http://doc.riot-os.org/mlenders_msc.pdf).

I have to admit that I'm weak at testing. I'm wondering how the RIOT community do performance tests, not feature tests. Do you share any particular scripts to do that? Please let me know if there is any useful tools for performance tests.

Thank you in advance!

jia200x, 2018-10-24:
maybe @miri64 or @PeterKietzmann can give some hints?

PeterKietzmann, 2018-10-25:
AFAIK there is no public tool yet to do that kind of measurement but I think it's worth to add at least some of these measurement tests to our release specs.
The measurements you referenced from @miri64's M.A. thesis focus 802.15.4 radios and thus incorporate 6LoWPAN **but** don't include the device driver processing times or radio transmission. So, for comparisons with your WiFi module they won't be the right reference. Am I mistaken?

miri64, 2018-10-25:
You can find the performance tests from my master thesis [here](https://github.com/miri64/RIOT_playground/tree/master/stack_comparison), but since they are as old as GNRC and untouched since (oops), I don't think they still work. And as @PeterKietzmann already stated: Those tests are the stack *traversal* times, so from radio to application, so they won't help you with testing WiFi radios.

kbumsik, 2018-11-02:
Of course I know miri64's test is not for my WiFi driver. I just wanted to study what kinds of approaches are taken to do performance tests. Please consider including any benchmark/performance tests in the future specs. It would be useful information for RIOT users.

@miri64 Thank you for your link and paper BTW. Your explanation on gnrc stack (especially the sequence diagram) was very useful when I was implementing the WiFi driver :)

PeterKietzmann, 2018-11-05:
@kbumsik can we close this issue?

kbumsik, 2018-11-05:
@PeterKietzmann Yes. I close the issue now.

closed 2018-11-05 by kbumsik