RIOT-OS/RIOT #10224: Region `ROM' overflowed by xxxx bytes | WSN430

npcode15, 2018-10-23:
Hello, 

I am trying to compile a program for WSN430 board. But I am getting the following error:
```
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: /home/vagrant/Tutorials/experiments/UDP/server/bin/wsn430-v1_4/server_udp.elf section `.rodata' will not fit in region `rom'
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: section .vectors loaded at [000000000000ffe0,000000000000ffff] overlaps section .rodata loaded at [000000000000ea76,0000000000010853]
/usr/lib/gcc/msp430/4.6.3/../../../../msp430/bin/ld: region `rom' overflowed by 2188 bytes
/usr/lib/gcc/msp430/4.6.3/mmpy-16/libcrt0.a(_copy_data.o): In function `__do_copy_data':
/build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/gcc/config/msp430/crt0.S:195: relocation truncated to fit: R_MSP430_16_BYTE against symbol `__data_load_start' defined in *ABS* section in /home/vagrant/Tutorials/experiments/UDP/server/bin/wsn430-v1_4/server_udp.elf
```

I read on the web that it is possible to increase the CODE_SIZE in the setting which could solve this issue. I tried searching for the setting in ```RIOT/boards``` but I couldn't find one. 

Please suggest a solution for the issue. 

Command used to compile: ```BOARD=wsn430-v1_4 make DEFAULT_CHANNEL=11```

**NOTE: The same program compiles successfully for iotlab-m3 and native. This is the reason why I have not included the code used but if needed I will upload it to a repository and share it here.**

Thanks.

dylad, 2018-10-23:
MSP430F1611 MCU has only 48k of ROM while stm32f103re has 512k. If your program doesn't fit in 48k  there is nothing to do unless you remove some features to lighten your code.
What is the size of your .bin program ?

npcode15, 2018-10-23:
@dylad 

size of the '.hex' file is 180 KB  and '.elf' file is 1.7MB

**NOTE:** I was able to successfully compile a different application for WSN430. And the size of the build('.hex' file) for that program was 128KB.

dylad, 2018-10-23:
At the end of RIOT build, there is a notice about ROM/RAM usage of your program.
If your text+data size exceeds 48k then  you need to either remove features or switch to another MCU with a bigger flash.

miri64, 2018-10-23:
> NOTE: I was able to successfully compile a different application for WSN430. And the size of that build('.hex' file) was 128KB.

@npcode15 as @dylad said, if you use a slimmer set of features (and an application is nothing else than a set of features), then it will fit. Don't look at the `hex` file for information, as it contains meta-information that isn't actually flashed to the chip. The size information printed at the end of a successful build is far more accurate:

```
$ BOARD=wsn430-v1_4 make -C examples/hello-world/
[…]
   text	   data	    bss	    dec	    hex	filename
   4686	     16	   1046	   5748	   1674	examples/hello-world/bin/wsn430-v1_4/hello-world.elf
```

If you are unsure which features to remove / slim down have a look with

```
$ BOARD=wsn430-v1_4 make -C examples/hello-world/ info-objsize
```

npcode15, 2018-10-23:
@dylad @miri64 

I got the following when I compiled for ```iotlab-m3```.

``` 
text      data     bss       dec      hex filename
60596  152    16544   77292   12dec /home/vagrant/Tutorials/experiments/UDP/server/bin/iotlab-m3/server_udp.elf
```

miri64, 2018-10-23:
`shell` will also remove all the `sc_` objects, as they are implementing shell commands (I think you don't even have to remove `shell_commands` for that, but would be cleaner if you do ;-)), so that would be enough space. You can't remove `xtimer` as it is a dependency of GNRC's neighbor discovery (which you need for working IPv6 services). Another optimization that you can do is to set `DEVELHELP = 0`.

miri64, 2018-10-23:
Furthermore, keep in mind that `iotlab-m3` is not only much larger than `wsn430-v1_4b` when it comes to flash (ROM) size, but also, that the first is based on a 32-bit ARM microcontroller and the latter on a 16-bit MSP430 microcontroller, so the buildsizes will differ significantly (usually the binaries on MSP430 boards are smaller by around 70% or less).

miri64, 2018-10-23:
(so you are comparing apples to oranges if you put those two against each other ;-))

npcode15, 2018-10-23:
@miri64 thanks for all the info.

Succesfully Compiled.

```
text    data     bss     dec     hex filename
31128   22    5608   36758    8f96 /home/vagrant/Tutorials/experiments/UDP/server/bin/wsn430-v1_4/server_udp.elf
```

The only reason I kept shell module was to get the IP/MAC addresses for nodes.

miri64, 2018-10-23:
You can get those with the `gnrc_netif` module (just look how the shell command implementation does it) and print them in regular intervals.

miri64, 2018-10-23:
But that's a different issue, so I'm closing this one now ;-)

closed 2018-10-23 by miri64

hvd007-harsh, 2022-10-09:
If you use window press window + R and type %temp% and press "enter" and delete all files to free the builds made by ide