RIOT-OS/RIOT #10258: Incorrect default $PORT building for esp32-wroom-32 on macOS

snej, 2018-10-25:
#### Description
I'm trying to run tutorial "task-01" on a SparkFun ESP32 Thing.
At the start of the build I see a warning `no PORT set!`. Compilation succeeds, but at the flash phase the esptool fails with `error: argument --port/-p: expected one argument`. (See below for build log.)

I'm new to RIOT, but figured out that PORT should be set to the path to the serial device, and that on macOS its value comes from PORT_DARWIN. In boards/common/esp32/Makefile.include this is set as:
```make
PORT_DARWIN ?= $(firstword $(sort $(wildcard /dev/tty.SLAB_USBtoUART*)))
```
I believe this is incorrect; there is no matching device on my box. The README for esptool says that the device name usually begins with `tty.usbserial-` on macOS:

>The `--port` argument specifies the serial port. This may take the form of something like COMx (Windows), /dev/ttyUSBx (Linux) or /dev/tty.usbserial (OS X) or similar names.

On my 2014 iMac, with the cable plugged directly into the back of the computer, it's `/dev/tty.usbserial-DN041OIM`. If I prepend `PORT=/dev/tty.usbserial-DN041OIM` to the command line, the binary is successfully flashed to the board.

This suggests that the line in Makefile.include quoted above is wrong and `SLAB_USBtoUART` should be changed to `tty.usbserial`. (There are about a dozen other board definitions that set PORT_DARWIN this way.)

#### Steps to reproduce the issue
1. Check out RIOT tutorial repo
2. Update RIOT submodule to latest, to pick up ESP32 support
3. Install ESP32 toolchain and ESP-IDF and set environment vars accordingly
4. Connect ESP32 Thing to USB port
5. `cd task-01`
6. `BOARD=esp32-wroom-32 make all flash term`

#### Expected results
Tutorial builds, uploads to board, and runs, allowing me to interact with the CLI.

#### Actual results
```
$  BOARD=esp32-wroom-32 make all flash term
Warning: no PORT set!
Building application "Task01" for "esp32-wroom-32" with MCU "esp32".

"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Embedded/RIOT-OS/RIOT/boards/esp32-wroom-32
...
...
/Embedded/ESP32/esp-idf//components/esptool_py/esptool/esptool.py --chip esp32 -p  -b 460800 --before default_reset --after hard_reset write_flash -z -fm dout -fs detect -ff 40m    0x1000 /Embedded/RIOT-OS/RIOT/cpu/esp32/bin/bootloader.bin 0x8000 /Embedded/RIOT-OS/task-01/bin/esp32-wroom-32/partitions.bin 0x10000 /Embedded/RIOT-OS/task-01/bin/esp32-wroom-32/Task01.elf.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument --port/-p: expected one argument
make: *** [flash] Error 2
```

#### Versions

- Building on: macOS 10.13.6
- Toolchain: xtensa-esp32-elf-gcc-5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- RIOT: Latest master commit 25-Oct-18 (99ae4952)
- Hardware: SparkFun ESP32 Thing

snej, 2018-10-25:
Disclaimer: I'm a total noob at RIOT, although I've been working with ESP-IDF and mbed-os for a few months (and am an old hand at desktop/mobile development.)

miri64, 2018-10-25:
@gschorcht @cladmi can you have a look please?

kYc0o, 2018-10-25:
See #10262.

gschorcht, 2018-10-26:
It is not an ESP32-specific problem, but rather a more general problem. Since, I have no access to MAC OSX, I took this definition from any other board. I don't remember from which one.

It seems that the following boards use the same wrong definition:
```
./boards/airfy-beacon/Makefile.include
./boards/cc2650stk/Makefile.include
./boards/common/esp8266/Makefile.include
./boards/common/msb-430/Makefile.include
./boards/common/stm32f103c8/Makefile.include
./boards/common/wsn430/Makefile.include
./boards/f4vi1/Makefile.include
./boards/firefly/Makefile.include
./boards/limifrog-v1/Makefile.include
./boards/maple-mini/Makefile.include
./boards/mbed_lpc1768/Makefile.include
./boards/microbit/Makefile.include
./boards/msbiot/Makefile.include
./boards/nrf51dongle/Makefile.include
./boards/nrf6310/Makefile.include
./boards/nz32-sc151/Makefile.include
./boards/remote-reva/Makefile.include
./boards/remote-revb/Makefile.include
./boards/seeeduino_arch-pro/Makefile.include
./boards/spark-core/Makefile.include
./boards/stm32f0discovery/Makefile.include
./boards/stm32f3discovery/Makefile.include
./boards/stm32f429i-disc1/Makefile.include
./boards/stm32f4discovery/Makefile.include
./boards/stm32f769i-disco/Makefile.include
./boards/yunjia-nrf51822/Makefile.include
./boards/z1/Makefile.include
```

gschorcht, 2018-10-26:
PR #10262 (thanks to @kYc0o) should solve the problem for ESP32. Unfortunatly, I can't test it since I don't have access to MAC OSX. Who has an ESP32 board and could test flashing with a MAC. The board definition should does not matter for the test.

gschorcht, 2018-10-27:
@kYc0o Should we leave this issue open until it is solved for all boards or should we create a new issue?

kYc0o, 2018-10-27:
I'd say we solve this for now, I'll try to create/extend documentation for porting new boards using the good schema. Anyways I don't know if lots of boards are wrong, the `/dev/tty.SLAB_USBtoUART` device is ok for boards embedding a Silabs USB to UART converter, which is the case on some ESP based boards.

gschorcht, 2018-10-27:
Does it mean that SiliconLabs CP2102 and CP2104 would result in a name like ```/dev/tty.SLAB_USBtoUART*```? If that is the case, we shouldn't make the change in ```boards/common/esp32/Makefile.include``` since all ESP32 DEVKIT clones (board ```esp32-wroom-32```) probably use the Silicon Labs USB to UART converter.

The boards I have, use the following converters:

Board | Converter
-------- | ------
DEVKITV1 | Silicon Labs CP2102
NoName WROOM-32 | Silicon Labs CP2102
MH-ET Live MiniKit | Silicon Labs CP2104
Olimex ESP32-EVB | QinHeng Electronics CH341
Epressif WROVER-KIT V3 | FTDI FTDI2232H

Maybe we should leave ```PORT_DARWIN``` in ```boards/common/esp32/Makefile.include``` as it is and overwrite it only in ```Makefile.include``` for boards that are using a different converter, probably for example for Espressif WROVER-KIT. The question is how CH341 converter devices are named under MAC OSX.

If we had a board definition for Sparkfun ESP32 Thing, the problem of @snej would be solved too.

gschorcht, 2018-10-27:
BTW, the Sparkfun ESP32 Thing is not an ESP32-WROOM-32 board and should therefore not be used as reference for ```esp32-wroom-32``` board definitions.

gschorcht, 2018-10-27:
@kYc0o I spent some more time to search for ESP32-WROOM-32 boards. Here are the results:

Board | Converter
------- | ------
GeeekNET ESP32 Development Board | SiliconLabs CP2102
Adafruit HUZZAH32 - ESP32 Feather Board | SiliconLabs CP2104
EzSBC ESP32-01 | SiliconLabs CP2104
Watterott ESP-WROOM32-Breakout | SiliconLabs CP2102
DoIT ESPduino32 | CH340C
ESP32 Test Fixture | SiliconLabs CP2102

The hundreds of no-name ESP32 DEVKIT clones also use SiliconLabs CP2102.

snej, 2018-10-27:
To confuse things further, the ESP-IDF docs [say](
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/establish-serial-connection.html#check-port-on-linux-and-macos) that on Mac the serial port name starts with `cu-`.

kYc0o, 2018-10-29:
Actually `/dev/cu*` is a kind of alternative serial device that might be used while `/dev/tty*` is busy: https://stackoverflow.com/questions/8632586/macos-whats-the-difference-between-dev-tty-and-dev-cu 

I propose we stick to the `/dev/tty*`.

For the boards supplying other kind of serial converters than Silabs, we need to open PRs to add them to boards/, so they can have their own particularities expressed there.

gschorcht, 2018-10-29:
@kYc0o Since Darwin is BSD based, I tried device naming under FreeBSD within a virtual maschine. It seems that ```/dev/cuaU*``` and ```/dev/ttyU*``` are created independent on whether the USB serial converter is from Silicon Labs, FTDI or anything else.

 it's the same on MAC OSX, why do not we just always use ```/dev/ttyU*```?

miri64, 2020-07-02:
Ping @gschorcht?

gschorcht, 2020-07-02:
Unfortunately, I don't have access to a MAC. So I can neither test nor investigate it. Therefore, I would rely on what @smlng in  https://github.com/RIOT-OS/RIOT/issues/10258#issuecomment-456310766 found out and follow his suggestion to close this issue.

Probably, it is nothing we can solve this issue for any board and any use case. Finally, `PORT_DARWIN` only defines the default value for `PORT` for MAC OS. Whenever another USB2UART chip is used on a board or even if not the first interface is used, the 'PORT' environment variable must be set anyway.

closed 2022-09-19 by maribu