RIOT-OS/RIOT #10259: tests/leds doesn't work, on ESP32 Thing

snej, 2018-10-25:
#### Description
I'm trying to get RIOT-OS to do anything on my ESP32 Thing board. I can't get any terminal I/O from the tutorial apps, so I dropped down to a lower level LED test.

When I try to run `tests/leds`, the build and flashing appear to succeed, but the [blue LED](https://cdn.sparkfun.com/assets/learn_tutorials/5/0/7/top-annotated.jpg) (attached to pin 5) just stays on without blinking.

#### Steps to reproduce the issue
1. Check out latest master branch of RIOT
2. Install ESP32 toolchain and ESP-IDF and set environment vars accordingly
3. Connect ESP32 Thing to USB port
4. `cd tests/leds`
5. `BOARD=esp32-wroom-32 PORT=/dev/tty.usbserial-DN041OIM make all flash term`

#### Expected results
Blue LED blinks

#### Actual results
Blue LED stays steady

Here's the build output, following the compile/link stages:
```
...
/Embedded/ESP32/esp-idf//components/esptool_py/esptool/esptool.py --chip esp32 elf2image -fm dout -fs 16m -ff 40m    -o /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/tests_leds.elf.bin /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/tests_leds.elf; echo "" > /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv; echo "nvs, data, nvs, 0x9000, 0x6000" >> /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv; echo "phy_init, data, phy, 0xf000, 0x1000" >> /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv; echo -n "factory, app, factory, 0x10000, " >> /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv; ls -l /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/tests_leds.elf.bin | awk '{ print $5 }' >> /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv; python /Embedded/RIOT-OS/RIOT/cpu/esp32/gen_esp32part.py --disable-sha256sum --verify /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.csv /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.bin
WARNING: Flash size arguments in megabits like '16m' are deprecated.
Please use the equivalent size '2MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.3.1
Parsing CSV input...
/Embedded/ESP32/esp-idf//components/esptool_py/esptool/esptool.py --chip esp32 -p /dev/tty.usbserial-DN041OIM -b 460800 --before default_reset --after hard_reset write_flash -z -fm dout -fs detect -ff 40m    0x1000 /Embedded/RIOT-OS/RIOT/cpu/esp32/bin/bootloader.bin 0x8000 /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/partitions.bin 0x10000 /Embedded/RIOT-OS/RIOT/tests/leds/bin/esp32-wroom-32/tests_leds.elf.bin
esptool.py v2.3.1
Connecting.......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0320
Compressed 20608 bytes to 12166...
Wrote 20608 bytes (12166 compressed) at 0x00001000 in 0.3 seconds (effective 572.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 85...
Wrote 3072 bytes (85 compressed) at 0x00008000 in 0.0 seconds (effective 1511.0 kbit/s)...
Hash of data verified.
Compressed 83664 bytes to 36541...
Wrote 83664 bytes (36541 compressed) at 0x00010000 in 1.1 seconds (effective 597.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
/Embedded/RIOT-OS/RIOT/dist/tools/pyterm/pyterm -p "/dev/tty.usbserial-DN041OIM" -b "115200"
No handlers could be found for logger "root"
2018-10-25 11:53:26,843 - INFO # Connect to serial port /dev/tty.usbserial-DN041OIM
Welcome to pyterm!
Type '/exit' to exit.
```

#### Versions
* Building on: macOS 10.13.6
* Toolchain: xtensa-esp32-elf-gcc-5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
* RIOT: Latest master commit 25-Oct-18 (99ae495)
* Hardware: SparkFun ESP32 Thing

snej, 2018-10-25:
A few more notes:
* The LED does turn off twice briefly during the flashing, presumably as the CPU is reset.
* If I press the reset button, the LED turns off till I release it, then comes back on and stays on.
* After releasing Reset, a bunch of garbage is written to pyterm, followed by:
```
  �lbĒ���b#`lߒb�n���IA AO�Ȓ``��l�l����	���l�#�I�;ets Jun  8 2016 00:22:57
2018-10-25 11:57:02,899 - INFO #
2018-10-25 11:57:02,900 - INFO # rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
2018-10-25 11:57:02,900 - INFO # configsip: 0, SPIWP:0xee
2018-10-25 11:57:02,902 - INFO # clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
2018-10-25 11:57:02,905 - INFO # mode:DOUT, clock div:2
2018-10-25 11:57:02,906 - INFO # load:0x3fff0018,len:4
2018-10-25 11:57:02,906 - INFO # load:0x3fff001c,len:5632
2018-10-25 11:57:02,911 - INFO # ho 0 tail 12 room 4
2018-10-25 11:57:02,911 - INFO # load:0x40078000,len:0
2018-10-25 11:57:02,916 - INFO # load:0x40078000,len:14868
2018-10-25 11:57:02,925 - INFO # ���xn�[�O����������)�^~
```
(exact output varies on each attempt.) Something seems to be wrong with the serial connection? Interesting that I never see this until I manually reset the board.

gschorcht, 2018-10-26:
@snej I have no Sparkfun ESP32 Thing here for testing. Even though ```esp32-wroom-32``` is a quite generic board definition, it does not define any LED pin. So I guess, you modified ```boards/esp32-wroom-32/board.h``` as following, right?
```
#define LED0_PIN        GPIO5
#define LED0_ACTIVE     1       /**< LED is high active */
```

What is the board configuration printed out at the terminal after the reset?

Concerning the garbage after manual reset. Could you try to use following to check whether it is a problem with pyterm and the synchronisation when UART data rate changes during the reset?
```
BOARD=esp32-wroom-32 PORT=/dev/tty.usbserial-DN041OIM make all flash
python -m serial.tools.miniterm /dev/tty.usbserial-DN041OIM 115200
```
Unfortunatly, I have no MAC to try it by myself.

gschorcht, 2018-10-26:
You reported that your LED is staying steady. Since the the default mode of ESP's GPIO5 is GPIO_IN_PU the LED could be always active when the GPIO is not initialized for any purpose.

It seems that you didn't define the LED pin. The ```esp32-wroom-32``` board definition doesn't define any LED pin. Thus, the default board definition has to be extended or overridden when you use it for another board.

Making ```tests/leds``` with the command
```
CFLAGS="-DLED0_PIN=GPIO5 -DLED0_ACTIVE=1" make flash BOARD=esp32-wroom-32 -C tests/leds
```
works for me on an ESP32 DevKit. GPIO5 output is alternating as expected.

Please take care when you try to use SPI with the  ```esp32-wroom-32``` board definition. It uses GPIO5 as the default CS signal for SPI_DEV(0). This would collide with the LED pin if you don't adopt the board definition.

gschorcht, 2018-10-26:
@snej Concerning the garbage: This garbage comes from or happens in bootloader which is not part of RIOT-OS. It uses as binary from ESP-IDF. What is the first correct output after this garbage?

snej, 2018-10-26:
After setting those CFLAGS, the LED blinks as promised. Thanks! Makes sense that the LED would be board-specific. So now that I know the program is running, the problem becomes how to see its output in pyterm...

I don't see any "Board configuration" logged after a reset. The output I [pasted above](
https://github.com/RIOT-OS/RIOT/issues/10259#issuecomment-433167662) is _everything_ that appears in pyterm, after the initial binary garbage. It does look as though it gets cut off at the end, after the `load:0x40078000,len:14868` line, and reverts back to binary.

I'll run with stdout redirected to a file and attach the file, in case you can get any information from the binary: 
[esp32_thing_log.zip](https://github.com/RIOT-OS/RIOT/files/2520049/esp32_thing_log.zip)
(Interestingly, in that file there is nothing after the `load:0x40078000,len:14868` line... some side effect of pyterm not writing to a TTY?)

snej, 2018-10-26:
> `python -m serial.tools.miniterm /dev/tty.usbserial-DN041OIM 115200`

That was no better ... but slightly different, in that (after I pressed RST) there was no garbage before the "`ets Jun  8 2016 00:22:57`..." but more garbage after "`load:0x40078000,len:14868`". 🤔

(I don't know if you'd rather close this bug since the LED test is working now, and have me open a new one on the console problems? I'll do so if you want.)

I just remembered something that may be a clue: Back when I first got the ESP32 Thing, I had some trouble getting terminal output from ESP-IDF, but their docs [had a workaround](
https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html#monitor):

>If instead of the messages above, you see a random garbage [...] your board is likely using 26MHz crystal. Most development board designs use 40MHz and the ESP-IDF uses this default value. Exit the monitor, go back to the menuconfig, change CONFIG_ESP32_XTAL_FREQ_SEL to 26MHz, then build and flash the application again.

This solved the problem for me. I assume the ESP32 side of the serial connection is using timing loops that are specific to the clock frequency? If you can point me to the equivalent clock configuration in the RIOT code, I can try making the same change.

snej, 2018-10-26:
I'm tracking down what this SDKCONFIG setting does in IDF ...
1. kconfig uses it to set the value of the preprocessor symbol `CONFIG_ESP32_XTAL_FREQ`
2. This constant is stored into `clk_cfg.xtal_freq` in `bootloader_clock_configure()` (bootloader_clock.c), which then passes that struct to `rtc_clk_init()` in rtc_clk.c, which configures the hardware.
3. `bootloader_clock_configure()` is called from `esp_clk_init()` in clk.c ... but only if `CONFIG_COMPATIBLE_PRE_V2_1_BOOTLOADERS` is defined, and then only if `rtc_clk_xtal_freq_get()` reports that the crystal freq hasn't been stored in a hw register yet.

And hey, RIOT has some of these source files in cpu/esp32/vendor/esp-idf/esp32 ... but it doesn't have bootloader_clock.c, so there is nothing that calls `rtc_clk_init()`.

So it looks as though the problem is that my board (perhaps all ESP32 Things?) have a pre-v2.1 bootloader which does not initialize the hardware register that stores the crystal frequency, and so RIOT assumes the default frequency of 40MHz.

snej, 2018-10-26:
I figured out where to splice in the `bootloader_clock_configure()` call — in `system_clk_init()` — and with the crystal freq set to 26, I now get RIOT console output in pyterm! 🎉

I'll try to clean this up a bit, fork RIOT, and submit a PR later today.

snej, 2018-10-26:
PR is #10273.

gschorcht, 2018-10-27:
@snej

> I figured out where to splice in the bootloader_clock_configure() call — in system_clk_init() — and with the crystal freq set to 26, I now get RIOT console output in pyterm!

That's great. Many thanks for investigating the problem in detail.

All the ESP32 boards I have (5 different boards), don't use the ESP32 SoC directly but rather either the WROOM-32 or the WROVER module which all have a 40 MHz crystal. The Sparcfun ESP32 Thing is the first board I saw that uses the ESP32 SoC and according to its schematic a 26 MHz. Thus, I thought all ESP32 board have a 40 MHz crystal and it wasn't clear to me that 26 MHz crystals are still an possible option.

I will review your PR immediatly.

gschorcht, 2018-11-06:
I close the issue for the following reasons:
- The problem originally described and used as title was not a bug, but rather an incorrect use of the board configuration.
- The boot frequency problem described in further explanations could be solved by an enhancement with PR # 10273.

closed 2018-11-06 by gschorcht