RIOT-OS/RIOT #10269: Undefined i2c behaviour after release

MrKevinWeiss, 2018-10-26:
Currently the i2c_acquire and release functions are not consistant between boards.

Should we have access to the bus on startup without i2c_acquire?

Should we be able to access the bus after an i2c_acquire and release?

Currently most boards support access to the bus without i2c_acquire first as it gets initialized in the cpu init.
On the nucleo-l073rz (and I assume other nucleo boards) the bus cannot be accessed after the bus has been acquired and released.  This is not true for the arduinomega-2560, nrf52dk, nucleo-f411re, samr21-xpro.  

Maybe it should work, if so how does this affect the pm layer?

gebart, 2018-10-26:
i2c_acquire, i2c_release are locking functions to ensure exclusive access to the bus when there is more than one device driver using the i2c bus. Therefore, i2c_acquire must _always_ be called before using any of the i2c_transfer_x functions.
If you find a device driver which uses i2c_transfer without first acquiring the bus, then that is a bug.

I don't understand this part:
>On the nucleo-l073rz (and I assume other nucleo boards) the bus cannot be accessed after the bus has been acquired and released. This is not true for the arduinomega-2560, nrf52dk, nucleo-f411re, samr21-xpro.

Does the nucleo-l073rz i2c stop working after the first i2c_release call? That would be a blocker bug IMO, and must be fixed.

MrKevinWeiss, 2018-11-05:
@gebart Sorry for not responding.  To clarify, 

> Does the nucleo-l073rz i2c stop working after the first i2c_release call?

Yes, as expected.  From what you described, after being released the bus should **not** work.  All other boards continue to work.

@dylad I did see that in some places but it seems that it gets enabled in the cpu init in many boards.

So from what I understand, the correct operation should be as follows
- Startup the i2c should **not** work
- Acquire i2c should work
- Release i2c should **not** work

If we all agree I will add these as critical tests to the HiL testing and we can start enforcing this rule.  I may even start a tracker to ensure this is happening everywhere.

@aabadie also has experience with this and it would be nice to get his feedback :)

gebart, 2018-11-05:
Thanks for clarifying. I originally understood it as "no more i2c_acquire calls an be made after the first i2c_release call (bug)", but you meant that no i2c transfers can be made after a call to i2c_release _unless_ a call to i2c_acquire is made first (enforcing correct application behavior), is that the correct interpretation?

MrKevinWeiss, 2018-11-05:
@gebart Correct

MrKevinWeiss, 2018-11-06:
Also I just found out that if you try to use the i2c with the frdm-k22f without first acquiring I get:
```
2018-11-06 11:31:32,921 - INFO # i2c_read_regs 0 85 152 3 0
2018-11-06 11:31:32,935 - INFO # Command: i2c_read_regs(0, 0x55, 0x98, 3, 0x00)
2018-11-06 11:31:32,937 - INFO # 
2018-11-06 11:31:32,954 - INFO # Context before hardfaultmain(): This is RIOT! 
```
...I don't know if this is how it should be.  It works fine if it has been acquired first.

MrKevinWeiss, 2018-11-22:
After talking to @haukepetersen it seems the conclusion is to not define what happens if calls are made to the i2c bus if not acquired as to not increase code size.  The documentation of this is currently

>  * // before accessing the bus, we need to acquire it

in the example in the drivers/include/periph/i2c.h

I would maybe suggest to add in each call a `@note` or `@requirement` stating that the bus must be acquired.

MrKevinWeiss, 2018-11-28:
Fixed with #10490

closed 2018-11-28 by MrKevinWeiss

bugagator, 2020-04-14:
I've run into similar issue with bluepill board and ds3231 slave device.

I use acqure-transfer-release sequences, but only first of them succeeds. The second transfer always fails with timeout.

After some investigation I've noticed two things:

1. Before first call to i2c_acquire clk and dio lines are both in high state. Releasing bus turns these lines low. The overall bus behaviour is inconsistent here.

2. Lowering the bus after release switches slave device into some weird mode, and reenabling bus with acquire is not sufficient to recover it.

I would prefer to turn off the bus disabling logic, and leave bus high after release. At least as an option.

MrKevinWeiss, 2020-04-14:
Having the bus high after release should be enforced. We should clarify that. The undefined behavior is more a question if the bus is enabled.

I noticed you say you are using the bluepill board as your master for I2C communication. It just so happens that the STM32F1xx series seems to not be able to provide an internal pullup, are you using pullup resistors?

Another thing which seems strange if pullups are there is that the release would be forcing the lines to ground. The release should at least make them inputs if anything.

bugagator, 2020-04-14:
Does explicit bus initialization required?

Pullups are used. Actually, the ds3231 is on a zs-042 module with all required components. I2C bus is on B6/B7 pins of STM32.

Clock goes down after periph_clk_dis call.

Minimal example to reproduce the issue:

```
#include <stdint.h>
#include "periph/i2c.h"
#include "board.h"
#include "xtimer.h"
int main(void) {
	uint8_t regs[19];
	int res;
	i2c_acquire(I2C_DEV(0));
    	res = i2c_read_regs(I2C_DEV(0), 0x68, 0, &regs, sizeof(regs), 0);
	(void)res; // res = 0
	i2c_release(I2C_DEV(0));
        // clk = 0 here
	xtimer_usleep(500000);
	i2c_acquire(I2C_DEV(0));
    	res = i2c_read_regs(I2C_DEV(0), 0x68, 0, &regs, sizeof(regs), 0);
	(void)res; // res = -116
	i2c_release(I2C_DEV(0));
	while (1);
	return 0;
}
```

```
APPLICATION = fail
BOARD ?= bluepill
QUIET ?= 1
CFLAGS_OPT ?= -O0
FEATURES_REQUIRED += periph_i2c
USEMODULE += xtimer stdio_null
PROGRAMMER ?= bmp
RIOTBASE ?= $(realpath $(CURDIR)/RIOT)
include $(RIOTBASE)/Makefile.include

print-%  : ; @echo $* = $($*)
```

MrKevinWeiss, 2020-04-15:
OK I will add a test to check the SDA and SCL lines after initialization and get back to you once I have the results.

MrKevinWeiss, 2020-04-15:
I am having some trouble reproducing the issue on my board (or any boards for that matter).

What happens if the only acquire and release are used and no reading of registers occur?

Could you also try adding a sleep before the release?

bugagator, 2020-04-15:
Only acquire and release turns the bus off. As well as `periph_clk_dis(i2c_config[0].bus,i2c_config[0].rcc_mask);` alone.

Sleeping do not help.

I also noticed with logic analyzer, that release makes bus completely disfunctional. Acquire restores high state of lines, but no clock and data signals present on the wires dirung read.

MrKevinWeiss, 2020-04-15:
OK, I will try to reproduce it tomorrow. I only have the bluepill and not the ds3212 though.  I can somewhat simulate the device.

Does this occur on master?
Can you toggle the pin somehow after release?

[This](https://ci.riot-os.org/hil/blue/organizations/jenkins/RIOT-HIL/detail/PR-62/2/pipeline/519) shows the `Pins High After Release Should Succeed` test passing for the nucleo-f103rb which is a similar CPU as the bluepill.  The only difference is the alternate function periph clock that must be enabled starting [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/periph/i2c_2.c#L115).

Maybe if you try adding [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/periph/i2c_2.c#L199) a `gpio_set` for the pins it would help.

```c

void i2c_release(i2c_t dev)
{
    assert(dev < I2C_NUMOF);
    gpio_set(i2c_config[dev].sda_pin);
    gpio_set(i2c_config[dev].scl_pin);
    periph_clk_dis(i2c_config[dev].bus, i2c_config[dev].rcc_mask);

#ifdef STM32_PM_STOP
    /* unblock STOP mode */
    pm_unblock(STM32_PM_STOP);
#endif

    mutex_unlock(&locks[dev]);
}
```

Still strange that I don't see it on my setup.

MrKevinWeiss, 2020-04-16:
[image]

It seems like it is raised for my setup.

[image]

bugagator, 2020-04-16:
Thanks for checking this. Probably, my device is a fake.

BTW, pins could not be toggled after release, but I can switch them explicitly into input mode to set high level, and I have to switch them back after release.

This basically works, except I get some extra impulses during gpio mode switching.