RIOT-OS/RIOT #10287: docker: /etc/localtime does not exist on OS X

emmanuelsearch, 2018-10-29:
#### Description
build with Docker on OS X fails because on OS X, /etc/localtime does not exist.

#### Steps to reproduce the issue
building HelloWorld with docker:

```
$ BUILD_IN_DOCKER=1 BOARD=samr21-xpro make -C examples/hello-world/ 
Warning: no PORT set!
Launching build container using image "riot/riotbuild:latest".
docker run --rm -t -u "$(id -u)" \
	    -v '/Users/emmanuel/Documents/work/CODING/RIOT:/data/riotbuild/riotbase' \
	    -v '/Users/emmanuel/Documents/work/CODING/RIOT/cpu:/data/riotbuild/riotcpu' \
	    -v '/Users/emmanuel/Documents/work/CODING/RIOT/boards:/data/riotbuild/riotboard' \
	    -v '/Users/emmanuel/Documents/work/CODING/RIOT/makefiles:/data/riotbuild/riotmake' \
	    -v '/Users/emmanuel/Documents/work/CODING/RIOT:/data/riotbuild/riotproject' \
	    -v /etc/localtime:/etc/localtime:ro \
	    -e 'RIOTBASE=/data/riotbuild/riotbase' \
	    -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' \
	    -e 'RIOTCPU=/data/riotbuild/riotcpu' \
	    -e 'RIOTBOARD=/data/riotbuild/riotboard' \
	    -e 'RIOTMAKE=/data/riotbuild/riotmake' \
	    -e 'RIOTPROJECT=/data/riotbuild/riotproject' \
	      \
	    -e 'BOARD=samr21-xpro' \
	    -w '/data/riotbuild/riotproject/examples/hello-world/' \
	    'riot/riotbuild:latest' make  
docker: Error response from daemon: Mounts denied: 
The path /etc/localtime
is not shared from OS X and is not known to Docker.
You can configure shared paths from Docker -> Preferences... -> File Sharing.
See https://docs.docker.com/docker-for-mac/osxfs/#namespaces for more info.
.
ERRO[0000] error waiting for container: context canceled 
make: *** [..in-docker-container] Error 125
```

#### Expected results
Build succeeds.

#### Actual results
Build fails.

#### Versions
RIOT master, on OS X 10.14, with Docker 18.06.1-ce.

emmanuelsearch, 2018-10-29:
seems related to this [issue](https://github.com/tomochain/tomochain-testnet/issues/2).

emmanuelsearch, 2018-10-29:
indeed, on OS X in my setup, the build succeed if you remove `-v /etc/localtime:/etc/localtime:ro \` from the docker makefile `riot/makefiles/docker.inc.mk` (line 107).

emmanuelsearch, 2018-10-29:
I'm rather ignorant in that matter: do we need localtime here?

gebart, 2018-10-29:
IIRC, without the link for /etc/localtime the time zone in the container is fixed at the default setting for the container and may not match the host time zone. The result is that we get different file modification times, which will confuse make.

emmanuelsearch, 2018-10-29:
OK. On this [thread](https://github.com/docker/for-mac/issues/2396) they discuss workarounds.
But none seem fantastic for our use case (or maybe I missed something?).

Do we care about solving the issue on OS X?
Should we update our [wiki page on docker](https://github.com/RIOT-OS/RIOT/wiki/Use-Docker-to-build-RIOT) with a workaround for this issue?

jia200x, 2018-10-31:
I'm facing exactly the same issue now

gebart, 2018-10-31:
How about using something like this? 
```make
$(if $(wildcard /etc/localtime), -v /etc/localtime:/etc/localtime,)
```

cladmi, 2018-11-05:
As the target file should exist I do not think that `wildcard` would help..

However this one could be a  workaround https://github.com/docker/for-mac/issues/2396#issuecomment-358090212

He does not really clear says what to mount where but could be worth trying one of these:

```
-v /private/etc/localtime:/etc/localtime
```
or
```
-v /private/etc/localtime:/private/etc/localtime
```

This one could also a workaround without touching files https://github.com/docker/for-mac/issues/2396#issuecomment-358150983

gebart, 2018-11-05:
@cladmi why shouldn't it work? the wildcard function removes any non existing files from the input list, even without any wildcard characters. The if checks if the input argument is non empty.

cladmi, 2018-11-05:
@gebart from what I understand from the external links, the issue is not that the file does not exist, but that it cannot be shared. Also in the output there is

```
The path /etc/localtime
is not shared from OS X and is not known to Docker.
```

But indeed that would mean the issue title is wrong and would need to be updated.

@emmanuelsearch can you confirm if you have a `/etc/localtime` file or not ?

gebart, 2018-11-05:
Ah, sorry, I should have read the detailed description better.

emmanuelsearch, 2018-11-06:
@cladmi yes I confirm /etc/localtime exists. Somehow I missed it last time I visited /etc let me update the issue title.

PR #10568 (2018-12-07): makefiles/docker: fix mounting localtime on OSX

### Contribution description

Mounting `/etc/localtime` directly does not work on macOS (anymore).
However, by resolving the symlink to its real path docker can handle
the mount on macOS again - should also work with Linux.

### Testing procedure

Try to run `BUILD_IN_DOCKER=1 make -C tests/minimal` on macOS, which will fail with one or the other error regarding either that `/etc` is not shared and hence not accessible by docker, or if you do add it to the list of shared folders, it will fail with mounting `/etc/localtime` into docker. With this [... cut]
closed 2018-12-07 by jcarrano