RIOT-OS/RIOT #10294: tests/leds doesn't work on Nucleo-F401RE

part1zano, 2018-10-29:
#### Description

Built tests/leds for `nucleo-f401re`, the LED doesn't blink.
#### Steps to reproduce the issue

#### Expected results

The LED blinks.
#### Actual results

The LED doesn't blink
#### Versions

I suppose that's caused by the fact the LED conflicts with SPI, but I don't know how to disable SPI.

aabadie, 2018-10-29:
If you want to have the LED blink on most of the nucleo 64 boards (including the nucleo-f401re), you have to define the AUTO_INIT_LED0 macro:
```
CFLAGS=-DAUTO_INIT_LED0 make BOARD=nucleo-f401re -C tests/leds flash term
```

aabadie, 2018-10-29:
For completeness, see [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/nucleo/board.c#L41) and [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/common/nucleo64/include/gpio_params.h#L35) (when using saul).

part1zano, 2018-10-29:
I suddenly found out the core issue here:
```(gdb) file tests_leds.elf 
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from tests_leds.elf...done.
(gdb) load
Loading section .text, size 0x2254 lma 0x8000000
Loading section .relocate, size 0x8c lma 0x8002254
Start address 0x80004c8, load size 8928
Transfer rate: 7 KB/sec, 4464 bytes/write.
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d3e in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:225
225         while (!(RCC->CR & RCC_CR_PLLRDY)) {}
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d42 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:225
225         while (!(RCC->CR & RCC_CR_PLLRDY)) {}
(gdb) 
```
How do I deal with this?

part1zano, 2018-10-29:
So the point is, it's not my fault: the program hangs forever trying to set up clocks.

aabadie, 2018-10-29:
This looks strange. What revision of nucleo-f401re are you using ?

Maybe you could try to change the value of CLOCK_HSE and CLOCK_LSE in the board clock configuration. See [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-f401re/include/periph_conf.h#L42) and [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-f401re/include/periph_conf.h#L45).
Can you try to set both CLOCK_HSE and CLOCK_LSE to 0 ?

part1zano, 2018-10-29:
Changed, no significant effect:
```
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:225
225         while (!(RCC->CR & RCC_CR_PLLRDY)) {}
(gdb) 
```

part1zano, 2018-10-29:
It now hangs at another line of code.

part1zano, 2018-10-29:
However, when I only set the latter to 1, I get the following:
```
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000802 in send_byte (byte=10 '\n', uart=0) at /usr/home/che/mcu/RIOT/cpu/stm32_common/periph/uart.c:129
129         while (!(dev(uart)->SR & USART_SR_TXE)) {}
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000802 in send_byte (byte=10 '\n', uart=0) at /usr/home/che/mcu/RIOT/cpu/stm32_common/periph/uart.c:129
129         while (!(dev(uart)->SR & USART_SR_TXE)) {}
(gdb) 
```

part1zano, 2018-10-29:
And when I set `CLOCK_LSE` to 1, I get the following:
```
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000ff4 in _puts_r ()
```

part1zano, 2018-10-29:
So maybe I could set `CLOCK_HSE` to something more appropriate?

part1zano, 2018-10-29:
No. Didn't help:
```
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000ff4 in _puts_r ()
(gdb) 
```

part1zano, 2018-10-29:
Besides, how do I define the revision for the board?
It says something like `MBT1136 C-1` and `MBT1136 rev C` on the bottom of the reverse side of the board

kaspar030, 2018-10-30:
> It says something like `MB1136 C-1` and `MB1136 rev C` on the bottom of the reverse side of the board

I just checked current master with a nucleo-f401re. My board says "MB1136 rev C" on the top, on the bottom it says "MB1136 C-04". With ```CFLAGS=-DAUTO_INIT_LED0``` the LED blinks just fine.
Can you check the crystals on your board? Mine does not have "x1" (below the MCU). It uses the one on the st-link part (x1).

part1zano, 2018-10-30:
In my case, X1 is to the left of the ST-Link MCU; the crystal below the STM32F4 is called X3, and it's absent.

kaspar030, 2018-10-30:
> the crystal below the STM32F4 is called X3

I had a typo in the relevant information. that one is called X3 here, too. sorry.

aabadie, 2018-10-30:
Just in case, are you building with arm-node-eabi-gcc 7.2 (or 7.3) ? (e.g what's our toolchain version?)

part1zano, 2018-10-30:
Yes:
```
% pkg info -ix gcc-arm-embedded
gcc-arm-embedded-7.3.20180627
```

part1zano, 2018-11-21:
Unfortunately, that didn't help. Still hangs at the same line (see above):
```

0x08000d34 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:204
204         while (!(RCC->CR & RCC_CR_HSERDY)) {}
```

part1zano, 2018-11-21:
To be a bit more clear, here's the full output with the applied patch:
``` 
(gdb) file tests_leds.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from tests_leds.elf...done.
(gdb) load
Loading section .text, size 0x2260 lma 0x8000000
Loading section .relocate, size 0x8c lma 0x8002260
Start address 0x80004c8, load size 8940
Transfer rate: 6 KB/sec, 4470 bytes/write.
(gdb) continue
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d32 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:183
183         FLASH->ACR = FLASH_ACR_CONFIG;
(gdb) continue
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d32 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:183
183         FLASH->ACR = FLASH_ACR_CONFIG;
(gdb) ^CQuit
(gdb) quit
A debugging session is active.

        Inferior 1 [Remote target] will be killed.

Quit anyway? (y or n) y
[0:11:54] 0 [che@xeon ~/mcu/RIOT/tests/leds/bin/nucleo-f401re]% arm-none-eabi-gdb
GNU gdb (GNU Tools for ARM Embedded Processors) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-unknown-freebsd11.1 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) tar remote :4242
Remote debugging using :4242
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x080004c8 in ?? ()
(gdb) file tests_leds.elf 
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Reading symbols from tests_leds.elf...done.
(gdb) load
Loading section .text, size 0x2260 lma 0x8000000
Loading section .relocate, size 0x8c lma 0x8002260
Start address 0x80004c8, load size 8940
Transfer rate: 6 KB/sec, 4470 bytes/write.
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d46 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:204
204         while (!(RCC->CR & RCC_CR_HSERDY)) {}
(gdb) continue 
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x08000d48 in stmclk_init_sysclk () at /usr/home/che/mcu/RIOT/cpu/stm32_common/stmclk.c:204
204         while (!(RCC->CR & RCC_CR_HSERDY)) {}
```

PeterKietzmann, 2019-02-27:
#11066 could fix this issue

PeterKietzmann, 2019-03-05:
Closing after #10322 and #11066 got merged

closed 2019-03-05 by PeterKietzmann