RIOT-OS/RIOT #10316: tests/emb6: wrong payload on udp send

jia200x, 2018-11-02:
#### Description

While running the Release interop tests I discovered there's a problem in the implementation of the udp send command in the emb6 test:

GNRC sending to emb6
```
1541163424.179446;m3-101;> udp send fe80::1711:6b10:65fc:5c22 80 abcdefgh
1541163424.181546;m3-101;Success: sent 8 byte(s) to [fe80::1711:6b10:65fc:5c22]:80
1541163424.182760;m3-100;Received from [c1d:20:a429:20:800:0:5000:0]:80:
1541163424.183683;m3-100;00000000  61  62  63  64  65  66  67  68
```

emb6 sending to GNRC
```
 1541163434.433180;m3-100;udp send fe80::1711:6b10:65f6:5d02 80 abcdefgh
1541163434.435101;m3-100;Success: send 3 byte to [fe80::1711:6b10:65f6:5d02]:80)
1541163434.436805;m3-101;> PKTDUMP: data received:
1541163434.437796;m3-101;~~ SNIP  0 - size:   3 byte, type: NETTYPE_UNDEF (0)
1541163434.437988;m3-101;00000000  AB  CD  EF
```
`emb6` is not encoding the payload in the same way as `gnrc_networking`. 

#### Steps to reproduce the issue

Compile one node with `examples/gnrc_networking` and another one with `tests/emb6`. Then send UDP packets between them (using the same payload) and compare the RX data.

#### Expected results

E.g for `udp send <dest_ip> <dest_port> abcdefgh`, both should receive `61  62  63  64  65  66  67  68`.
#### Actual results

GNRC node receives `AB CD EF` because the payload is parsed incorrectly from the shell.

jia200x, 2018-11-02:
I've renamed the title. We double checked and in theory is not a bug because the help command of `emb6` explicitly says the data is in hex format.

Nevertheless I still think we should have the same udp command for all examples/tests. So, I will live it open.

So, it's not a bug for the scope of the release.

miri64, 2018-11-05:
The encoding for those commands was intentional. Yes its differnent than the GNRC commands, but I prefer to send bytes rather than strings with this stack (same goes for the lwIP one).

closed 2018-11-05 by miri64

jia200x, 2018-11-05:
> The encoding for those commands was intentional. Yes its differnent than the GNRC commands, but I prefer to send bytes rather than strings with this stack (same goes for the lwIP one).

Ok, I see.

Does it make sense to add a `0x` prefix to the bytes in that case?

jia200x, 2018-11-05:
or even the other way around, adapt all udp commands to bytes. Sometimes is hard to decode pktdump messages if the payload is in ASCII

miri64, 2018-11-05:
> or even the other way around, adapt all udp commands to bytes. Sometimes is hard to decode pktdump messages if the payload is in ASCII

If we do it I would make it optional. Which other command*s* are you talking about? I'm only aware of the one in `gnrc_networking`. `posix_sockets` is working differently as well IIRC.

miri64, 2018-11-05:
Just to give some further reasons why I did it this way: for the interops e.g. we request a number of *bytes* as payload. With a string that can be confusing, as the `\0` might or might not be included. With bytes this is clearer.

miri64, 2018-11-05:
> Does it make sense to add a `0x` prefix to the bytes in that case? 

IMHO the self-documentation on the command is clear enough. `0x` as a requirement before the payload would only make the command more difficult to use.