RIOT-OS/RIOT #10317: tests/trickle: nondeterministic failure

jia200x, 2018-11-02:
#### Description
I've got results very random results with the `trickle` test. This snippet has been reported in at least `samr21-xpro`, `arduino-mega2560`, `iotlab-m3` and `nucleo-l152re`.
This is the kind of output of the test:
```
make RIOT_CI_BUILD=1 CC_NOCOLOR=1 --no-print-directory -C /home/kevinweiss/WorkingDirectory/RIOT/tests/trickle test
ssh -oProxyCommand="ssh weiss@riot-ci.inet.haw-hamburg.de -W %h:%p" -t pi@10.28.48.11 'socat - open:/dev/ttyACM0,b115200,echo=0,raw'
main(): This is RIOT! (Version: buildtest)
[START]
now = 44987, prev_now = 3935, diff = 41052
now = 138904, prev_now = 44987, diff = 93917
now = 380000, prev_now = 138904, diff = 241096
now = 830277, prev_now = 380000, diff = 450277
now = 1648560, prev_now = 830277, diff = 818283
[TRICKLE_RESET]
now = 1666973, prev_now = 1648560, diff = 18413
now = 1694341, prev_now = 1666973, diff = 27368
now = 1772710, prev_now = 1694341, diff = 78369
now = 1845080, prev_now = 1772710, diff = 72370
[FAILURE]
Timeout in expect script at "child.expect(u"now = \d+, prev_now = \d+, diff = \d+")" (tests/trickle/tests/01-run.py:22)

/home/kevinweiss/WorkingDirectory/RIOT/tests/trickle/../../Makefile.include:557: recipe for target 'test' failed
make: *** [test] Error 1

Return value: 2
```

This doesn't seem to be related to #9052 

#### Steps to reproduce the issue
This bug is hard to reproduce. However, it's run via `make -C tests/trickle flash test`.

#### Expected results

Test passes
#### Actual results

Test fails randomly.

jia200x, 2018-11-02:
With the help of @MrKevinWeiss we discovered the error is triggered with certain seeds. That explains why it's hard to reproduce (but it always happen in the same board!)

E.g `seed = 2392459496` makes it fail always:
```2018-11-02 17:25:13,706 - INFO # now = 158113, prev_now = 56132, diff = 101981
2018-11-02 17:25:13,707 - INFO # 200
2018-11-02 17:25:13,710 - INFO # trickle: I == 224, diff == 37, msg_time == 237000  ,trickle->t == 200 , old_interval: 112
2018-11-02 17:25:13,712 - INFO # now = 407721, prev_now = 158113, diff = 249608
2018-11-02 17:25:13,713 - INFO # 422
2018-11-02 17:25:13,716 - INFO # trickle: I == 448, diff == 24, msg_time == 446000  ,trickle->t == 422 , old_interval: 224
2018-11-02 17:25:13,718 - INFO # now = 866423, prev_now = 407721, diff = 458702
2018-11-02 17:25:13,719 - INFO # 788
2018-11-02 17:25:13,722 - INFO # trickle: I == 896, diff == 26, msg_time == 814000  ,trickle->t == 788 , old_interval: 448
2018-11-02 17:25:14,319 - INFO # now = 1693133, prev_now = 866423, diff = 826710
2018-11-02 17:25:14,320 - INFO # 1013
2018-11-02 17:25:14,328 - INFO # trickle: I == 1792, diff == 108, msg_time == 1121000  ,trickle->t == 1013 , old_interval: 896
2018-11-02 17:25:14,328 - INFO # 14
2018-11-02 17:25:14,335 - INFO # trickle: I == 16, diff == 0, msg_time == 14000  ,trickle->t == 14 , old_interval: 8
2018-11-02 17:25:14,337 - INFO # [TRICKLE_RESET]
2018-11-02 17:25:14,354 - INFO # now = 1728204, prev_now = 1693133, diff = 35071
2018-11-02 17:25:14,355 - INFO # 21
2018-11-02 17:25:14,362 - INFO # trickle: I == 32, diff == 2, msg_time == 23000  ,trickle->t == 21 , old_interval: 16
2018-11-02 17:25:14,389 - INFO # now = 1763459, prev_now = 1728204, diff = 35255
2018-11-02 17:25:14,390 - INFO # 63
2018-11-02 17:25:14,397 - INFO # trickle: I == 64, diff == 11, msg_time == 74000  ,trickle->t == 63 , old_interval: 32
2018-11-02 17:25:14,476 - INFO # now = 1849803, prev_now = 1763459, diff = 86344
2018-11-02 17:25:14,477 - INFO # 67
2018-11-02 17:25:14,484 - INFO # trickle: I == 128, diff == 1, msg_time == 68000  ,trickle->t == 67 , old_interval: 64
2018-11-02 17:25:14,557 - INFO # now = 1930149, prev_now = 1849803, diff = 80346
2018-11-02 17:25:14,557 - INFO # 180
2018-11-02 17:25:14,565 - INFO # trickle: I == 256, diff == 61, msg_time == 241000  ,trickle->t == 180 , old_interval: 128
2018-11-02 17:25:14,566 - INFO # [FAILURE]
```

cgundogan, 2018-11-02:
Looking further into this issue: The actual behaviour displayed here is **not critical**. The trickle intervals are still monotonically increasing.

The test application has very strict conditions and shows that something is off with certain seeds (see comment of @jia200x). Nevertheless, for the actual operation of trickle and RPL, which is the only *user* of trickle currently, this behaviour does **not lead to issues**.

It's still unclear why certain seeds behave this way and we should definitively investigate further.

IMO this is no show-stopper for the release, because it poses no problems for the actual trickle and RPL operation.

PR #10328 (2018-11-05): test: trickle: fix condition for success
### Contribution description

The current test implementation wrongly assumes that the diff between
two fired events (e1, e2) must always increase. That is not true, as
event e1 may reside on the upper part of [I/2, I) and e2 on the lower
part of [I, 2*I).

This commit fixes the test to look at the actual time that was randonmly
chosen from both intervals (t1, t2). Given that the intervals are
doubled, t1 must always be smaller than t2.

### Testing procedure
--

### Issues/PRs references
Fixes #10317
PR #10332 (2018-11-05): test: trickle: fix condition for success [backport 2018.10]
# backport of #10328 

### Contribution description

The current test implementation wrongly assumes that the diff between
two fired events (e1, e2) must always increase. That is not true, as
event e1 may reside on the upper part of [I/2, I) and e2 on the lower
part of [I, 2*I).

This commit fixes the test to look at the actual time that was randonmly
chosen from both intervals (t1, t2). Given that the intervals are
doubled, t1 must always be smaller than t2.

### Testing procedure
--

### Issues/PRs references
Fixes #10317
closed 2018-11-05 by cgundogan