RIOT-OS/RIOT #10318: ds1307: multiple read/write regs don't increment pointer

MrKevinWeiss, 2018-11-02:
#### Description
On the ds1307 RTC the pointer doesn't seem to increment when trying to i2c_read_reg or i2c_write_reg.  As well as the reg 15 seems to have some issue either reading or writing.

#### Steps to reproduce the issue
`BOARD=iotlab-m3 make flash term -C tests/periph_i2c/`
`i2c_read_regs 0 104 0 2 0`

#### Expected results
`i2c_0 read 2 byte(s) from reg 0x00 : [0x42, 0x10]`
Since it is an RTC values could be the same, just increase read amount to a larger number, a different number should be expected.
#### Actual results
`i2c_0 read 2 byte(s) from reg 0x00 : [0x42, 0x42]`

MrKevinWeiss, 2018-11-02:
Replacing all the i2c_write/read_regs commands with individual i2c_write/read_reg commands call several times so the pointer is set each time seems to make it work.  However I cannot get the i2c_write/read_reg for register 15 (aka 0x0f) working.  It is supposed to be ram meaning I should be able to write a value and it saves it then I should be able to read it back.  Maybe the clk is too fast seeing as it is only on 0x0f and all the other registers are fine.

kYc0o, 2018-11-02:
Side question: how did you manage to attach a ds1307 to an iotlab-m3 board?

kYc0o, 2018-11-02:
Just executed the command in a nrf52dk with a ds1307 shield and I have successful output:

```console
> i2c_read_regs 0 104 0 2 0
2018-11-02 17:31:40,723 - INFO #  i2c_read_regs 0 104 0 2 0
2018-11-02 17:31:40,727 - INFO # Command: i2c_read_regs(0, 0x68, 0x00, 2, 0x00)
2018-11-02 17:31:40,730 - INFO # Success: i2c_0 read 2 byte(s) from reg 0x00 : [0xc2, 0x10]
> 
```

MrKevinWeiss, 2018-11-02:
Hmmm, the iotlab-m3 node has one on board.  I used that.  I also connected a nucleo-l4xx and a samr21-xpro to the iotlab-m3 i2c bus (to get access to the ds1307 since i don't have a breakout board).  Everything was the same.

Can you also try 
```
i2c_write_reg 0 104 15 0 0
i2c_read_reg 0 104 15 0
i2c_write_reg 0 104 15 1 0
i2c_read_reg 0 104 15 0
```
to see if the write/read to the RTC ram problem is only with the iotlab-m3 node.

kYc0o, 2018-11-05:
```console
> i2c_write_reg 0 104 15 0 0
2018-11-05 16:17:42,594 - INFO #  i2c_write_reg 0 104 15 0 0
2018-11-05 16:17:42,598 - INFO # Command: i2c_write_reg(0, 0x68, 0x0f, 0x00, [0x00])
2018-11-05 16:17:42,601 - INFO # Success: i2c_0 wrote 1 byte
> i2c_read_reg 0 104 15 0
2018-11-05 16:17:50,229 - INFO #  i2c_read_reg 0 104 15 0
2018-11-05 16:17:50,231 - INFO # Command: i2c_read_reg(0, 0x68, 0x0f, 0x00)
2018-11-05 16:17:50,235 - INFO # Success: i2c_0 read 1 byte(s) from reg 0x0f : [0x00]
> i2c_write_reg 0 104 15 1 0
2018-11-05 16:18:04,150 - INFO #  i2c_write_reg 0 104 15 1 0
2018-11-05 16:18:04,152 - INFO # Command: i2c_write_reg(0, 0x68, 0x0f, 0x00, [0x01])
2018-11-05 16:18:04,155 - INFO # Success: i2c_0 wrote 1 byte
> i2c_read_reg 0 104 15 0
2018-11-05 16:18:10,933 - INFO #  i2c_read_reg 0 104 15 0
2018-11-05 16:18:10,937 - INFO # Command: i2c_read_reg(0, 0x68, 0x0f, 0x00)
2018-11-05 16:18:10,941 - INFO # Success: i2c_0 read 1 byte(s) from reg 0x0f : [0x01]
>
```

Just to clear the situation, and avoid confusion, I strongly believe the iotlab-m3 node **has no** ds1307 on board, according to the [doc](https://github.com/iot-lab/iot-lab/wiki/Hardware-M3-node) and [schematics](http://github.com/iot-lab/iot-lab/wiki/Docs/openm3-schematics.pdf).

Is it relevant for your issue according to my results?

Again my setup: nrf52dk + arduino shield with ds1307. tests/driver_ds1307/ also pass with my setup.

MrKevinWeiss, 2018-11-06:
...Thanks @kYc0o.  This whole time I was testing on the L3G4200 on the iotlab-m3 as it has the same address (104).  I think the tests/driver_ds1307 is a bit misleading as it checks the address and if it is there it assumes it is the ds1307 is connected then attempts to run tests.

MrKevinWeiss, 2018-11-06:
At least after your tests we can check off the untested ACK that was given from the i2c embargo.

closed 2018-11-06 by MrKevinWeiss

miri64, 2018-11-06:
> I think the tests/driver_ds1307 is a bit misleading as it checks the address and if it is there it assumes it is the ds1307 is connected then attempts to run tests.

It was my very first driver implementation and I wasn't aware that different devices can have different addresses. How can one make sure the ds1307 is connected?

MrKevinWeiss, 2018-11-06:
@miri64 Sometimes the drivers have a WHOAMI register.  This one doesn't so it isn't that straightforward.  I really should have known that it was a different device... I even checked the schematic and was like 'maybe they just didn't include it'.  Silly mistake on my part.

MrKevinWeiss, 2018-11-06:
I am not sure if the test should have a whitelist or ideally a driver provided macro or something.

cladmi, 2018-11-13:
That is a thing I would like to start introducing, that hardware peripherals (== what we have in drivers) are exposed as `FEATURES` and application depend on them with FEATURES_REQUIRED.
But there are still things to do before it can be nicely integrated.