RIOT-OS/RIOT #10335: Add identifier to the ds1307 driver

MrKevinWeiss, 2018-11-06:
#### Description
The ds1307_init will return a positive result if the address is ACK, since it shares the same address as the L3G4200 (address 104) it would think it is a correct ds1307.

It would be nice to have an additional check to ensure it is the correct device.  Maybe it can try to write an invalid day or read an invalid register to do so (since there is no WHOAMI register).

### Useful links
https://datasheets.maximintegrated.com/en/ds/DS1307.pdf
https://www.elecrow.com/download/L3G4200_AN3393.pdf

jcarrano, 2018-11-06:
There are some bits in some resisters that are always zero. It's not super-robust, but it may work to read those and check they are all zero.

MrKevinWeiss, 2018-11-06:
Ya I wanted to see if I could write 1's to them then check if they are still 0.  I need a board though (on the next order list).

gebart, 2018-11-06:
Is it safe to write some random value to a device which may or may not be the device you want to use?
IMO reads are mostly safe, but writing might well cause some big problems if it is the wrong device..

MrKevinWeiss, 2018-11-06:
Ya, true.  Should it just to a compare to known values or maybe focus on changing the test?  Or should we just say you know know what devices are connected to your boards.

miri64, 2018-11-06:
Side info: I just found my DS1307 breakout board again, so I can maintain and test that thing again when the time comes ;-)

PR #10697 (2019-01-03): tests/driver_ds1307: Blacklist iotlab boards
### Contribution description

This states iotlab boards shouldn't run these tests.
This is because the i2c address of L3G4200 is the same ad the ds1307.
This causes an error that thinks the test is valid when it is not.

### Testing procedure

The iotlab-m3 and iotlab-a8-m3 board should now state to expect errors for the test.

### Issues/PRs references
Fixes #10335
closed 2019-01-03 by miri64