RIOT-OS/RIOT #10338: xbee: setting PAN ID sometimes fails

miri64, 2018-11-06:
#### Description
While testing arduino-zero+xbee to samr21-xpro communication during the testing phase of 2018.10 I noticed, that sometimes the initialization of the network interface fails. To check what happened I used the DEBUG output and it apears to be an issue in the PAN ID initialization:

```
[xbee] AT_CMD: +++
[xbee] AT_CMD: ATMM2
[xbee] AT_CMD: ATAP1
[xbee] AT_CMD: ATAC
[xbee] AT_CMD: ATCN
[xbee] AT_CMD: SH
[xbee] AT_CMD: SL
[xbee] AT_CMD: CH
[xbee] AT_CMD: ID
[xbee] api_at_cmd: response timeout
[xbee] init: error setting PAN ID
gnrc_netif: netdev init failed: -5
```

(CH also seems to be somewhat errorneous, as it prints a non-ASCII symbol)

I'm not 100% sure, but I think it happens if a reset happens while the device is sending something, but I also saw it directly on the first initialization after flash.

Would be great (or not so great for our stability) if someone can check if that also happens with xbee on other boards than arduino-zero.

#### Steps to reproduce the issue
Use a board (preferrably to make it easier one without a radio) and attach an xbee shield (e.g. arduino-zero in IoT-LAB. Compile `gnrc_networking` with the `xbee` module and flash it to the board and open the terminal. Try to reset several times, sometimes it works quite stable.

#### Expected results
Always reset to a workable state.

#### Actual results
Sometimes you will see the `gnrc_netif: netdev init failed: -5` message shown above (or if you miss it, you won't be able to use the shell, since it isn't initialized due to the error).

#### Versions
[IoT-LAB's arduino-zero](https://www.iot-lab.info/hardware/custom/) + my local setup:

```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "16.04.5 LTS (Xenial Xerus)"
                 Kernel: Linux 4.4.0-138-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bits) 7.2.0
                  clang: clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: "2.1.0"
riscv-none-embed-newlib: "2.5.0"
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.0
               cppcheck: Cppcheck 1.72
                doxygen: 1.8.11
                 flake8: 2.5.4 (pep8: 1.7.0, pyflakes: 1.1.0, mccabe: 0.2.1) CPython 3.5.2 on Linux
                    git: git version 2.7.4
                openocd: Open On-Chip Debugger 0.10.0+dev-00207-g4109263 (2017-11-02-16:35)
                 python: Python 2.7.12
                python2: Python 2.7.12
                python3: Python 3.5.2
             coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

miri64, 2020-07-02:
@fjmolinas do you know if https://github.com/RIOT-OS/RIOT/pull/13224 maybe has fixed that as well?

fjmolinas, 2020-07-03:
I'll try to reproduce, was this easy to reproduce at the time?

fjmolinas, 2020-07-03:
I still get non ASCII characters sometimes

```
> [xbee] AT_CMD: +++
[xbee] AT_CMD: ATMM2
[xbee] AT_CMD: ATAP1
[xbee] init: WARNING if using an arduino BOARD + arduino xbee shield with ICSP connector, hardware flow control can't be used since CTS pin is connected to ICSP RESET pin
[xbee] AT_CMD: ATD6 0
[xbee] AT_CMD: ATD7 0
[xbee] AT_CMD: ATAC
[xbee] AT_CMD: ATCN
[xbee] AT_CMD: SH
[xbee] AT_CMD: SL
[xbee] AT_CMD: CH
[xbee] AT_CMD: ID
[xbee] init: Initialization successful
[xbee] AT_CMD: MYD
```

But I have reseted and flashed multiple times and it succeeds every time.

fjmolinas, 2020-07-03:
This is a local test, not over iot-lab.

miri64, 2020-07-03:
Ok, non-ASCII chars I consider not in the scope of this PR, so I consider this fixed.

fjmolinas, 2020-07-16:
I'm re-opening this since I saw it again during release testing, maybe having debugging enabled somehow affected the initial result, but I got `gnrc_netif: netdev init failed: -5` message a couple of times when running 04-single-hop-6lowpan-icmp task 07-08 . I'll try to debug this again.