RIOT-OS/RIOT #10341: driver_my9221 prevents further flashing of nucleo-l073rz

MrKevinWeiss, 2018-11-06:
#### Description
When running the compile_and_test_for_board script the driver_my9221 test gets flashed on the nucleo-l073rz and as a result all further attempts to flash and reset are prevented.  This can be fixed by manually holding down the reset pin just before flashing.  Once flashed with working firmware after the manual reset button press it appears to be able to flash like normal again.

#### Steps to reproduce the issue
`BOARD=nucleo-l073rz make flash -C tests/driver_my9221`
`BOARD=nucleo-l073rz make flash -C examples/hello-world`

#### Expected results
hello-world can flash

#### Actual results
```Error: init mode failed (unable to connect to the target)
in procedure 'init' 
in procedure 'ocd_bouncer'

~/RIOT/tests/driver_my9221/../../Makefile.include:517: recipe for target 'flash' failed
make: *** [flash] Error 1
```

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "16.04.5 LTS (Xenial Xerus)"
                 Kernel: Linux 4.15.0-38-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 4.9.2
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: clang version 5.0.0-3~16.04.1 (tags/RELEASE_500/final)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "1.8.0svn" ("20111229")

Installed development tools
---------------------------
                  cmake: cmake version 3.12.3
               cppcheck: missing
                doxygen: missing
                 flake8: 3.5.0 (mccabe: 0.6.1, pycodestyle: 2.3.1, pyflakes: 1.6.0) CPython 3.5.2 on Linux
                    git: git version 2.7.4
                openocd: Open On-Chip Debugger 0.10.0+dev-00399-g09076d1 (2018-04-12-17:12)
                 python: Python 2.7.12
                python2: Python 2.7.12
                python3: Python 3.5.2
             coccinelle: missing
```

aabadie, 2018-11-06:
Maybe try to use the same openocd configuration as [the one used by b-l072z-lrwan1](https://github.com/RIOT-OS/RIOT/blob/master/boards/b-l072z-lrwan1/dist/openocd.cfg)

MrKevinWeiss, 2018-11-07:
It is not the only test that crashes it but it appears adding the `connect_assert_srst` in the openocd config fixes it.  Thanks @aabadie.  The frdm-k22f also seems to have this problem.  I will try to see if this can fix it then make a PR.

MrKevinWeiss, 2018-11-07:
it was emb6 test that locked the frdm-k22f board

aabadie, 2018-11-07:
I tried locally and can't reproduce the issue. The board is flashed smoothly in all cases.

MrKevinWeiss, 2018-11-07:
In master it doesn't happen anymore but with RC2 or old releases I was able to reproduce.
Could it be my openocd version?
```
openocd --version
Open On-Chip Debugger 0.10.0+dev-00399-g09076d1 (2018-04-12-17:12)
```

MrKevinWeiss, 2018-11-07:
The frdm-k22f board is something different.  It seems to get into a debug more but I don't have one on my desk so it is hard to test.

before flashing
```
START...                                                                                                                                                                                                                                                                           
Info : kx.cpu: hardware has 6 breakpoints, 4 watchpoints                                                                                                                                                                                                                           
Info : MDM: Chip is unsecured. Continuing.                                                                                                                                                                                                                                         
Info : Listening on port 41033 for gdb connections                                                                                                                                                                                                                                 
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* kx.cpu             cortex_m   little kx.cpu             running
START...
Info : MDM: Chip is unsecured. Continuing.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0000074c msp: 0x1fff0200
auto erase enabled
Info : Kinetis MK22FN512xxx12 detected: 2 flash blocks
Info : 2 PFlash banks: 512k total
Info : Padding image section 1 with 48 bytes
Warn : Missing bank 1 configuration, FCF protection flags may be incomplette
Info : This device supports Program Longword execution only.
Info : Disabling Kinetis watchdog (initial WDOG_STCTRLH = 0x01d3)
Info : WDOG_STCTRLH = 0x01d2
Warn : Missing bank 1 configuration, FCF protection flags may be incomplette
Info : This device supports Program Longword execution only.
wrote 51200 bytes from file /home/pi/RIOT/tests/emb6/bin/frdm-k22f/tests_emb6.elf in 2.821835s (17.719 KiB/s)
34 bytes written at address 0x20000000
downloaded 34 bytes in 0.003017s (11.005 KiB/s)
target halted due to breakpoint, current mode: Thread
xPSR: 0x01000000 pc: 0x20000020 msp: 0x1fff0200
verified 50600 bytes in 0.707899s (69.804 KiB/s)
START...
Info : MDM: Chip is unsecured. Continuing.
Polling target kx.cpu failed, trying to reexamine
START...
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 100ms
shutdown command invoked
Done flashing
make: Leaving directory '/home/pi/RIOT/tests/emb6'
```

Then all following flashing looks like:
```
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: SWD connect failed
in procedure 'init'
in procedure 'ocd_bouncer'

/home/pi/RIOT/examples/hello-world/../../Makefile.include:516: recipe for target 'flash' failed
make: *** [flash] Error 1
```

aabadie, 2018-11-07:
Please don't mix boards in this issue, or change the issue title.

aabadie, 2018-11-07:
Otherwise, I can indeed reproduce the issue on nucleo-l073rz and RC2 but not master. It would be interesting to track what was changed in between the 2.
Because then #10343 might not be the only solution.

MrKevinWeiss, 2018-11-07:
Yup I will open a separate issue, I thought it was the same problem but it appears to be something different.

MrKevinWeiss, 2018-11-13:
I found a related issue https://github.com/vsergeev/arm-bmw-sw/issues/1

It seems the fix is reset_config none separate

PR #10343 (2019-03-14): boards/stm32l0: Fix openocd to prevent flash locking
### Contribution description

This commit add the connect_assert_srst which requires the SRST to be asserted before trying to connect.
Certain firmwares will cause it to get to a state where flashing fails and the reset button must be pushed at the correct time to recover.

### Testing procedure
on 7645c66aeb584a643d7356ba41f1ecf80b06dfe8 commit
`BOARD=nucleo-l073rz make flash -C tests/driver_my9221`
then try flashing anything after.
_It seems in master the driver_my9221 doesn't cause lockup anymore but if it is already locked up it doesn't  [... cut]
closed 2019-03-14 by aabadie