RIOT-OS/RIOT #10345: frdm-k22f cannot flash after certain firmware flashed

MrKevinWeiss, 2018-11-07:
#### Description
When flashing the tests/emb6, flash stops working after and the reset button must be pressed at the right time to reflash.
_only emb6 firmware was found to cause this problem but other firmwares may also cause this_

#### Steps to reproduce the issue
`BOARD=frdm-k22f make flash -C tests/emb6`
then flash any firmware

#### Expected results
flash succeed

#### Actual results
before flashing
```
START...                                                                                                                                                                                                                                                                           
Info : kx.cpu: hardware has 6 breakpoints, 4 watchpoints                                                                                                                                                                                                                           
Info : MDM: Chip is unsecured. Continuing.                                                                                                                                                                                                                                         
Info : Listening on port 41033 for gdb connections                                                                                                                                                                                                                                 
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* kx.cpu             cortex_m   little kx.cpu             running
START...
Info : MDM: Chip is unsecured. Continuing.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0000074c msp: 0x1fff0200
auto erase enabled
Info : Kinetis MK22FN512xxx12 detected: 2 flash blocks
Info : 2 PFlash banks: 512k total
Info : Padding image section 1 with 48 bytes
Warn : Missing bank 1 configuration, FCF protection flags may be incomplette
Info : This device supports Program Longword execution only.
Info : Disabling Kinetis watchdog (initial WDOG_STCTRLH = 0x01d3)
Info : WDOG_STCTRLH = 0x01d2
Warn : Missing bank 1 configuration, FCF protection flags may be incomplette
Info : This device supports Program Longword execution only.
wrote 51200 bytes from file /home/pi/RIOT/tests/emb6/bin/frdm-k22f/tests_emb6.elf in 2.821835s (17.719 KiB/s)
34 bytes written at address 0x20000000
downloaded 34 bytes in 0.003017s (11.005 KiB/s)
target halted due to breakpoint, current mode: Thread
xPSR: 0x01000000 pc: 0x20000020 msp: 0x1fff0200
verified 50600 bytes in 0.707899s (69.804 KiB/s)
START...
Info : MDM: Chip is unsecured. Continuing.
Polling target kx.cpu failed, trying to reexamine
START...
Error: Could not initialize the debug port
Examination failed, GDB will be halted. Polling again in 100ms
shutdown command invoked
Done flashing
make: Leaving directory '/home/pi/RIOT/tests/emb6'
```

Then all following flashing looks like:
```
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: SWD connect failed
in procedure 'init'
in procedure 'ocd_bouncer'

/home/pi/RIOT/examples/hello-world/../../Makefile.include:516: recipe for target 'flash' failed
make: *** [flash] Error 1
```

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Raspbian GNU/Linux" "9 (stretch)"
                 Kernel: Linux 4.14.52-v7+ armv7l unknown

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.4.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.7.2
               cppcheck: Cppcheck 1.76.1
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.11.0
                openocd: Open On-Chip Debugger 0.10.0+dev-00313-gbae76053 (2018-03-02-11:48)
                 python: Python 2.7.13
                python2: Python 2.7.13
                python3: Python 3.5.3
             coccinelle: missing
```

may relate to #10341

gebart, 2018-11-07:
I can confirm this problem is seen sometimes. It sometimes works with reset_config connect_assert_srst, but not always. Still looking for a cure.

MrKevinWeiss, 2018-11-07:
Ya I tried with adding that and it didn't fix it

MrKevinWeiss, 2018-11-13:
I found a related issue https://github.com/vsergeev/arm-bmw-sw/issues/1

It seems the fix is reset_config none separate

MrKevinWeiss, 2019-08-12:
I haven't had it in a while.   I think we can close and if it happens again just reopen!

closed 2019-08-12 by MrKevinWeiss