RIOT-OS/RIOT #10353: cpu/nrf5x_common: hwrng hangs

PeterKietzmann, 2018-11-09:
#### Description
The hwrng driver hangs when calling `hwrng_read()`

#### Steps to reproduce the issue
- Compile+flash 'tests/rng* to a `nrf840dk` board
- change the random source to hwrng by `source 2`
- try to dump some values `dump 3`
- when there's no output, call `help` 3 times

#### Expected results
The terminal outputs a 32-bit number.

#### Actual results
The driver sucks supposedly [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/nrf5x_common/periph/hwrng.c#L47). 
A. 
Adding `puts("abc")` one line above theevent flag polling makes it work again.
B.
Not applying the above mentioned puts **but** calling any other shell function makes the driver proceed. 

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "18.04.1 LTS (Bionic Beaver)"
                 Kernel: Linux 4.15.0-38-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.17.1
                openocd: Open On-Chip Debugger 0.10.0
                 python: Python 2.7.15rc1
                python2: Python 2.7.15rc1
                python3: Python 3.6.6
             coccinelle: missing
```

PeterKietzmann, 2018-11-09:
@SemjonKerner can you have a look at this?

SemjonKerner, 2018-11-09:
@PeterKietzmann I am very busy with nrf52s radio driver, if this can wait a week I will look into it.

PeterKietzmann, 2018-11-09:
Yes it can wait. Would be great if you had a look as soon as you find the time.

SemjonKerner, 2018-11-14:
I can reproduce the error. When I keep trying some dumps and helps eventually it even loops in the dump state. I'll see whether I can find the source of it.

SemjonKerner, 2018-11-14:
Just for completeness: I found that the function `cortexm_sleep_until_event()` was not correctly initialized to the rng interrupt. There was also a simple solution given in [nordics dev forum](https://devzone.nordicsemi.com/f/nordic-q-a/2959/isn-t-nrf_rng--events_valrdy-an-event-you-can-wait-for-with-__wfe?ReplySortBy=CreatedDate&ReplySortOrder=Ascending) to the exact same problem.

PR #10387 (2018-11-14): cpu/nrf5x: fix hwrng hang
This PR fixes a minor bug as mentioned in  #10353.
The original implementation did not utilize the corresponding rng-interrupt leading to a freeze until an uncorellated interrupt happens.

Resolves #10353.
closed 2018-11-14 by miri64